fwupd/fwupd #4378: A different "RemoveDelay" setting in quirk is required after firmware update and in case of simple device removal

chlee75, 2022-03-14:
RemoveDelay value in quirk is used for device removal timeout when device is detached.
However device removal timeout could be different when device is removed after firmware update and when device is simply removed without firmware update.
It is because that device need more time to validating firmware after firmware update or writing firmware stored in flash memory after firmware update  rather than when device is simply detached.

If we use same device remove timeout, device may not be recognized in time when device is detached after firmware update.

So I suggest to use different device remove timeout value in case of when firmware is updated and when device is simply detached.

e.g. 
 RemoveDelay is device removal timeout when device is simply detached.
 UpdateRemoveDelay is device removal timeout when device is detached after firmware update.

superm1, 2022-03-14:
What device type is this request in relation to?

I think in general the request makes sense, but why not just pick the longer delay to start? Make it a worst case value.

If the device comes back more quickly I would think that the daemon should handle that just fine.

hughsie, 2022-03-14:
I think Ryan is more talking about the:

 * remove device
 * 20 seconds later the device is still in the `fwupdmgr get-devices --show-all` list

@chlee75 is that correct?

hughsie, 2022-03-14:
@chlee75 would something like the wip/hughsie/ONLY_WFR_REMOVE_DELAY branch work? 100.0% untested.

chlee75, 2022-03-15:
@hughsie 

I have been checking  wip/hughsie/ONLY_WFR_REMOVE_DELAY branch.
DMC dock works fine. 
But I can't check it for Lenovo Dock Gen2 due to firmware update error. See https://github.com/fwupd/fwupd/issues/4382.
Issue#4382 is not due to wip/hughsie/ONLY_WFR_REMOVE_DELAY but due to current main line.

I will check  wip/hughsie/ONLY_WFR_REMOVE_DELAY for Lenovo Gen2 Dock after Issue#4382 has been fixed.

Thanks.

closed 2022-03-15 by hughsie