fwupd/fwupd #4388: gpio: device is not open for /sys/class/gpio/gpiochip152

CragW, 2022-03-15:
**Describe the bug**
if plugin `gpio` is enabled then g_warning message will be seen:
```
14:14:39:0300 FuEngine             failed to add device /sys/devices/platform/INT34C5:00/gpio/gpiochip152: failed to get chipinfo: 45e57a8f68f89837238630ab65cb49a342f1a4d8 [(null)] has not been opened
```

**Steps to Reproduce**
fwupdtool get-devices

**Expected behavior**
No warning message

**fwupd version information**
compile   org.freedesktop.fwupd         1.8.0
runtime   org.freedesktop.gusb          0.3.5
runtime   org.kernel                    5.13.0-35-generic
compile   org.freedesktop.gusb          0.3.5
compile   com.hughsie.libjcat           0.1.9
runtime   org.freedesktop.fwupd         1.8.0

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
source

**fwupd device information**
```
Loading…                 [***************************************]
WARNING: This package has not been validated, it may not work properly.
XPS 13 9310
│
├─11th Gen Intel Core™ i7-1185G7 @ 3.00GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x0000009a
│     Vendor:             Intel
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f ← cpu
│                         30249f37-d140-5d3e-9319-186b1bd5cac3 ← CPUID\PRO_0&FAM_06
│                         2347ad97-b545-5bae-b6b3-b70a9553cbba ← CPUID\PRO_0&FAM_06&MOD_8C
│                         1df99df5-22f4-5505-98cc-22dcc53b132a ← CPUID\PRO_0&FAM_06&MOD_8C&STP_1
│     Device Flags:       • Internal device
│   
├─KXG60PNV2T04 NVMe KIOXIA 2048GB:
│     Device ID:          c430a03ca2a65dfe2412ff950c79c51f6aec1317
│     Summary:            NVM Express solid state drive
│     Current version:    10704104
│     Vendor:             Toshiba Corporation (NVME:0x1179)
│     Serial Number:      Z0LF305KFSH1
│     Update Error:       Cannot install update when not on AC power
│     GUIDs:              73a0c81f-2555-56a5-bb6a-c2aeab2213d1 ← STORAGE-DELL-108638
│                         558810bc-de64-11e8-9f32-f2801f1b9fd1
│     Device Flags:       • System requires external power source
│                         • Needs a reboot after installation
│                         • Updatable
│                         • Signed Payload
│   
├─System Firmware:
│ │   Device ID:          a45df35ac0e948ee180fe216a5f703f32dda163f
│ │   Summary:            UEFI ESRT device
│ │   Current version:    197632
│ │   Minimum Version:    197632
│ │   Vendor:             Dell Inc. (DMI:Dell Inc.)
│ │   Update State:       Success
│ │   Update Error:       Cannot install update when not on AC power
│ │   GUIDs:              cfb08d7c-8d3d-41b1-846b-48f634dfac9d
│ │                       230c8b18-8d9b-53ec-838b-6cfc0383493a ← main-system-firmware
│ │   Device Flags:       • Internal device
│ │                       • System requires external power source
│ │                       • Needs a reboot after installation
│ │                       • Cryptographic hash verification is available
│ │                       • Device is usable for the duration of the update
│ │                       • Updatable
│ │ 
│ └─UEFI dbx:
│       Device ID:        362301da643102b9f38477387e2193e57abaa590
│       Summary:          UEFI revocation database
│       Current version:  83
│       Minimum Version:  83
│       Vendor:           UEFI:Linux Foundation
│       Install Duration: 1 second
│       GUIDs:            00fe3755-a4d8-5ef7-ba5f-47979fbb3423 ← UEFI\CRT_E28D59CA489BD2AD580F2EA5D62D6A29BB9C02AE5A818434A37DA7FC11DFF9E9
│                         4a6cd2cb-8741-5257-9d1f-89a275dacca7 ← UEFI\CRT_E28D59CA489BD2AD580F2EA5D62D6A29BB9C02AE5A818434A37DA7FC11DFF9E9&ARCH_X64
│                         c6682ade-b5ec-57c4-b687-676351208742 ← UEFI\CRT_A1117F516A32CEFCBA3F2D1ACE10A87972FD6BBE8FE0D0B996E09E65D802A503
│                         f8ba2887-9411-5c36-9cee-88995bb39731 ← UEFI\CRT_A1117F516A32CEFCBA3F2D1ACE10A87972FD6BBE8FE0D0B996E09E65D802A503&ARCH_X64
│       Device Flags:     • Internal device
│                         • Updatable
│                         • Needs a reboot after installation
│                         • Only version upgrades are allowed
│                         • Signed Payload
│     
├─TPM:
│     Device ID:          c6a80ac3a22083423992a3cb15018989f37834d6
│     Current version:    1.257.0.0
│     Vendor:             ST Microelectronics (TPM:STM)
│     GUIDs:              ff71992e-52f7-5eea-94ef-883e56e034c6 ← system-tpm
│                         84df3581-f896-54d2-bd1a-372602f04c32 ← TPM\VEN_STM&DEV_0001
│                         bfaed10a-bbc1-525b-a329-35da2f63e918 ← TPM\VEN_STM&MOD_
│                         70b7b833-7e1a-550a-a291-b94a12d0f319 ← TPM\VEN_STM&DEV_0001&VER_2.0
│                         06f005e9-cb62-5d1a-82d9-13c534c53c48 ← TPM\VEN_STM&MOD_&VER_2.0
│     Device Flags:       • Internal device
│   
├─TPM 2.0:
│     Device ID:          a3487e128cf1413519bce8e9a1ab3f5981e61458
│     Summary:            UEFI ESRT device
│     Current version:    0.1.1.1
│     Vendor:             Dell Inc. (PCI:0x1028)
│     Update State:       Success
│     Update Error:       Cannot install update when not on AC power
│     GUIDs:              fd112308-8a5f-5c8e-9e89-1a8df757f4a1 ← 0a43-2.0
│                         ff71992e-52f7-5eea-94ef-883e56e034c6 ← system-tpm
│                         73730635-f6c2-53da-9df2-948bb5ac1022 ← DELL-TPM-2.0-STM-
│     Device Flags:       • Internal device
│                         • System requires external power source
│   
├─UEFI Device Firmware:
│     Device ID:          349bb341230b1a86e5effe7dfe4337e1590227bd
│     Summary:            UEFI ESRT device
│     Current version:    154
│     Minimum Version:    154
│     Vendor:             DMI:Dell Inc.
│     Update State:       Success
│     Update Error:       Cannot install update when not on AC power
│     GUID:               176e090d-0ddb-495e-8173-bc998ccfecd0
│     Device Flags:       • Internal device
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│                         • Updatable
│   
├─UEFI Device Firmware:
│     Device ID:          2292ae5236790b47884e37cf162dcf23bfcd1c60
│     Summary:            UEFI ESRT device
│     Current version:    275792132
│     Minimum Version:    275792132
│     Vendor:             DMI:Dell Inc.
│     Update State:       Success
│     Update Error:       Cannot install update when not on AC power
│     GUID:               558810bc-de64-11e8-9f32-f2801f1b9fd1
│     Device Flags:       • Internal device
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│                         • Updatable
│   
├─USB4 host controller:
│     Device ID:          7b9f9bd417dd96a9b1c432fb5e585fa397c4cf63
│     Summary:            Unmatched performance for high-speed I/O
│     Update Error:       Cannot install update when not on AC power, No vendor ID set
│     GUID:               e72e778e-94f7-5ed2-b560-1c1262ee217c ← TBT-fixed
│     Device Flags:       • Internal device
│                         • System requires external power source
│                         • Updatable
│                         • Signed Payload
│   
└─USB4 host controller:
      Device ID:          3555a1f8ba8dfbfe0f6b0c3f385dc520067ff475
      Summary:            Unmatched performance for high-speed I/O
      Update Error:       Cannot install update when not on AC power, No vendor ID set
      GUID:               e72e778e-94f7-5ed2-b560-1c1262ee217c ← TBT-fixed
      Device Flags:       • Internal device
                          • System requires external power source
                          • Updatable
                          • Signed Payload
```

**Additional questions**
- Operating system and version: Ubuntu 21.10
- Have you tried rebooting? Yes
- Is this a regression? No

CragW, 2022-03-15:
Should we only consider `/dev/gpiochip*` and skip devices from `/sys/class/gpio/`?

superm1, 2022-03-15:
BTW I think that's unexpected that the USB4 host controller is marked updatable from the OS.  I don't believe that's possible to update.

CragW, 2022-03-16:
After a closer look to my system, hereunder are the udevices. The 1st node from /sys/class/gpio/gpiochip152 can't be opened successfully as `device_file` is `0x0` so `priv->fd` won't get the file descriptor. The 2nd node from `/dev/gpiochip0` can be opened and added to engine successfully.

```
P: /devices/platform/INT34C5:00/gpio/gpiochip152
L: 0
E: DEVPATH=/devices/platform/INT34C5:00/gpio/gpiochip152
E: SUBSYSTEM=gpio

P: /devices/platform/INT34C5:00/gpiochip0
N: gpiochip0
L: 0
E: DEVPATH=/devices/platform/INT34C5:00/gpiochip0
E: SUBSYSTEM=gpio
E: DEVNAME=/dev/gpiochip0
E: MAJOR=254
E: MINOR=0
```
```
$ sudo ./dist/fwupdtool.sh get-devices --plugins gpio --show-all -v
02:10:06:0002 FuDebug              Verbose debugging enabled (on console 1)
02:10:06:0002 FuMain               locked /run/lock/fwupdtool
02:10:06:0002 FuConfig             trying to load config values from /home/crabx/mywork/fwupd/dist/etc/fwupd/daemon.conf
02:10:06:0002 FuCommon             reading /home/crabx/mywork/fwupd/dist/etc/fwupd/daemon.conf with 1930 bytes
02:10:06:0002 FuConfig             trying to load config values from /home/crabx/mywork/fwupd/dist/etc/fwupd/daemon.conf
02:10:06:0002 FuCommon             reading /home/crabx/mywork/fwupd/dist/etc/fwupd/daemon.conf with 1930 bytes
02:10:06:0002 FuConfig             using autodetected max archive size 18.4 EB
02:10:06:0003 GLib-GIO             _g_io_module_get_default: Found default implementation local (GLocalVfs) for ‘gio-vfs’
02:10:06:0003 FuRemoteList         loading /home/crabx/mywork/fwupd/dist/share/fwupd/metainfo
02:10:06:0003 XbSilo               attempting to load /home/crabx/mywork/fwupd/dist/cache/fwupd/metainfo.xmlb
02:10:06:0003 XbSilo               file: 58cf8bac-0a93-f0ad-a52f-98cc9a4baceb, current:58cf8bac-0a93-f0ad-a52f-98cc9a4baceb, cached: (null)
02:10:06:0003 XbSilo               loading silo with file contents
02:10:06:0003 FuRemoteList         loading remote from /home/crabx/mywork/fwupd/dist/etc/fwupd/remotes.d/lvfs-testing.conf
02:10:06:0003 FuRemoteList         loading remote from /home/crabx/mywork/fwupd/dist/etc/fwupd/remotes.d/lvfs.conf
02:10:06:0003 FuRemoteList         loading remote from /home/crabx/mywork/fwupd/dist/etc/fwupd/remotes.d/vendor.conf
02:10:06:0003 FuRemoteList         loading remote from /home/crabx/mywork/fwupd/dist/etc/fwupd/remotes.d/vendor-directory.conf
02:10:06:0003 FuRemoteList         loading remote from /home/crabx/mywork/fwupd/dist/etc/fwupd/remotes.d/dell-esrt.conf
02:10:06:0003 FuRemoteList         loading remote from /home/crabx/mywork/fwupd/dist/etc/fwupd/remotes.d/fwupd-tests.conf
02:10:06:0003 FuRemoteList         ordering lvfs-testing=lvfs+1
02:10:06:0003 FuRemoteList         ignoring unfound remote fwupd
02:10:06:0003 Jcat                 ignoring GPG-KEY-Linux-Foundation-Firmware as not PKCS-7 certificate
02:10:06:0003 Jcat                 trying to load certificate from /home/crabx/mywork/fwupd/dist/etc/pki/fwupd/LVFS-CA.pem
02:10:06:0003 Jcat                 reading /home/crabx/mywork/fwupd/dist/etc/pki/fwupd/LVFS-CA.pem with 1679 bytes
02:10:06:0004 Jcat                 loaded 1 certificates
02:10:06:0004 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
02:10:06:0004 Jcat                 ignoring GPG-KEY-Linux-Foundation-Metadata as not PKCS-7 certificate
02:10:06:0004 Jcat                 trying to load certificate from /home/crabx/mywork/fwupd/dist/etc/pki/fwupd-metadata/LVFS-CA.pem
02:10:06:0004 Jcat                 reading /home/crabx/mywork/fwupd/dist/etc/pki/fwupd-metadata/LVFS-CA.pem with 1679 bytes
02:10:06:0004 Jcat                 loaded 1 certificates
02:10:06:0004 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
02:10:06:0004 Jcat                 reading /home/crabx/mywork/fwupd/dist/lib/fwupd/pki/secret.key with 2455 bytes
02:10:06:0004 Jcat                 reading /home/crabx/mywork/fwupd/dist/lib/fwupd/pki/client.pem with 1383 bytes
02:10:06:0004 FuEngine             client certificate now exists: nothing to do
02:10:06:0004 FuHistory            trying to open database '/home/crabx/mywork/fwupd/dist/lib/fwupd/pending.db'
02:10:06:0004 FuHistory            got schema version of 7
02:10:06:0004 FuQuirks             loading quirks from /home/crabx/mywork/fwupd/dist/share/fwupd/quirks.d
02:10:06:0007 FuQuirks             loading quirks from /home/crabx/mywork/fwupd/dist/lib/fwupd/quirks.d
02:10:06:0007 XbSilo               attempting to load /home/crabx/mywork/fwupd/dist/cache/fwupd/quirks.xmlb
02:10:06:0007 XbSilo               file: 94431eb2-7a58-bbb1-7bdc-9940ff7e2cf9, current:94431eb2-7a58-bbb1-7bdc-9940ff7e2cf9, cached: (null)
02:10:06:0007 XbSilo               loading silo with file contents
02:10:06:0007 FuSmbios             trying to read /sys/class/dmi/id
02:10:06:0007 FuHwids              loading HwId overrides from /home/crabx/mywork/fwupd/dist/etc/fwupd/daemon.conf
02:10:06:0007 FuHwids              not loading HwId overrides from /home/crabx/mywork/fwupd/dist/lib/fwupd/daemon.conf
02:10:06:0007 FuHwids              smbios property Manufacturer=Dell Inc.
02:10:06:0007 FuHwids              smbios property EnclosureKind=a
02:10:06:0007 FuHwids              smbios property Family=XPS
02:10:06:0007 FuHwids              smbios property ProductName=XPS 13 9310
02:10:06:0007 FuHwids              smbios property ProductSku=0A43
[... cut ...]
02:10:06:0036 FuEngine             writing motd target /home/crabx/mywork/fwupd/dist/cache/fwupd/motd.d/85-fwupd
WARNING: This package has not been validated, it may not work properly.
02:10:06:0036 FuMain               FuGpioDevice:
  DeviceId:             63223cc2d1b0b8d4c45ef8cedcb1f45315d3ff20
  Guid:                 ca5c154c-a46a-59d8-bf77-0ca40522cc85 ← GPIO\ID_INT34C5:00
  Plugin:               gpio
  Flags:                registered
  Created:              2022-03-16

02:10:06:0036 FuPlugin             destroy(gpio)
```

hughsie, 2022-03-16:
hey @CragW can you try https://github.com/fwupd/fwupd/pull/4389 again please; I added one more fix.

closed 2022-03-16 by hughsie