fwupd/fwupd #4394: modem-manager: device removed or did not come back

DylanVanAssche, 2022-03-16:
**Describe the bug**

It seems that in some rare cases ModemManager --> fastboot hand-over seems to fail as seen by LVFS reports.
I have a hard time to reproduce this personally on my setups, but in the past I have seem some similar things and the modem appears though with `fastboot devices`. Error messages:

- `failed to get device before update: failed to wait for detach replug: device 976c4a39e87f61e6940ea6a8d39c583cfa99615f did not come back`
- `Device has been removed`

AFAIK, retrying the upgrade 'fixes' it.
Most reports come from Debian but I saw a few as well from postmarketOS.
I hope somebody else has an idea here what may go wrong in this case.

**Steps to Reproduce**
1. Upgrade a modem with fwupd through ModemManager + fastboot
2. Upgrade fails because the modem seems to be lost when handing over, the modem appears though with `fastboot devices`

**Expected behavior**
Upgrade just works

**fwupd version information**
Please provide the version of the daemon and client.

```shell
pine64-pinephone:/home/dylan# fwupdmgr --version
runtime   org.freedesktop.fwupd         1.7.6
compile   org.freedesktop.gusb          0.3.10
runtime   org.kernel                    5.16.4
compile   com.hughsie.libjcat           0.1.10
compile   org.freedesktop.fwupd         1.7.6
runtime   org.freedesktop.gusb          0.3.10
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
Pine64 PinePhone (1.2)
│
└─QUECTEL Mobile Broadband Module:
      Device ID:          976c4a39e87f61e6940ea6a8d39c583cfa99615f
      Summary:            Quectel EG25-G modem
      Current version:    0.5.7
      Vendor:             QUALCOMM INCORPORATED (USB:0x2C7C, USB:0x18D1)
      Release Branch:     community
      GUIDs:              db379a33-254f-5140-b37e-d36ae7e5c039
                          d18f31f1-a3fa-55a2-b4ed-decfbc1e004d ← USB\VID_2C7C&PID_0125&REV_0318&NAME_EG25GGB
                          1a2996cb-f86e-5583-a464-e1b96e1c6ae9 ← USB\VID_2C7C&PID_0125&REV_0318
                          587bf468-6859-5522-93a7-6cce552a0aa3 ← USB\VID_2C7C&PID_0125
                          22ae45db-f68e-5c55-9c02-4557dca238ec ← USB\VID_2C7C
      Device Flags:       • Updatable
                          • System requires external power source

```

**Additional questions**

- Operating system and version: postmarketOS edge
- Have you tried rebooting? Yes
- Is this a regression? No idea

superm1, 2022-03-16:
Is the delay waiting for it to come back too short maybe?

DylanVanAssche, 2022-03-16:
Well it is set to 20 000ms so that is sufficient I think, but would this fail if the device disconnects shortly from the USB bus or something? Like re-enumerate twice or something? The bootloader is known to be a mess so...

DylanVanAssche, 2022-03-17:
@martijnbraam did the upgrade as well and got the `Device Removed` message after a successful upgrade: https://paste.sr.ht/~martijnbraam/4ac301e8a133f037280946c8284a65e9ca5734ae

We don't have verbose logs though (yet)

DylanVanAssche, 2022-03-17:
Another upgrade: https://paste.sr.ht/~martijnbraam/2d161ff7ab3a8b6239131684c5a59b866d93e2bf
However, no issue report is made now :shrug: 

The logs now indicate that ModemManager didn't got the message that the inhibition is over while fwupd seems to notify ModemManager:

```
19:51:12:0205 FuIdle               uninhibiting: update
```

But ModemManager is not notified:

```
19:51:34:0210 FuPluginMm           added port to inhibited modem: /dev/ttyUSB1 (ifnum 1)
19:51:34:0210 FuPluginMm           ignoring port /dev/ttyUSB1 (tty)
```

As long as you don't restart MM, fwupd still thinks it is on the old firmware: 
https://paste.sr.ht/~martijnbraam/9414f94a4d51423174335a9dbdb7b9e6f062c943

DylanVanAssche, 2022-03-27:
When digging into this, I tried to trace this further but on my side the uninhibiting actually works properly, but causes a segfault:

```
14:20:45:0986 FuIdle               uninhibiting: update
14:20:46:0024 FuMain               Called GetDevices()
14:20:46:0065 FuMain               Called GetReleases(976c4a39e87f61e6940ea6a8d39c583cfa99615f)
14:20:46:0079 FuEngine             requirement 1.7.6 ge 1.7.6 -> org.freedesktop.fwupd passed
14:20:46:0079 FuEngine             HWID provided adad873f-33d3-5477-bd5e-0870eca37444
14:20:46:0412 FuEngine             auto-removing backend device Fastboot MDM9607 [cf294bf55b333004beb7c41f952c1838c23e1f4a]
14:20:46:0414 FuHistory            modifying device Fastboot MDM9607 [cf294bf55b333004beb7c41f952c1838c23e1f4a]
14:20:46:0415 FuDeviceList         waiting 60000ms for Fastboot MDM9607 device removal
14:21:17:0114 FuPlugin             backend_device_added(modem_manager)
14:21:17:0123 FuUsbDevice          USB device 2:18 is /sys/devices/platform/soc/1c1b000.usb/usb2/2-1
14:21:17:0128 FuPluginMm           added port to inhibited modem: /dev/ttyUSB0 (ifnum 0)
14:21:17:0128 FuPluginMm           uninhibit modemmanager device with uid /sys/devices/platform/soc/1c1b000.usb/usb2/2-1

Thread 1 "fwupd" received signal SIGSEGV, Segmentation fault.
0x0000fffff634bc50 in fu_mm_device_udev_new () from /usr/lib/fwupd-plugins-5/libfu_plugin_modem_manager.so
(gdb) bt
#0  0x0000fffff634bc50 in fu_mm_device_udev_new () from /usr/lib/fwupd-plugins-5/libfu_plugin_modem_manager.so
#1  0x0000fffff6348f88 in ?? () from /usr/lib/fwupd-plugins-5/libfu_plugin_modem_manager.so
#2  0x0000fffff7429ce0 in ?? () from /usr/lib/libffi.so.8
#3  0x0000fffff7429258 in ?? () from /usr/lib/libffi.so.8
#4  0x0000fffff7bfc510 in g_cclosure_marshal_generic () from /usr/lib/libgobject-2.0.so.0
#5  0x0000fffff7bfbdcc in g_closure_invoke () from /usr/lib/libgobject-2.0.so.0
#6  0x0000fffff7c0ca70 in ?? () from /usr/lib/libgobject-2.0.so.0
#7  0x0000fffff7c10b2c in g_signal_emit_valist () from /usr/lib/libgobject-2.0.so.0
#8  0x0000fffff7c11234 in g_signal_emit () from /usr/lib/libgobject-2.0.so.0
#9  0x0000fffff7a5914c in ?? () from /usr/lib/libgudev-1.0.so.0
#10 0x0000fffff7b0bde8 in g_main_context_dispatch () from /usr/lib/libglib-2.0.so.0
#11 0x0000fffff7b0c048 in ?? () from /usr/lib/libglib-2.0.so.0
#12 0x0000fffff7b0c498 in g_main_loop_run () from /usr/lib/libglib-2.0.so.0
#13 0x0000aaaaaaab4d60 in ?? ()
#14 0x0000fffffffffc58 in ?? ()
Backtrace stopped: previous frame inner to this frame (corrupt stack?)
```

So either fwupd does not crash, but the modem stays inhibited, or fwupd crashes but the modem is unhibited.
Since fwupd is DBus activated, you won't notice the crash as fwupd will just restart immediately to serve fwupd clients such as GNOME Firmware Updater.

DylanVanAssche, 2022-03-28:
@hughsie I wish, but https://github.com/fwupd/fwupd/pull/4227 breaks fwupd when you have not libmbim 1.27.5 installed:

```
cannot load: failed to open plugin /usr/lib/fwupd-plugins-5/libfu_plugin_modem_manager.so: Error relocating /usr/lib/fwupd-plugins-5/libfu_plugin_modem_manager.so: mbim_message_qdu_quectel_reboot_set_new: symbol not found
```

hughsie, 2022-03-30:
@DylanVanAssche I don't think that PR caused those link issues; can you try rebuilding with the same development and library package versions installed?

closed 2022-04-12 by hughsie