fwupd/fwupd #4396: shim isn't installed to the EFI system partition error when updating T14/G2 AMD laptop on RHEL9

vbenes, 2022-03-16:
**Describe the bug**
After installing clean RHEL9 on T14G2 with AMD internals I am unable to update firmware from testing repo via software:
getting error:
Unable to update Thinkpad T14/P14s Gen 2 AMD system update, Secure boot is enabled but shim isn't installed to the EFI system partition. 

**Steps to Reproduce**
clean install from net iso, via anaconda, adding repos (I think c9s would behave the same), using the quite standard disk partitioning with LVM and when in GNOME, just start Software, enable testing repo where 1.15 firmware is available from and start update. Error seen.

**Expected behavior**
Reboot and firmware update performed.

**fwupd version information**

```shell
fwupdmgr --version
```
client version:	1.5.9
compile-time dependency versions
	gusb:	0.3.6

daemon version:	1.5.9

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices
```
[vbenes@localhost ~]$ fwupdmgr get-devices --show-all-devices
20XLS1WL00
│
├─AMD Ryzen 7 PRO 5850U with Radeon Graphics     :
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Vendor:             AMD
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f
│                         22f9ecf4-588d-5c0a-8326-6ebff3655c6d
│                         79759cdc-94db-5098-be7b-eb02521fbbec
│                         20b595b0-5892-5870-8e4c-688133ad6e34
│     Device Flags:       • Internal device
│   
├─Integrated Camera:
│     Device ID:          301046452a49d84af6356d23e43a684b8f10660f
│     Current version:    61.18
│     Vendor:             Chicony Electronics Co.,Ltd. (USB:0x04F2)
│     GUIDs:              d6d1a39a-3fbc-5ad1-ab40-336ec0776a38
│                         f8d483f9-9284-51b6-9363-7e6d59c40a97
│     Device Flags:       • Updatable
│   
├─Prometheus:
│ │   Device ID:          96a0be68fa5cf5d60d3d2d392797c6844ab64d33
│ │   Summary:            Fingerprint reader
│ │   Current version:    10.01.3478575
│ │   Vendor:             Synaptics (USB:0x06CB)
│ │   Install Duration:   2 seconds
│ │   GUIDs:              09e8bf16-3e69-50f5-bb66-c7a040248352
│ │                       8088f861-6318-5b1e-9ce4-fbddbedb09ac
│ │   Device Flags:       • Updatable
│ │                       • Supported on remote server
│ │                       • Cryptographic hash verification is available
│ │ 
│ └─Prometheus IOTA Config:
│       Device ID:        b869c71992c6719542d6dd2a02ef4e18617778ea
│       Current version:  0029
│       Minimum Version:  0029
│       Vendor:           Synaptics (USB:0x06CB)
│       GUIDs:            7c5a1e62-38fa-5859-9337-09dbac6377e4
│                         8a12a1dd-5e54-5cb8-9607-c3a4e3e9a7e4
│       Device Flags:     • Updatable
[... 195 lines cut ...]
├── loader
│   └── entries
│       ├── e248ce683379481dbacbcf35d20b0805-0-rescue.conf
│       └── e248ce683379481dbacbcf35d20b0805-5.14.0-70.1.1.el9.x86_64.conf
├── symvers-5.14.0-70.1.1.el9.x86_64.gz -> /lib/modules/5.14.0-70.1.1.el9.x86_64/symvers.gz
├── System.map-5.14.0-70.1.1.el9.x86_64
├── vmlinuz-0-rescue-e248ce683379481dbacbcf35d20b0805
└── vmlinuz-5.14.0-70.1.1.el9.x86_64

10 directories, 24 files

**Additional questions**

- Operating system and version:
RHEL9.0
- Have you tried rebooting?
yes, no luck
- Is this a regression?
no idea, a new machine so probably no
- Are you using an NVMe disk?
yes, Samsung Electronics Co Ltd (NVME:0x144D) 
- Is secure boot enabled?
yes, it is
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS?
Lenovo yes, Boot order lock turned off

vbenes, 2022-03-16:
ok, so as suggested in #3762 I've copied over shimx64.efi and shimx64-redhat.efi to the rhel directory and I got rid of the issue. But, after reboot, I am in Verification failed: (0x1A) Security Violation blue tab and the only thing I can do is press OK.

hmm hmm, we need something more to do, but definitely, something is missing here in the directory structure.

superm1, 2022-03-16:
> rhel directory

That's not expected!  @hughsie is the OS directory set wrong for the RHEL build of fwupd?  It should be "redhat" from that above output it seems.

superm1, 2022-03-16:
Also I don't know why there are two shims, if shimx64.efi doesn't work, what happens if you copy shimx64-redhat.efi over it?  Does that work?

hughsie, 2022-03-16:
> is the OS directory set wrong for the RHEL build of fwupd

Doh. It appears efi_os_dir is unset in RHEL 9. I'm guessing it's falling back to the "ID" in /etc/os-release? -- @vbenes can you attach the os-release here for a sanity check please.

superm1, 2022-03-16:
> > Also I don't know why there are two shims, if shimx64.efi doesn't work, what happens if you copy shimx64-redhat.efi over it? Does that work?
> 
> I've renamed shimx64-redhat.efi to shimx64.efi in rhel subdir but then I see normal grub after reboot, so of no help here :-(

Did you stage the update again presumably?

Can you give efibootmgr output specifically after it's staged with the copied over shim in rhel directory?

superm1, 2022-03-16:
After you overwrote the binary, re-run the update and when it prompts you to reboot run `efibootmgr -v` in another terminal and then share that and let it reboot.

vbenes, 2022-03-16:
[root@localhost boot]# efibootmgr -v
BootNext: 0001
BootCurrent: 0000
Timeout: 0 seconds
BootOrder: 0000,0017,0018,0019,001A,001B,001C,0001
Boot0000* Red Hat Enterprise Linux	HD(1,GPT,e44192e1-1557-48ef-a298-686b61c93c34,0x800,0x12c000)/File(\EFI\redhat\shimx64.efi)
Boot0001* Linux-Firmware-Updater	HD(1,GPT,e44192e1-1557-48ef-a298-686b61c93c34,0x800,0x12c000)/File(\EFI\rhel\shimx64.efi)\.f.w.u.p.d.x.6.4...e.f.i...
Boot0010  Setup	FvFile(721c8b66-426c-4e86-8e99-3457c46ab0b9)
Boot0011  Boot Menu	FvFile(126a762d-5758-4fca-8531-201a7f57f850)
Boot0012  Diagnostic Splash Screen	FvFile(a7d8d9a6-6ab0-4aeb-ad9d-163e59a7a380)
Boot0013  Lenovo Diagnostics	FvFile(3f7e615b-0d45-4f80-88dc-26b234958560)
Boot0014  ThinkShield secure wipe	FvFile(3593a0d5-bd52-43a0-808e-cbff5ece2477)
Boot0015  Startup Interrupt Menu	FvFile(f46ee6f4-4785-43a3-923d-7f786c3c8479)
Boot0016  Rescue and Recovery	FvFile(665d3f60-ad3e-4cad-8e26-db46eee9f1b5)
Boot0017* USB CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,86701296aa5a7848b66cd49dd3ba6a55)
Boot0018* USB FDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,6ff015a28830b543a8b8641009461e49)
Boot0019* NVMe0	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,001c199932d94c4eae9aa0b6e98eb8a400)
Boot001A* ATA HDD0	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f600)
Boot001B* USB HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,33e821aaaf33bc4789bd419f88c50803)
Boot001C* PXE BOOT	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,78a84aaf2b2afc4ea79cf5cc8f3d3803)
Boot001D  Regulatory Information	FvFile(478c92a0-2622-42b7-a65d-5894169e4d24)
Boot001E* Boot Next Boot Option	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,a7ca6d35b2c2684783721826a7404894)

vbenes, 2022-03-16:
and again, normal grub and system boot

superm1, 2022-03-16:
Okay so that matches my expectations with knowing the OS directory set as it is.

If that doesn't work now on reboot then there is something wrong with shim (second bug in rhel9).

hughsie, 2022-03-16:
@vbenes can you open a RH bz against RHEL 9 please and I'll do a c9s build with the s/rhel/redhat fix please.

hughsie, 2022-03-17:
@vbenes great, thanks. When i get qa-ack I'll do a test build.

superm1, 2022-03-17:
What about the second bug though? Even when shim is in the place fwupd expects it isn't working.

hughsie, 2022-03-18:
Yup, that looks like a shim issue.

closed 2022-04-27 by stale[bot]