fwupd/fwupd #4397: Upgrading Huawei Matebook 14S BIOS. `fwupdx64.efi` file missing

mil-ad, 2022-03-16:
I'm trying to upgrade the BIOS of my Huawei Matebook 14s. I'm following the steps in [this gist](https://gist.github.com/derjohn/2c19554415cea09830225314c2072807) which is for another (but hopefully similar) Huawei laptop and I'm running into exactly the issue reported by @tczank in [their comments](https://gist.github.com/derjohn/2c19554415cea09830225314c2072807?permalink_comment_id=4015573#gistcomment-4015573), i.e. rebooting doesn't actually install the update. 

I can convert the EXE firmware to a cap file using the script in the `contrib` directory and try to update using:

```
$ fwupdmgr --allow-older install ./firmware_bios_update.cab
```

`fwupdmgr` reports success but rebooting doesn't actually do the update. I have never done this before but I noticed Arch's wiki for systemd-boot (which is what I'm using) [talks about](https://wiki.archlinux.org/title/systemd-boot#EFI_Shells_or_other_EFI_applications) how to add a loader entry to fwupd's EFI file and I don't seem to have this file in my `/boot`!. I tried to manually copy it from `/usr/lib/fwupd/efi/fwupdx64.efi` into `/boot/EFI` and add 
the loader entry from the wiki. This made the upgrade option appear in boot menu but choosing it just shows "Reset System". 

Any idea what could be going wrong here?

**fwupd version information**
```
runtime   org.freedesktop.fwupd         1.7.6
runtime   com.dell.libsmbios            2.4
compile   org.freedesktop.gusb          0.3.10
runtime   org.kernel                    5.15.28-1-lts
compile   com.hughsie.libjcat           0.1.9
compile   org.freedesktop.fwupd         1.7.6
runtime   org.freedesktop.gusb          0.3.10
```

**fwupd device information**

```
HKD-WXX
│
├─11th Gen Intel Core™ i7-11370H @ 3.30GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x0000009a
│     Vendor:             Intel
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f
│                         30249f37-d140-5d3e-9319-186b1bd5cac3
│                         2347ad97-b545-5bae-b6b3-b70a9553cbba
│                         1df99df5-22f4-5505-98cc-22dcc53b132a
│     Device Flags:       • Internal device
│   
├─KPART512GBC2DVT:
│     Device ID:          aa27657c75074b0e4a4fe4c7da5d49d64e06e9c3
│     Summary:            NVM Express solid state drive
│     Current version:    FW000003
│     Vendor:             Intel Corporation (NVME:0xEFFF)
│     GUIDs:              7497261d-2cfb-50d9-8740-77640969772a
│                         46260061-1804-571a-9622-a5b3c59241cb
│                         539c71d1-4f2d-5419-9346-b0c7e5a32848
│                         95e9c4d3-6796-574f-a639-f561354e4457
│                         0c04df4d-9e5b-5eec-a3b4-95e4dc19cd59
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─System Firmware:
│     Device ID:          a45df35ac0e948ee180fe216a5f703f32dda163f
│     Summary:            UEFI ESRT device
│     Current version:    65541
│     Minimum Version:    65536
│     Vendor:             HUAWEI (DMI:HUAWEI)
│     Update State:       Success
│     GUIDs:              884756e0-d4a3-7ff7-1025-229246197d1a
│                         230c8b18-8d9b-53ec-838b-6cfc0383493a
│                         61e89d69-9cc6-50b8-a2c5-ff10919b5a0b
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Cryptographic hash verification is available
[... 119 lines cut ...]
├── initramfs-linux-fallback.img
├── initramfs-linux.img
├── initramfs-linux-lts-fallback.img
├── initramfs-linux-lts.img
├── intel-ucode.img
├── loader
│   ├── entries
│   │   ├── arch.conf
│   │   ├── arch-fallback.conf
│   │   ├── arch-lts.conf
│   │   └── arch-lts-fallback.conf
│   ├── loader.conf
│   └── random-seed
├── vmlinuz-linux
└── vmlinuz-linux-lts
```

**Additional questions**

- Operating system and version: **Arch 5.15.28-1-lts**
- Have you tried rebooting? **Yes**
- Is this a regression? **No**
- Are you using an NVMe disk? **Yes**
- Is secure boot enabled? **No**
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? **No**

superm1, 2022-03-16:
> I can convert the EXE firmware to a cap file using the script in the contrib directory and try to update using:

FYI, that is only guaranteed to work on Dell designs based on the design for their BIOS upgrade.  Dragons ahead for others.  But if we end up with success, please PR to a README some docs about it working on !Dell.

> Any idea what could be going wrong here?

Entries are supposed to be made automatically.  I don't see in your file listing a fwupdx64.efi anywhere.

Suggest to share the output of `sudo fwupdtool --verbose install foo.cab` so we can see what goes wrong with copying the fwupdx64.efi binary.

mil-ad, 2022-03-16:
> FYI, that is only guaranteed to work on Dell designs based on the design for their BIOS upgrade. Dragons ahead for others. But if we end up with success, please PR to a README some docs about it working on !Dell.

Thanks for the heads up! Is there a danger of bricking my laptop? 😰 This definitely worked with my previous Huawei laptop (Matebook 13) as documented [in this repo](https://github.com/nekr0z/linux-on-huawei-matebook-13-2019/blob/master/README.md) so maybe there's hope?

> Suggest to share the output of sudo fwupdtool --verbose install foo.cab so we can see what goes wrong with copying the fwupdx64.efi binary.

here you go:

```
19:40:24:0360 FuDebug              Verbose debugging enabled (on console 1)
19:40:24:0360 FuMain               locked /run/lock/fwupdtool
19:40:24:0367 FuMain               Failed to stop daemon: failed to find fwupd.service: GDBus.Error:org.freedesktop.systemd1.NoSuchUnit: Unit fwupd.service not loaded.
19:40:24:0367 FuConfig             trying to load config values from /etc/fwupd/daemon.conf
19:40:24:0367 FuCommon             reading /etc/fwupd/daemon.conf with 1930 bytes
19:40:24:0367 FuConfig             trying to load config values from /var/etc/fwupd/daemon.conf
19:40:24:0367 FuConfig             using autodetected max archive size 18.4 EB
19:40:24:0367 GLib-GIO             _g_io_module_get_default: Found default implementation local (GLocalVfs) for ‘gio-vfs’
19:40:24:0368 FuRemoteList         loading /usr/share/fwupd/metainfo
19:40:24:0368 XbSilo               attempting to load /var/cache/fwupd/metainfo.xmlb
19:40:24:0368 XbSilo               file: 322677be-01ca-dcf5-869d-ab36d5c3d879, current:79eb7ecf-0ce5-ef3a-8265-50e4ccd9ab73, cached: (null)
19:40:24:0375 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs.conf
19:40:24:0375 FuRemoteList         loading remote from /etc/fwupd/remotes.d/fwupd-tests.conf
19:40:24:0375 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor-directory.conf
19:40:24:0375 FuRemoteList         loading remote from /etc/fwupd/remotes.d/dell-esrt.conf
19:40:24:0375 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs-testing.conf
19:40:24:0376 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor.conf
19:40:24:0376 FuRemoteList         ignoring unfound remote fwupd
19:40:24:0376 FuRemoteList         ordering lvfs-testing=lvfs+1
19:40:24:0376 Jcat                 ignoring GPG-KEY-Linux-Foundation-Firmware as not PKCS-7 certificate
19:40:24:0376 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
19:40:24:0376 Jcat                 trying to load certificate from /etc/pki/fwupd/LVFS-CA.pem
19:40:24:0376 Jcat                 reading /etc/pki/fwupd/LVFS-CA.pem with 1679 bytes
19:40:24:0376 Jcat                 loaded 1 certificates
19:40:24:0376 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
19:40:24:0376 Jcat                 trying to load certificate from /etc/pki/fwupd-metadata/LVFS-CA.pem
19:40:24:0376 Jcat                 reading /etc/pki/fwupd-metadata/LVFS-CA.pem with 1679 bytes
19:40:24:0376 Jcat                 loaded 1 certificates
19:40:24:0376 Jcat                 ignoring GPG-KEY-Linux-Foundation-Metadata as not PKCS-7 certificate
19:40:24:0376 Jcat                 reading /var/lib/fwupd/pki/secret.key with 2459 bytes
19:40:24:0376 Jcat                 reading /var/lib/fwupd/pki/client.pem with 1383 bytes
19:40:24:0376 FuEngine             client certificate now exists: nothing to do
19:40:24:0376 FuHistory            trying to open database '/var/lib/fwupd/pending.db'
19:40:24:0377 FuHistory            got schema version of 7
19:40:24:0377 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d
19:40:24:0381 FuQuirks             loading quirks from /var/lib/fwupd/quirks.d
19:40:24:0381 XbSilo               attempting to load /var/cache/fwupd/quirks.xmlb
19:40:24:0381 XbSilo               file: c808e562-68a8-0a7c-8d48-bf93a8d07b90, current:065d7625-89ba-973a-6a93-2ca97cc8c8b5, cached: (null)
19:40:24:0416 FuQuirks             invalid key names: CcgxFlashRowSize,CcgxFlashSize,CcgxImageKind,CxaudioChipIdBase,CxaudioPatch1ValidAddr,CxaudioPatch2ValidAddr,DellDockBlobBuildOffset,DellDockBlobMajorOffset,DellDockBlobMinorOffset,DellDockBlobVersionOffset,DellDockBoardMin,DellDockHubVersionLowest,DellDockInstallDurationI2C,DellDockUnlockTarget,DellDockVersionLowest,DfuAltName,DfuForceTimeout,DfuForceVersion,ElantpI2cTargetAddress,ElantpIapPassword,ElantpIcPageCount,FastbootBlockSize,FastbootOperationDelay,GenesysScalerGpioEnableRegister,GenesysScalerGpioOutputRegister,GenesysScalerGpioValue,GenesysUsbhubReadRequest,GenesysUsbhubSwitchRequest,GenesysUsbhubWriteRequest,IntelSpiBarProxy,IntelSpiBiosCntl,IntelSpiKind,Issue,JabraMagic,LogitechHidppModelId,ModemManagerBranchAtCommand,NordicHidBootloader,ParadeLspconAuxDeviceName,PciBcrAddr,RealtekMstDpAuxName,RealtekMstDrmCardKernelName,Rts54I2cSpeed,Rts54RegisterAddrLen,Rts54TargetAddr,SteelSeriesDeviceKind,SuperioAutoloadAction,SuperioControlPort,SuperioDataPort,SuperioGType,SuperioId,SuperioPort,SuperioTimeout,SynapticsMstDeviceKind,VliDeviceKind,WacomI2cFlashBaseAddr,WacomI2cFlashBlockSize,WacomI2cFlashSize
19:40:24:0416 FuSmbios             trying to read /sys/class/dmi/id
19:40:24:0416 FuHwids              loading HwId overrides from /etc/fwupd/daemon.conf
[... cut ...]
19:40:27:0570 FuCabinet            processing file: firmware.metainfo.xml
19:40:27:0570 FuCabinet            processing release: 1.09
19:40:27:0570 FuEngine             Emitting PropertyChanged('Status'='idle')
Idle…: 0%
19:40:27:0570 FuMain               first pass requirement on 4bde70ba4e39b28f9eab1628f9dd6e6244c03027:org.Huawei.guid230c8b18 failed: No supported devices found
19:40:27:0570 FuMain               first pass requirement on aa27657c75074b0e4a4fe4c7da5d49d64e06e9c3:org.Huawei.guid230c8b18 failed: No supported devices found
```

mil-ad, 2022-04-18:
I hope @superm1 can find some time to go through the log above 🙏🏼

superm1, 2022-04-18:
Thanks for reminding me.  So looking at this, it's actually setting up a UefiCodDevice - meaning this is not being delivered by fwupdx64.efi, it's delivered via an capsule on disk capsule.  Some code landed in main just 6 days ago that improves this support: https://github.com/fwupd/fwupd/commit/bef1d742b3af21690f7120cedfe7177ca164f973

Can you please upgrade to a checkout from main, and show the full run end to end from

`sudo fwupdtool install foo.cab --verbose` up until the reboot?  If it works then great, if not then we can see if it's still a problem in main.

mil-ad, 2022-04-20:
Thanks! I had a hard time compiling it and the `fwupd-git` AUR package on arch seems to be broken at the moment. I'll test this as soon as a release lands on arch's official repo.

mil-ad, 2022-04-29:
Ok I'm not getting the old error anymore but I still see the old version in the bios. Here's the log (from the second time I ran the command after reboot. Forgot to copy the first time around)

```
❯ sudo fwupdtool --allow-older install firmware_bios_update.cab --verbose                
21:52:10:0587 FuDebug              Verbose debugging enabled (on console 1)
21:52:10:0588 FuMain               locked /run/lock/fwupdtool
21:52:10:0593 FuMain               Failed to stop daemon: failed to find fwupd.service: GDBus.Error:org.freedesktop.systemd1.NoSuchUnit: Unit fwupd.service not loaded.
21:52:10:0594 FuConfig             trying to load config values from /etc/fwupd/daemon.conf
21:52:10:0594 FuCommon             reading /etc/fwupd/daemon.conf with 2080 bytes
21:52:10:0594 FuConfig             trying to load config values from /var/etc/fwupd/daemon.conf
21:52:10:0594 FuConfig             using autodetected max archive size 18.4 EB
21:52:10:0596 GLib-GIO             _g_io_module_get_default: Found default implementation local (GLocalVfs) for ‘gio-vfs’
21:52:10:0597 FuRemoteList         loading /usr/share/fwupd/metainfo
21:52:10:0597 XbSilo               attempting to load /var/cache/fwupd/metainfo.xmlb
21:52:10:0597 XbSilo               file: aafa54b5-e0b9-6f82-3f46-85fb049fe2a9, current:aafa54b5-e0b9-6f82-3f46-85fb049fe2a9, cached: (null)
21:52:10:0597 XbSilo               loading silo with file contents
21:52:10:0597 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs.conf
21:52:10:0598 FuRemoteList         loading remote from /etc/fwupd/remotes.d/fwupd-tests.conf
21:52:10:0598 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor-directory.conf
21:52:10:0598 FuRemoteList         loading remote from /etc/fwupd/remotes.d/dell-esrt.conf
21:52:10:0599 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs-testing.conf
21:52:10:0599 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor.conf
21:52:10:0599 FuRemoteList         ignoring unfound remote fwupd
21:52:10:0599 FuRemoteList         ordering lvfs-testing=lvfs+1
21:52:10:0600 Jcat                 ignoring GPG-KEY-Linux-Foundation-Firmware as not PKCS-7 certificate
21:52:10:0600 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
21:52:10:0600 Jcat                 reading /etc/pki/fwupd/LVFS-CA.pem with 1679 bytes
21:52:10:0601 Jcat                 loaded 1 certificates
21:52:10:0601 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
21:52:10:0601 Jcat                 reading /etc/pki/fwupd-metadata/LVFS-CA.pem with 1679 bytes
21:52:10:0601 Jcat                 loaded 1 certificates
21:52:10:0601 Jcat                 ignoring GPG-KEY-Linux-Foundation-Metadata as not PKCS-7 certificate
21:52:10:0601 Jcat                 reading /var/lib/fwupd/pki/secret.key with 2459 bytes
21:52:10:0601 Jcat                 reading /var/lib/fwupd/pki/client.pem with 1383 bytes
21:52:10:0601 FuEngine             client certificate now exists: nothing to do
21:52:10:0601 FuHistory            trying to open database '/var/lib/fwupd/pending.db'
21:52:10:0602 FuHistory            got schema version of 7
21:52:10:0602 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d
21:52:10:0614 FuQuirks             loading quirks from /var/lib/fwupd/quirks.d
21:52:10:0614 XbSilo               attempting to load /var/cache/fwupd/quirks.xmlb
21:52:10:0614 XbSilo               file: 850eb8ef-e96e-60d0-810e-bf30974e3b99, current:850eb8ef-e96e-60d0-810e-bf30974e3b99, cached: (null)
21:52:10:0614 XbSilo               loading silo with file contents
21:52:10:0615 FuSmbios             trying to read /sys/class/dmi/id
21:52:10:0615 FuHwids              loading HwId overrides from /etc/fwupd/daemon.conf
21:52:10:0615 FuHwids              not loading HwId overrides from /var/lib/fwupd/daemon.conf
21:52:10:0615 FuHwids              smbios property Manufacturer=HUAWEI
21:52:10:0615 FuHwids              smbios property EnclosureKind=a
21:52:10:0615 FuHwids              smbios property Family=MateBook
21:52:10:0615 FuHwids              smbios property ProductName=HKD-WXX
21:52:10:0615 FuHwids              smbios property ProductSku=C100
21:52:10:0615 FuHwids              smbios property BiosVendor=HUAWEI
21:52:10:0615 FuHwids              smbios property BiosVersion=1.05
21:52:10:0615 FuHwids              smbios property BiosMajorRelease=01
21:52:10:0615 FuHwids              smbios property BiosMinorRelease=05
21:52:10:0615 FuHwids              smbios property FirmwareMajorRelease=01
21:52:10:0615 FuHwids              smbios property FirmwareMinorRelease=05
21:52:10:0615 FuHwids              smbios property BaseboardManufacturer=HUAWEI
21:52:10:0615 FuHwids              smbios property BaseboardProduct=HKD-WXX-PCB
21:52:10:0619 FuEngine             ignoring: Error opening directory “/var/lib/fwupd/local.d”: No such file or directory
21:52:10:0619 FuEngine             ignoring: Error opening directory “/usr/share/fwupd/local.d”: No such file or directory
21:52:10:0619 XbSilo               attempting to load /var/cache/fwupd/metadata.xmlb
21:52:10:0619 XbSilo               file: 5a81484b-874a-81b2-04ec-3c3352dbdf5f, current:5a81484b-874a-81b2-04ec-3c3352dbdf5f, cached: (null)
21:52:10:0619 XbSilo               loading silo with file contents
21:52:10:0621 FuEngine             1313 components now in silo
[... cut ...]
21:52:13:0461 FuPluginLogind       closed logind fd 17
21:52:13:0461 FuDeviceList         no replug or re-enumerate required
21:52:13:0461 FuEngine             Updating System Firmware took 0.069420 seconds
21:52:13:0461 FuHistory            modifying device System Firmware [a45df35ac0e948ee180fe216a5f703f32dda163f]
21:52:13:0463 FuPlugin             composite_cleanup(dell_dock)
21:52:13:0463 FuPlugin             composite_cleanup(thunderbolt)
21:52:13:0464 FuIdle               uninhibiting: update
```

mil-ad, 2022-04-29:
Another random question. Which GUID should I use when generating the .cab file. I see two:

```
├─System Firmware:
│     Device ID:          a45df35ac0e948ee180fe216a5f703f32dda163f
│     Summary:            UEFI ESRT device
│     Current version:    65541
│     Minimum Version:    65536
│     Vendor:             HUAWEI (DMI:HUAWEI)
│     Update State:       Needs reboot
│     GUIDs:              884756e0-d4a3-7ff7-1025-229246197d1a
│                         230c8b18-8d9b-53ec-838b-6cfc0383493a ← main-system-firmware
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Cryptographic hash verification is available
│                         • Device is usable for the duration of the update
```

superm1, 2022-04-29:
Either will work for your purpose but you want the one from the ESRT ideally ( not from a string ).

superm1, 2022-07-20:
This is fixed in upcoming 1.8.3 release.
https://github.com/fwupd/fwupd/commit/77880751cddab056ebc653b1e991ff88f60603eb

closed 2022-07-20 by superm1