fwupd/fwupd #4400: "Full disk Encryption Detected" warning message is confusing

milianw, 2022-03-17:
**Describe the bug**

I'm using a ThinkPad P1 G4i with full disk encryption enabled via the bios, i.e. I am _not_ using Bitlocker. I got this warning then:

```
╔══════════════════════════════════════════════════════════════════════════════╗
║ Full Disk Encryption Detected                                                ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ Some of the platform secrets may be invalidated when updating this           ║
║ firmware. Please ensure you have the volume recovery key before continuing.  ║
╚══════════════════════════════════════════════════════════════════════════════╝
```

I was very confused by this, as the bios disk encryption doesn't allow me to see any "volume recovery key" anywhere. I could also not find any help on what this could mean online.

Looking at https://github.com/fwupd/fwupd/commit/d3706e0e0b0fc210796da839b84ac391f7a251f8 it seems that this warning is bitlocker specific? As such, could you maybe reword the message slightly indicating that it could be ignored if BitLocker is wrongly detected.

Additionally, please also reference some resource that explains how to obtain the "volume recovery key".

Finally note that due to this scary message, I refrained from running the update from linux. Instead, I rebooted into Windows and used the official Lenovo utility to update the bios. No warning like the above was shown there, which made me more confident in applying the update.

**Steps to Reproduce**

Enable full disk encryption from bios on ThinkPad and then install bios update.

**Expected behavior**

The warning should be more helpful and tell me that I can ignore it if BitLocker was wrongly detected. BitLocker users would also like to know how to even obtain the "volume recovery key" I believe.

**fwupd version information**

```
fwupdmgr --version
runtime   org.freedesktop.fwupd         1.7.6
runtime   com.dell.libsmbios            2.4
compile   org.freedesktop.gusb          0.3.10
runtime   org.kernel                    5.16.14-arch1-1
compile   com.hughsie.libjcat           0.1.9
compile   org.freedesktop.fwupd         1.7.6
runtime   org.freedesktop.gusb          0.3.10
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): pacman

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices
1000979;2110176
│
├─BCM920702 Bluetooth 4.0:
│     Device ID:          1f914e619b232dc4da68ec3389b98b20163003a2
│     Current version:    1.12
│     Vendor:             Broadcom Corp (USB:0x0A5C)
│     GUIDs:              6faa2765-8bc2-51d3-bf6b-ffd6312c24d2
│                         a1e76ca3-d35b-53b8-a17c-454b700274ea
│     Device Flags:       • Updatable
│   
├─GIGABYTE GP-ASM2NE6200TTTD:
│     Device ID:          04e17fcf7d3de91da49a163ffe4907855c3648be
│     Summary:            NVM Express solid state drive
│     Current version:    EGFM13.0
│     Vendor:             Phison Electronics Corporation (NVME:0x1987)
│     GUIDs:              95711de3-8cc3-5db4-b813-6f4535e75c66
│                         12c32247-6129-573a-8e42-f8bf95e33f43
│                         e0beed4b-7f9d-58e2-ae74-0dd4cf071aad
│                         3cb8c6f1-e686-5c7b-9ddf-cd66cd34350f
│                         e9e90b91-7095-5d6a-9241-7a6279609a2a
│     Device Flags:       • Updatable
│                         • System requires external power source
│                         • Needs shutdown after installation
│   
├─Multiple Reader:
│     Device ID:          268b8794e1a87266d2903c7cafe79aa2b193b6e4
│     Summary:            SCSI device
│     Current version:    1.11
│     Vendor:             Generic- (SCSI:Generic)
│     GUIDs:              195005c5-ab02-527d-aa2b-9aee4672a90b
│                         97ac5924-dda2-54a2-ade3-252b7c03087b
│   
├─Navi 10 [Radeon RX 5600 OEM/5600 XT / 5700/5700 XT] (Radeon RX 5700 XT Gaming OC):
│     Device ID:          21ba7fbcdfb80c824895aad926418475726bcc79
│     Current version:    c1
│     Vendor:             Advanced Micro Devices, Inc. [AMD/ATI] (PCI:0x1002)
│     GUIDs:              2a8138cd-16fb-59a8-9c96-1a99fa377c41
│                         52bf2704-faeb-5dd2-9bc6-c140b2d2d07f
│                         afa52ef7-5494-518d-980c-54328b60c0f3
│                         dd0fc8c4-f037-5a78-bdea-c85eb0fe68fa
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
├─Ryzen 9 3900X 12-Core Processor:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Vendor:             AMD
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f
│                         997104da-ec11-5e04-9d94-ac5501fbf609
│                         b54f3c89-8a41-5b9f-9a5c-0367c19a9451
│                         d951a23c-a920-50ac-ad89-0660aa0be95c
│     Device Flags:       • Internal device
│   
├─SSD 970 EVO Plus 1TB:
│     Device ID:          71b677ca0f1bc2c5b804fa1d59e52064ce589293
│     Summary:            NVM Express solid state drive
│     Current version:    2B2QEXM7
│     Vendor:             Samsung (NVME:0x144D)
│     GUIDs:              0b4d773a-7ac3-58c1-a541-e22ef1cdfe02
│                         c9d531ea-ee7d-5562-8def-c64d0d144813
│                         6e54c992-d302-59ab-b454-2d26ddd63e6d
│                         47335265-a509-51f7-841e-1c94911af66b
│                         f05ecac9-5cc2-5f77-b23b-a08ad8879ada
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─System Firmware:
│ │   Device ID:          a45df35ac0e948ee180fe216a5f703f32dda163f
│ │   Summary:            UEFI ESRT device
│ │   Current version:    5127
│ │   Minimum Version:    5127
│ │   Vendor:             Wortmann_AG (DMI:American Megatrends Inc.)
│ │   Update State:       Success
│ │   GUIDs:              ae204fc0-355b-54a9-948f-79e9a9e22a16
│ │                       230c8b18-8d9b-53ec-838b-6cfc0383493a
│ │                       bbbf4a35-1b72-591d-bd5f-06d487f5ca49
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Needs a reboot after installation
│ │                       • Cryptographic hash verification is available
│ │                       • Device is usable for the duration of the update
│ │                       • Full disk encryption secrets may be invalidated when updating
│ │ 
│ └─UEFI dbx:
│       Device ID:        362301da643102b9f38477387e2193e57abaa590
│       Summary:          UEFI revocation database
│       Current version:  77
│       Minimum Version:  77
│       Vendor:           UEFI:Linux Foundation
│       Install Duration: 1 second
│       GUIDs:            fda6234b-adcb-5105-8515-9af647d29775
│                         f8ff0d50-c757-5dc3-951a-39d86e16f419
│                         c6682ade-b5ec-57c4-b687-676351208742
│                         f8ba2887-9411-5c36-9cee-88995bb39731
│                         7d5759e5-9aa0-5f0c-abd6-7439bb11b9f6
│                         0c7691e1-b6f2-5d71-bc9c-aabee364c916
│       Device Flags:     • Internal device
│                         • Updatable
│                         • Needs a reboot after installation
│                         • Only version upgrades are allowed
│                         • Signed Payload
│     
├─TPM:
│     Device ID:          c6a80ac3a22083423992a3cb15018989f37834d6
│     Summary:            TPM 2.0 Device
│     Current version:    3.37.0.5
│     Vendor:             AMD (TPM:AMD)
│     GUIDs:              ff71992e-52f7-5eea-94ef-883e56e034c6
│                         9305de1c-1e12-5665-81c4-37f8e51219b8
│                         78a291ae-b499-5b0f-8f1d-74e1fefd0b1c
│                         65a3fced-b423-563f-8098-bf5c329fc063
│                         5e704f0d-83cb-5364-8384-f46d725a23b8
│     Device Flags:       • Internal device
│   
└─iHAS124 F:
      Device ID:          82539ee6f74ab53606091313311e17e96b003c09
      Summary:            SCSI device
      Current version:    CL9N
      Vendor:             ATAPI (SCSI:ATAPI)
      GUIDs:              704829d5-9643-5d9a-8bd4-61e85b8598fa
                          c6752a35-8173-5c9a-9232-a8458147d1bc
```

**Additional questions**

- Operating system and version:  latest ArchLinux and Windows 11
- Have you tried rebooting? yes
- Is this a regression? don't know

hughsie, 2022-03-17:
Can we have the output from `udisksctl dump` please?

milianw, 2022-03-17:
Sure thing:

```
/org/freedesktop/UDisks2/Manager:
  org.freedesktop.UDisks2.Manager:
    DefaultEncryptionType:      luks1
    SupportedEncryptionTypes:   luks1
                                luks2
    SupportedFilesystems:       ext2
                                ext3
                                ext4
                                vfat
                                ntfs
                                exfat
                                xfs
                                reiserfs
                                nilfs2
                                btrfs
                                minix
                                udf
                                f2fs
                                swap
    Version:                    2.9.4

/org/freedesktop/UDisks2/block_devices/dm_2d0:
  org.freedesktop.UDisks2.Block:
    Configuration:              [('fstab', {'fsname': <b'UUID=c4134a74-8c83-448a-8cbb-11723509303e'>, 'dir': <b'/'>, 'type': <b'ext4'>, 'opts': <b'rw,relatime,noatime,discard'>, 'freq': <0>, 'passno': <1>})]
    CryptoBackingDevice:        '/org/freedesktop/UDisks2/block_devices/nvme0n1p5'
    Device:                     /dev/dm-0
    DeviceNumber:               65024
    Drive:                      '/'
    HintAuto:                   false
    HintIconName:               
    HintIgnore:                 false
    HintName:                   
    HintPartitionable:          false
    HintSymbolicIconName:       
    HintSystem:                 true
    Id:                         by-id-dm-name-cryptroot
    IdLabel:                    
    IdType:                     ext4
    IdUUID:                     c4134a74-8c83-448a-8cbb-11723509303e
    IdUsage:                    filesystem
    IdVersion:                  1.0
    MDRaid:                     '/'
    MDRaidMember:               '/'
    PreferredDevice:            /dev/mapper/cryptroot
    ReadOnly:                   false
    Size:                       736561725440
    Symlinks:                   /dev/disk/by-id/dm-name-cryptroot
                                /dev/disk/by-id/dm-uuid-CRYPT-LUKS2-1ed084c295de4955a4c85e6e8650c785-cryptroot
                                /dev/disk/by-uuid/c4134a74-8c83-448a-8cbb-11723509303e
                                /dev/mapper/cryptroot
    UserspaceMountOptions:      
  org.freedesktop.UDisks2.Filesystem:
    MountPoints:        /
    Size:               736561725440

/org/freedesktop/UDisks2/block_devices/dm_2d1:
  org.freedesktop.UDisks2.Block:
    Configuration:              [('fstab', {'fsname': <b'UUID=0e013dca-45ba-4c63-be6e-4a7c69195217'>, 'dir': <b'/mnt/data1/'>, 'type': <b'ext4'>, 'opts': <b'rw,relatime,noatime,discard'>, 'freq': <0>, 'passno': <1>})]
    CryptoBackingDevice:        '/org/freedesktop/UDisks2/block_devices/nvme1n1'
    Device:                     /dev/dm-1
    DeviceNumber:               65025
    Drive:                      '/'
    HintAuto:                   false
    HintIconName:               
    HintIgnore:                 false
    HintName:                   
    HintPartitionable:          false
[... 529 lines cut ...]
    Media:                      
    MediaAvailable:             true
    MediaChangeDetected:        true
    MediaCompatibility:         
    MediaRemovable:             false
    Model:                      Samsung SSD 970 EVO Plus 1TB
    Optical:                    false
    OpticalBlank:               false
    OpticalNumAudioTracks:      0
    OpticalNumDataTracks:       0
    OpticalNumSessions:         0
    OpticalNumTracks:           0
    Removable:                  false
    Revision:                   2B2QEXM7
    RotationRate:               0
    Seat:                       seat0
    Serial:                     S4EWNJ0N224037T
    SiblingId:                  
    Size:                       1000204886016
    SortKey:                    00coldplug/00fixed/nvme0n1
    TimeDetected:               1647501668630580
    TimeMediaDetected:          1647501668630580
    Vendor:                     
    WWN:                        eui.0025385201419a76
```

milianw, 2022-03-17:
Sorry, I just realize I'm writing this from the wrong machine (desktop, instead of laptop). I'll update the requested output now

milianw, 2022-03-17:
OK, I updated the first comment with the output from my laptop. And here's the output of `udiskctl dump` from there:

```
/org/freedesktop/UDisks2/Manager:
  org.freedesktop.UDisks2.Manager:
    DefaultEncryptionType:      luks1
    SupportedEncryptionTypes:   luks1
                                luks2
    SupportedFilesystems:       ext2
                                ext3
                                ext4
                                vfat
                                ntfs
                                exfat
                                xfs
                                reiserfs
                                nilfs2
                                btrfs
                                minix
                                udf
                                f2fs
                                swap
    Version:                    2.9.4

/org/freedesktop/UDisks2/block_devices/nvme0n1:
  org.freedesktop.UDisks2.Block:
    Configuration:              []
    CryptoBackingDevice:        '/'
    Device:                     /dev/nvme0n1
    DeviceNumber:               66304
    Drive:                      '/org/freedesktop/UDisks2/drives/Samsung_SSD_970_EVO_Plus_2TB_S4J4NX0R810423M'
    HintAuto:                   false
    HintIconName:               
    HintIgnore:                 false
    HintName:                   
    HintPartitionable:          true
    HintSymbolicIconName:       
    HintSystem:                 true
    Id:                         by-id-nvme-Samsung_SSD_970_EVO_Plus_2TB_S4J4NX0R810423M
    IdLabel:                    
    IdType:                     
    IdUUID:                     
    IdUsage:                    
    IdVersion:                  
    MDRaid:                     '/'
    MDRaidMember:               '/'
    PreferredDevice:            /dev/nvme0n1
    ReadOnly:                   false
    Size:                       2000398934016
    Symlinks:                   /dev/disk/by-id/nvme-Samsung_SSD_970_EVO_Plus_2TB_S4J4NX0R810423M
                                /dev/disk/by-id/nvme-eui.0025385811b028f5
                                /dev/disk/by-path/pci-0000:04:00.0-nvme-1
    UserspaceMountOptions:      
  org.freedesktop.UDisks2.PartitionTable:
    Partitions:         /org/freedesktop/UDisks2/block_devices/nvme0n1p2
                        /org/freedesktop/UDisks2/block_devices/nvme0n1p3
                        /org/freedesktop/UDisks2/block_devices/nvme0n1p4
                        /org/freedesktop/UDisks2/block_devices/nvme0n1p5
                        /org/freedesktop/UDisks2/block_devices/nvme0n1p1
    Type:               gpt

/org/freedesktop/UDisks2/block_devices/nvme0n1p1:
  org.freedesktop.UDisks2.Block:
    Configuration:              []
    CryptoBackingDevice:        '/'
    Device:                     /dev/nvme0n1p1
    DeviceNumber:               66305
    Drive:                      '/org/freedesktop/UDisks2/drives/Samsung_SSD_970_EVO_Plus_2TB_S4J4NX0R810423M'
    HintAuto:                   false
    HintIconName:               
[... 229 lines cut ...]
    Media:                      
    MediaAvailable:             true
    MediaChangeDetected:        true
    MediaCompatibility:         
    MediaRemovable:             false
    Model:                      Samsung SSD 970 EVO Plus 2TB
    Optical:                    false
    OpticalBlank:               false
    OpticalNumAudioTracks:      0
    OpticalNumDataTracks:       0
    OpticalNumSessions:         0
    OpticalNumTracks:           0
    Removable:                  false
    Revision:                   2B2QEXM7
    RotationRate:               0
    Seat:                       seat0
    Serial:                     S4J4NX0R810423M
    SiblingId:                  
    Size:                       2000398934016
    SortKey:                    00coldplug/00fixed/nvme0n1
    TimeDetected:               1647519798304514
    TimeMediaDetected:          1647519798304514
    Vendor:                     
    WWN:                        eui.0025385811b028f5
```

milianw, 2022-03-17:
Looking at that output, I clearly see that it detects BitLocker. But I double checked on Windows and BitLocker is disabled.

I did not install Windows myself on this machine, it's the default installation that got shipped with the machine after buying it. I only updated it to Windows 11. I am unsure if in that process bit locker was ever enabled or not, but I did not ever use it myself from what I recall. And right now, it is clearly disabled, as I'm relying on the bios full disk encryption feature instead.

milianw, 2022-03-17:
The BitLocker Drive Encryption settings on Windows says:

> Windows (C:) BitLocker waiting for activation

And there is a button to "Turn BitLocker on".

hughsie, 2022-03-17:
All we get from UDisks is:

    IdType:                     BitLocker

We don't know if BitLocker is turned on or off. If there's a way to read that back from the partition then we could add it, but I think this is probably private to Windows.

milianw, 2022-03-17:
Alright, but could you please add this information to the fwupd warning? I.e. not knowing that this warning even talks about BitLocker in the first place, I tried to find a "volume backup key" for the bios full disk encryption, but couldn't find anything...

hughsie, 2022-03-17:
The word BitLocker is trademarked by Microsoft and I'm slightly hesitant to use it in this context. In the future we'll also be detecting other FDE like Clevis. Maybe we could write a wiki page about it and include a link from the warning?

milianw, 2022-03-17:
Yes, that sounds like a great solution to me.

hughsie, 2022-03-17:
@milianw is that something you could have a go at? I'm terrible at technical prose like this.

milianw, 2022-03-18:
I have given it a shot over at https://github.com/fwupd/fwupd/wiki/Full-Disk-Encryption-Detected - improvements welcome :)

closed 2022-03-18 by hughsie

hughsie, 2025-06-27:
I think it's okay to ignore. It's really only valid if you have full disk encryption via BitLocker or Snap.

GiorgioAresu, 2025-06-29:
I ignored the warning on my framework 13 with Fedora and LUKS, worked without any issues. Thanks

artkpv, 2025-06-29:
> I think it's okay to ignore. It's really only valid if you have full disk encryption via BitLocker or Snap.

Thank you for the answer. I've updated successfully.

ddmgh, 2025-07-21:
Came across this from a search and Just wanted to chime in to say I too got this confusing message and updated with no issue.

FWIW:

```
$ udisksctl dump | grep IdType
    IdType:                     ext4
    IdType:                     
    IdType:                     
    IdType:                     
    IdType:                     
    IdType:                     
    IdType:                     
    IdType:                     
    IdType:                     
    IdType:                     
    IdType:                     vfat
    IdType:                     crypto_LUKS
```