fwupd/fwupd #4402: FWUPD for retimer FW update should support for DA mode

krankun, 2022-03-17:
**Describe the bug**
The TBT3 device seems to be in DP ALT when fwupd probes retimer at boot time, 
This issue should cover all the DA cases , USB , DP ALT , USB4 etc

**Steps to Reproduce**
connect a TBT3 dock 
login the DUT of chromebook

**Expected behavior**
TBT3 device should enumerate fine,

**fwupd version information**

```shell
fwupdmgr --version
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices
```

**Additional questions**

- Operating system and version:
- Have you tried rebooting?
- Is this a regression?
NO

krankun, 2022-03-17:
after login the mode doen't transition to TBT=1
instead 
ectool usbpdmuxinfo
Port 0: USB=1 DP=1 POLARITY=INVERTEDHPD_IRQ=0 HPD_LVL=0 SAFE=0 TBT=0 USB4=0 
Port 1: USB=0 DP=0 POLARITY=NORMALHPD_IRQ=0 HPD_LVL=0 SAFE=0 TBT=0 USB4=0 

[tbt_enumarate_issue.txt](https://github.com/fwupd/fwupd/files/8290678/tbt_enumarate_issue.txt)

krankun, 2022-03-17:
cc: @djcampello @superm1 @hughsie @westeri @jzhao80 @ feng li and raj will be added too

djcampello, 2022-03-17:
How is this actionable in user space? If we can use the same offline/rescan sysfs interfaces to make the retimers enumerate then it should be already working. If that is not the case, and we don't have a way to make the retimers enumerate from fwupd then this needs to be handled at another level in the system.

krankun, 2022-03-17:
@djcampello this is issue is created to get the understanding of the underlying issue and see how this can be fixed, 
This issue got exposed after fwupd probes retimer in DA case
we need to agree on the component(CB/EC/kernel/fwupd) where the fix can land not necessarily fwupd.
since most of the community is available UPSTREAM I created this issue here , we are also going to disable retimer probing from fwupd until this is fixed even thou I understand this issue is not related to fwupd.

I already left a note to Li and Raj to chime in here too