fwupd/fwupd #4403: PinePhone Pro not recognized when booting through UEFI

MartijnBraam, 2022-03-17:
**Describe the bug**
When booting the PinePhone Pro using UEFI instead of just loading it with an U-Boot image fwupd  no longer knows about
updates for the Quectel EG25-g modem. The device is now shown as "Unknown Product"

The full log is available at https://paste.sr.ht/~martijnbraam/3b99d51bc3df121a321fb02beac1626b2bcbc9fd

**Steps to Reproduce**
Install the UEFI branch of postmarketOS to the PinePhone Pro, run fwupd get-devices or fwupd upgrade

**Expected behavior**
I expect the community firmware to be available and it lets me switch branches

**fwupd version information**

```shell
runtime   org.freedesktop.fwupd         1.7.6
compile   org.freedesktop.gusb          0.3.10
runtime   org.kernel                    5.17.0-rc8
compile   com.hughsie.libjcat           0.1.10
compile   org.freedesktop.fwupd         1.7.6
runtime   org.freedesktop.gusb          0.3.10
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```
pine64-pinephonepro:~# fwupdmgr get-devices
WARNING: This package has not been validated, it may not work properly.
Unknown Product
│
└─QUECTEL Mobile Broadband Module:
      Device ID:          27730920c2119d128e431bbd20a8f225a2013d3a
      Summary:            Quectel EG25-G modem
      Current version:    EG25GGBR07A08M2G_01.002.01.002
      Vendor:             QUALCOMM INCORPORATED (USB:0x2C7C)
      GUIDs:              db379a33-254f-5140-b37e-d36ae7e5c039
                          d18f31f1-a3fa-55a2-b4ed-decfbc1e004d ← USB\VID_2C7C&PID_0125&REV_0318&NAME_EG25GGB
                          1a2996cb-f86e-5583-a464-e1b96e1c6ae9 ← USB\VID_2C7C&PID_0125&REV_0318
                          587bf468-6859-5522-93a7-6cce552a0aa3 ← USB\VID_2C7C&PID_0125
                          22ae45db-f68e-5c55-9c02-4557dca238ec ← USB\VID_2C7C
      Device Flags:       • Updatable
                          • System requires external power source
```

**Additional questions**

- Operating system and version: postmarketOS edge
- Have you tried rebooting? yes
- Is this a regression? maybe?

hughsie, 2022-03-17:
IIRC the supplied smbios data was incomplete. Do `sudo fwupdtool hwids` to verify.

MartijnBraam, 2022-03-17:
Oh something is definetly wrong here

```
pine64-pinephonepro:~$ sudo fwupdtool  hwids
[sudo] password for user: 
Computer Information
--------------------
ProductName: Pine64 PinePhonePro
EnclosureKind: 8

Hardware IDs
------------
not available as 'Manufacturer' unknown
not available as 'Manufacturer' unknown
not available as 'Manufacturer' unknown
not available as 'Manufacturer' unknown
not available as 'Manufacturer' unknown
not available as 'Manufacturer' unknown
not available as 'Manufacturer' unknown
not available as 'Manufacturer' unknown
not available as 'Manufacturer' unknown
not available as 'Manufacturer' unknown
not available as 'Manufacturer' unknown
not available as 'Manufacturer' unknown
not available as 'Manufacturer' unknown
not available as 'Manufacturer' unknown
not available as 'Manufacturer' unknown
```

MartijnBraam, 2022-03-17:
Got the right one:

```
Computer Information
--------------------
BiosVendor: U-Boot
BiosVersion: 2021.10
BiosMajorRelease: 21
BiosMinorRelease: 10
FirmwareMajorRelease: ff
FirmwareMinorRelease: ff
Manufacturer: Unknown
Family: 
ProductName: Unknown Product
ProductSku: 
EnclosureKind: 3
BaseboardManufacturer: Unknown
BaseboardProduct: Unknown Product

Hardware IDs
------------
{641d78a2-8383-59e4-867a-e4dde7268166}   <- Manufacturer + Family + ProductName + ProductSku + BiosVendor + BiosVersion + BiosMajorRelease + BiosMinorRelease
{0ec56064-d285-589b-bd7a-6de35fb9c1c6}   <- Manufacturer + Family + ProductName + BiosVendor + BiosVersion + BiosMajorRelease + BiosMinorRelease
{ff5286e0-e0fe-5be3-ad22-e3e2d8cc3086}   <- Manufacturer + ProductName + BiosVendor + BiosVersion + BiosMajorRelease + BiosMinorRelease
{7bea3c53-5915-52f7-9832-dadf569ffd7a}   <- Manufacturer + Family + ProductName + ProductSku + BaseboardManufacturer + BaseboardProduct
{5d22a0ea-939c-5522-9560-db2ef07f5d9d}   <- Manufacturer + Family + ProductName + ProductSku
{f93ef407-fe7b-5e56-81e8-27da5dafb84a}   <- Manufacturer + Family + ProductName
{7ee9b36b-642d-596e-bc37-11de5c9d5693}   <- Manufacturer + ProductSku + BaseboardManufacturer + BaseboardProduct
{709c856c-52e3-5acb-b4bc-dcaaf69cba0d}   <- Manufacturer + ProductSku
{5049feab-7a22-5c13-be5d-4f734a5e1ce8}   <- Manufacturer + ProductName + BaseboardManufacturer + BaseboardProduct
{eeb4844b-e76e-5b5b-8238-f981556d1e88}   <- Manufacturer + ProductName
{7ee9b36b-642d-596e-bc37-11de5c9d5693}   <- Manufacturer + Family + BaseboardManufacturer + BaseboardProduct
{709c856c-52e3-5acb-b4bc-dcaaf69cba0d}   <- Manufacturer + Family
{e6518a3f-b88f-5fb1-bb31-ae832b5fd740}   <- Manufacturer + EnclosureKind
{6cd0cec5-1915-5123-965f-c428c5f724eb}   <- Manufacturer + BaseboardManufacturer + BaseboardProduct
{a143f026-6752-57b0-ad82-fb21735ca366}   <- Manufacturer
```

hughsie, 2022-03-17:
I think somebody needs to populate that data 🙂

closed 2022-04-27 by stale[bot]