fwupd/fwupd #4406: After successful upgrade, fwupdmgr syn-bkc ask for downgrade to the same version

tcap26, 2022-03-18:
**Describe the bug**
A clear and concise description of what the bug is.

**Steps to Reproduce**

1. fwupdmgr syn-bkc, upgrade broadcom NIC to new version ( 219.0.114.0 )
2. reboot the server
3. After reboot, run "fwupdmgr syn-bkc", it says "downgrade the NIC from 219.0.114.0 to 219.0.114.0

**Expected behavior**
No downgrade is required, as the NIC is at same version

**fwupd version information**
Please provide the version of the daemon and client.

```shell
fwupdmgr --version
```
1.7.5
Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices
```

**Additional questions**

- Operating system and version:
- Have you tried rebooting?
- Is this a regression?

hughsie, 2022-03-18:
Hmm, I can't reproduce here, and I'm getting an error when trying to sync the latest UEFI firmware `Base.1.8.InternalError;The request failed due to an internal service error.  The service is still operational.` -- but lets try to debug based on what you see. Can you get me the output of `fwupdmgr sync-bkc --verbose` on your machine please?

hughsie, 2022-04-19:
@tcap26 verbose data please; or by email if you're more comfortable with that.

closed 2022-07-31 by stale[bot]