fwupd/fwupd #4417: fwupd fails to detect Logitech Unifying Receiver

spiwocoal, 2022-03-20:
**Describe the bug**
Running `fwupdmgr get-devices` with a Logitech Unifying Receiver (ID `046d:c534`) plugged into a USB port fails to detect the device, showing only PCIe devices.

**Steps to Reproduce**
1. Run `fwupdmgr get-devices` with a Logitech Unifying Receiver (ID `046d:c534`) plugged into a USB port.

**Expected behavior**
The Receiver should show up among the detected devices. This should be something like this:
```
Unifying Receiver
  DeviceId:             dfae5e57c0357cad4c2423cd787bde43114ae1e9
  Guid:                 279ed287-3607-549e-bacc-f873bb9838c4
  Guid:                 21e75d9a-5ce6-5da2-b7ab-910c7f3f6836
  Guid:                 9d131a0c-a606-580f-8eda-80587250b8d6
  Summary:              A miniaturised USB wireless receiver
  Plugin:               unifying
  Flags:                updatable|supported|registered
  Vendor:               Logitech
  VendorId:             USB:0x046D
  Version:              RQR12.09_B0030
  VersionBootloader:    BOT01.04_B0016
  Icon:                 preferences-desktop-keyboard
  Created:              2018-09-25
```

**fwupd version information**

```shell
$ fwupdmgr --version
runtime   org.freedesktop.fwupd         1.7.6
runtime   com.dell.libsmbios            2.4
compile   org.freedesktop.gusb          0.3.10
compile   com.hughsie.libjcat           0.1.9
runtime   org.kernel                    5.15.28-1-MANJARO
compile   org.freedesktop.fwupd         1.7.6
runtime   org.freedesktop.gusb          0.3.10
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
fwupd was installed from `pacman`.

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
$ fwupdmgr get-devices --show-all-devices
To Be Filled By O.E.M.
│
├─USB2.0 Hub:
│     Device ID:          0d5d05911800242bb1f35287012cdcbd9b381148
│     Summary:            USB 3.x hub
│     Current version:    5.64
│     Vendor:             VIA Labs, Inc. (USB:0x2109)
│     Install Duration:   15 seconds
│     GUIDs:              8b6303f4-7a3a-5577-91d2-4ab5ab5ccdc2
│                         3f0581b6-15ed-5f87-91eb-296db71e0838
│                         d390a7f5-ea48-5370-87a7-eba89bf5606e
│                         e3af93e4-9387-5afc-b83b-6a7a21ce93c5
│                         0adb66da-75a7-58ef-9abd-11baf5acd519
│     Device Flags:       • Updatable
│                         • Cryptographic hash verification is available
│                         • Device stages updates
│                         • Device can recover flash failures
│                         • Unsigned Payload
│
├─USB3.0 Hub:
│     Device ID:          68212d3b78e1f4971b1df26b80046d3cedb84611
│     Summary:            USB 3.x hub
│     Current version:    5.64
│     Vendor:             VIA Labs, Inc. (USB:0x2109)
│     Install Duration:   15 seconds
│     GUIDs:              40dff49b-65d9-5e31-bee3-ff7c80f64db5
│                         26f33695-3a3e-5c08-badb-f6141390ebd9
│                         10eb3a15-c177-5810-af53-1963e9200e65
│                         5527ecdd-2db5-54a6-83fc-f0db6511b4e6
│                         d87dbd95-2f89-5d8d-bd8f-899b3781d57d
│     Device Flags:       • Updatable
│                         • Cryptographic hash verification is available
│                         • Device stages updates
│                         • Device can recover flash failures
│                         • Unsigned Payload
│
├─400 Series Chipset SATA Controller:
│     Device ID:          b640caac3e7ebce3633ffc8d0f718061d1f11812
│     Current version:    01
│     Vendor:             Advanced Micro Devices, Inc. [AMD] (PCI:0x1022)
│     GUIDs:              fa5e6cc7-75d3-5ba4-919f-bd9c1c544ac5
│                         c7dfec4f-7bf2-50c6-b3e7-eb7da3d74bc7
│                         37d930ee-e811-585c-a094-f430aa4d42fb
│                         40dd51ce-4d2b-5863-a55b-e5e05b828e18
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│
├─Navi 10 [Radeon RX 5600 OEM/5600 XT / 5700/5700 XT]:
│     Device ID:          f0c42c4d354e719e9d5cd3366ebe367cbd98e2bf
│     Current version:    c1
│     Vendor:             Advanced Micro Devices, Inc. [AMD/ATI] (PCI:0x1002)
│     GUIDs:              f604da8c-121f-576a-8b6f-c7ccc76830b7
│                         07f449d0-5ce9-50dc-bad6-2d648c30b8ca
│                         afa52ef7-5494-518d-980c-54328b60c0f3
│                         dd0fc8c4-f037-5a78-bdea-c85eb0fe68fa
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│
├─Ryzen 7 3700X 8-Core Processor:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Vendor:             AMD
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f
│                         997104da-ec11-5e04-9d94-ac5501fbf609
│                         b54f3c89-8a41-5b9f-9a5c-0367c19a9451
│                         d951a23c-a920-50ac-ad89-0660aa0be95c
│     Device Flags:       • Internal device
│
├─ST3000DM001-1ER166:
│     Device ID:          98a419e2ed3f093585976ba88bdd85005199fef8
│     Summary:            ATA drive
│     Current version:    CC26
│     Vendor:             Seagate (ATA:0x1BB1, OUI:000c50)
│     GUIDs:              38c8eb5a-4733-5f10-8486-b25d6c5644de
│                         abf665a2-f4ee-5499-be13-d3026b05b5b1
│                         e531282b-708d-5d5e-aa45-671d3f1c02dc
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│
├─WDC WD20EZAZ-00GGJB0:
│     Device ID:          f78273dfd86c017eb3fdac51ca76f149af4aa87f
│     Summary:            ATA drive
│     Current version:    80.00A80
│     Vendor:             Western Digital (ATA:0x101C, OUI:0014ee)
│     GUIDs:              f7e3fc20-dd83-5c81-afc3-87b13ddef1f3
│                         88578003-fa12-54d0-9b1a-f51d41e2efd8
│                         51461388-9e49-540c-b2b8-cac7c8dffa1e
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│
└─WDC WDS100T2B0C-00PXH0:
      Device ID:          71b677ca0f1bc2c5b804fa1d59e52064ce589293
      Summary:            NVM Express solid state drive
      Current version:    211070WD
      Vendor:             Sandisk Corp (NVME:0x15B7)
      GUIDs:              58b87924-c1a3-5e04-8242-8afea5bd4dd8
                          83579426-bdbb-5944-9af0-393070f66a99
                          1656943f-1d12-5880-b7ae-18ebbb53c90d
                          dd75b173-c635-57e1-9569-c9da6dca84bb
                          fd4b0ada-7adb-544b-84a2-12ab288ee812
      Device Flags:       • Internal device
                          • Updatable
                          • System requires external power source
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update
```

**Additional questions**
The reason I am trying to update my Logitech Unifying Receiver's firmware is that it isn't working properly (I can't get `ltunify` nor `libratbag` to recognize the device), but when running `lsusb` I do see the device appear as `Bus 001 Device 004: ID 046d:c534 Logitech, Inc. Unifying Receiver`.
Also I think it is important to add is that my fwupd's logs (obtained from running `systemctl status fwupd`) have a couple errors:
```
mar 20 17:36:13 spiwocoal-desktop systemd[1]: Starting Firmware update daemon...
mar 20 17:36:13 spiwocoal-desktop fwupd[20956]: 20:36:13:0833 FuPluginUefiCapsule  skipping device that failed coldplug: ESRT GUID '00000000-0000-0000-0000-000000000000' was not valid
mar 20 17:36:14 spiwocoal-desktop fwupd[20956]: 20:36:14:0051 FuEngine             failed to add device usb:03:00:03:05: failed to get version: endpoint stalled or request not supported
mar 20 17:36:14 spiwocoal-desktop systemd[1]: Started Firmware update daemon.
```

- Operating system and version:
```shell
$ uname -a
Linux spiwocoal-desktop 5.15.28-1-MANJARO #1 SMP PREEMPT Fri Mar 11 14:12:57 UTC 2022 x86_64 GNU/Linux
```
- Have you tried rebooting?
Yes.
- Is this a regression?
Not sure.

hughsie, 2022-03-20:
I don't think that's a unifying receiver -- that's a "nano" protocol device -- e.g. see https://pwr-solaar.github.io/Solaar/devices and https://en.wikipedia.org/wiki/Logitech_Unifying_receiver

spiwocoal, 2022-03-20:
You're right :/

I installed Solaar and it immediately told me that it was a Nano receiver and sadly the device I'm trying to pair it to is not compatible (an M570) :(

closed 2022-03-20 by spiwocoal