fwupd/fwupd #4438: Intel USB4 host controller showing with Update Error: No Vendor ID set

mrhpearson, 2022-03-24:
**Describe the bug**
On a number of Lenovo platforms with the latest fwupd we're seeing

├─USB4 host controller:
│     Device ID:          56b4d077430ab6c4a18481fe170c00e98c6b6282
│     Summary:            Unmatched performance for high-speed I/O
**│     Update Error:       No vendor ID set**
│     GUID:               e72e778e-94f7-5ed2-b560-1c1262ee217c
│     Device Flags:       • Internal device
│                         • System requires external power source
│                         • Updatable
│                         • Signed Payload

Customers are flagging the "update error : No vendor ID set" as a problem with our systems

**Steps to Reproduce**
fwupdmgr get-devices

**Expected behavior**
Shouldn't display as an update error.

**fwupd version information**
runtime   com.dell.libsmbios            2.4
runtime   org.freedesktop.gusb          0.3.10
compile   com.hughsie.libjcat           0.1.9
compile   org.freedesktop.gusb          0.3.10
runtime   org.kernel                    5.10.0-12-amd64
compile   org.freedesktop.fwupd         1.7.6
runtime   org.freedesktop.fwupd         1.7.6

**Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):**
N/A

**fwupd device information**
N/A

**Additional questions**

- Operating system and version: Multiple
- Have you tried rebooting? N/A
- Is this a regression? I don't think so - looks like new functionality

superm1, 2022-03-24:
I think the root of the problem is these shouldn't be marked updatable in the first place.

hughsie, 2022-03-24:
@superm1 do we know when to export the "USB4 host controller" -- or is the answer "never"?

PR #4439 (2022-03-24): Don't export USB4 host controllers as updatable if they don't have un…
…ique GUIDs

Fixes: #4438

Type of pull request:

- [ ] New plugin (Please include [new plugin checklist](https://github.com/fwupd/fwupd/wiki/New-plugin-checklist))
- [x] Code fix
- [ ] Feature
- [ ] Documentation
commit 1241e156 (2022-03-24): Don't export USB4 host controllers as updatable if they don't have unique GUIDs
Fixes: #4438
closed 2022-03-24 by hughsie