fwupd/fwupd #4445: Export device serials and InstanceIds to non-root users

lancer1268, 2022-03-27:
**Is your feature request related to a problem? Please describe.**
Currently the command `lsusb` can list the USB devices. I want to add an extra field `firmware_version` to each item that has firmware information provided by fwupd.

I'm trying to read device `serial`s and `InstanceIds` using the `GetDevices` D-Bus call from a dedicated non-root user. These fields are used to match the items in `lsusb` to items in `fwupd`. However, the information is only available to trusted requests, which are requests sent from the root user. (by checking that `uid == 0`)

**Describe the solution you'd like**
Add another D-Bus method to export all fields, including `serial` and `InstanceIds`. The access control of this new method can be configured by the D-Bus .conf files so that the dedicated user can be granted the access.

**Describe alternatives you've considered**
None so far.

hughsie, 2022-03-28:
> from a dedicated non-root user

Both of these things might be privileged, hence requiring root. I suppose if the device is removable then that argument is less convincing as you can just unplug it and re-plug it on a machine where you *do* have root access. @superm1 what's your take here?

superm1, 2022-03-28:
> I do also see the argument for removable devices to advertise the instance IDs.

Actually how about we look at what udev exports today for serial on the devices for an unpriv'ed user?  `udevadm info --export-db | grep ID_SERIAL` should give a sampling of that.  If this info is already there, we're artificially hiding it in fwupd already anyway this means.

lancer1268, 2022-03-29:
> 1. Do you really need serial to be able to connect the two?  I think not - I think you just need the PhysicalID.  Maybe that's what we want to export instead as part of the get-devices payload?  We could potentially even link it back to a specific kernel device node instead of what we do today (`PhysicalId: usb:03:00:04`) so it's easier to map from other software.

You're right. Serial is not critical for matching the devices. It's just a workaround. 

Currently it is hard to connect the devices in fwupd to those from other sources.  It would be great if fwupd could provide such information like physical ids or device nodes.

closed 2022-03-30 by hughsie