fwupd/fwupd #4460: Override of UEFI checking via config is not working in Debian.

rahmanshaber, 2022-03-30:
**Describe the bug**
Override of UEFI checking via config is not working in Debian. My device sony viao ux

**Steps to Reproduce**
Follow this guide https://github.com/fwupd/fwupd/wiki/PluginFlag:legacy-bios.

**Expected behavior**
As i followed this guide, (https://github.com/fwupd/fwupd/wiki/PluginFlag:legacy-bios) it should show the option to update the firmware in the GUI app, also stop showing the  "Firmware can't be updated in legacy BIOS Mode"

**fwupd version information**
Please provide the version of the daemon and client.

```shell
fwupdmgr --version
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

**fwupd device information**
VGN-UX380N
│
├─Intel(R) Core™ Solo CPU     U1500  @ 1.33GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x00000054
│     Vendor:             Intel
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f
│                         30249f37-d140-5d3e-9319-186b1bd5cac3
│                         638417e7-ec26-53e7-8f9e-80df9bc4a29e
│                         c1dcb1ab-1acc-5c2d-ab28-c3584cc44f97
│     Device Flags:       • Internal device
│   
├─Mobile 945GM/GMS, 943/940GML Express Integrated Graphics Controller:
│     Device ID:          5792b48846ce271fab11c4a545f7a3df0d36e00a
│     Current version:    03
│     Vendor:             Intel Corporation (PCI:0x8086)
│     GUIDs:              3b15b0e1-70ad-5c8e-913d-c07d65705248
│                         8fd495cf-2aa2-514e-9de8-49ceb45f41d1
│                         22ce31f7-5e60-58d9-8c4d-417ba6214cd3
│                         11496496-987d-58ed-be5c-a99bcf2eb5b7
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
└─UGX:
      Device ID:          7622d5fdbf1d1e08138156da7d83bf693986ad16
      Current version:    19.15
      Vendor:             ALPS (USB:0x044E)
      GUIDs:              d7c11f1a-8255-5a85-aefd-850f441ae18e
                          0bcca859-e153-53fa-b5cd-265f48329b61
      Device Flags:       • Updatable

```shell
fwupdmgr get-devices --show-all-devices
```

**Additional questions**

- Debian 11.3
- yes
- Is this a regression?

hughsie, 2022-03-30:
> it should show the option to update the firmware in the GUI app

No, changing the `DisabledPlugins` option just hides the warning, rather than making the hardware support UEFI updates.

rahmanshaber, 2022-03-30:
My device don't have UEFI, only bios. What can be done to use the FWUPD?

hughsie, 2022-03-30:
fwupd only supports UEFI system firmware updates unfortunately.

closed 2022-03-30 by rahmanshaber

superm1, 2022-03-30:
> That's a bummer. So what's the point of that warning? i think a clear text that Firmware update only possible in UEFI enabled system will be better.  

A number of machines can run in both modes, but firmware updates only work in UEFI mode.

So this warning message is a line of best fit.

klemensn, 2022-05-27:
https://github.com/fwupd/fwupd/issues/2576#issuecomment-726493409 explained to add "bios" rather than "uefi" to `DisabledPlugins` which does work work for me on a ThinkPad X250 booted in BIOS mode.

Leaving this here as a note to other BIOS users who want to disable this warning on UEFI capable devices with UEFI support enabled in fwupd.

I suggest updating https://github.com/fwupd/fwupd/wiki/PluginFlag:legacy-bios accordingly.

hughsie, 2022-05-27:
Can you update it please; it's a wiki

rahmanshaber, 2022-05-29:
My device with is from 2008 don't have UEFI support and when i run get-devices then i can see there is an update available, so what's the point of that new firmware if that device don't have UEFI support?