fwupd/fwupd #4469: Brand-new Dell Precision 7760:  Failed to activate service 'org.freedesktop.fwupd': timed out

cptanalatriste, 2022-03-31:
**Describe the bug**
I got a brand-new Dell Precision 7760 with Ubuntu 20.04 installed. Many times when I try to install software, I see this error message from the Ubuntu Software app:

```
Failed to connect to daemon: Error calling StartServiceByName for org.freedesktop.fwupd: Failed to activate service 'org.freedesktop.fwupd': timed out (service_start_timeout=25000ms)
```

**Steps to Reproduce**
The error on the Ubuntu Software app happens randomly. However, I noticed I see the same error message running `fwupdmgr update` on terminal.

**Expected behavior**
No error messages, when running the Ubuntu Software app or running terminal commands.

**fwupd version information**
This is the command output. Notice the same error message appears.
```shell
$ fwupdmgr --version
client version:	1.5.11
compile-time dependency versions
	gusb:	0.3.4

Failed to connect to daemon: Error calling StartServiceByName for org.freedesktop.fwupd: Failed to activate service 'org.freedesktop.fwupd': timed out (service_start_timeout=25000ms)
```

I did not install `fwupd` manually. I'm using whatever came by default in the laptop.

**fwupd device information**

I see the same error when running the following:

```shell
$ fwupdmgr get-devices --show-all-devices
Failed to connect to daemon: Error calling StartServiceByName for org.freedesktop.fwupd: Failed to activate service 'org.freedesktop.fwupd': timed out (service_start_timeout=25000ms)
```

**Additional questions**

- Operating system and version: Ubuntu 20.04.4 LTS (64-bit)
- Have you tried rebooting?: Yes, many times.
- Is this a regression? Not sure. After first starting it, the system crashed, and not even booted. After a few restarts, is now booting but that error started appearing.

superm1, 2022-03-31:
Can you please run all the updates in update manager? There is a newer version available. If it keeps failing with that we can debug from there.

cptanalatriste, 2022-03-31:
When running the `update-manager` I got a `Not all updated can be installed message` message, so I only run a `Partial Upgrade`. After restarting, I can now run `fwupdmgr --version` and `fwupdmgr get-devices --show-all-devices` without a problem.

It appears to be solved. Thanks a lot for the help!

closed 2022-03-31 by cptanalatriste