fwupd/fwupd #4514: Allow to specify the VERSION after DEVICE-ID|GUID to the downgrade command

gportay, 2022-04-11:
**Is your feature request related to a problem? Please describe.**
The `downgrade` command requires two parameters: the `device-id` and the `version`.

The utility prompts for these parameters if unset through the command-line.

However, the utility accepts the `DEVICE-ID|GUID` to be specified, but it does not accepts for second argument that would be the version in this case.

Implementing this would help me to get rid of doing some hacky things for automating some testing; such as

``` sh
choice=$(fwupdmgr downgrade b0a78eb71f4eeea7df8fb114522556ba8ce22074 </dev/null | sed -n "/2.0.6/s,\.\t.*$,,p")
fwupdmgr downgrade --assume-yes b0a78eb71f4eeea7df8fb114522556ba8ce22074 <<<"$choice"
```

_note_: not sure the command above is working actually.

**Describe the solution you'd like**

```
$ fwupdmgr --help
  downgrade [DEVICE-ID|GUID] [VERSION]
                                                           Downgrades the firmware on a device
```

This would be nicer `fwupdmgr downgrade b0a78eb71f4eeea7df8fb114522556ba8ce22074 2.0.6`

**Describe alternatives you've considered**
None.

**Additional context**
It is all about automating the testing for device upgrades.

Does that make sense?

superm1, 2022-04-11:
How about `fwupdgmr install $DEVICEID $VERSION`?  Then the meaning is grab this specific version from a metadata source.  It's just overloading the first argument from being a URI/filename to also a device ID/GUID.

hughsie, 2022-04-11:
we've got install for local file installing -- doesn't it get very messy for overloading it with different args?

superm1, 2022-04-11:
I can't think of another verb that is more fitting to this specific use case of install a specific firmware version unless you want to add a very explicit `install-version` command.

BTW IIRC it's already overloaded for:

* Local filename
* Remote URI