fwupd/fwupd #4516: fwupd builds for Windows look for BSD kenv and then fail to attempt an update

cwsystem76, 2022-04-11:
**Describe the bug**
Attempting to install a firmware update via `fwupdtool` on a Windows machine results in a `kenv not supported` error:
```
PS C:\Program Files (x86)\fwupd\bin> .\fwupdtool.exe install --allow-older C:\Users\wagne\Downloads\ccc3c5e911fe4d87f4c7ee8a1fb7d6f0ffc73fd1f63c3c566781bfe8d3a2d103-launch_0.7.103-4007-ga49720.cab b43e0d166325bdb94f19c5b7971726f10d4ac65b
Loading…                 [***************************************]
WARNING: This package has not been validated, it may not work properly.
Decompressing…           [***************************************]
kenv not supported
```
I believe that this is due to this code from https://github.com/fwupd/fwupd/blob/main/src/fu-engine.c, which calls `fu_kenv_get_string` from https://github.com/fwupd/fwupd/blob/main/libfwupdplugin/fu-kenv.c:
```
fu_engine_get_kernel_cmdline(GError **error)
{
/* Linuxish */
if (g_file_test("/proc/cmdline", G_FILE_TEST_EXISTS))
return fu_engine_get_proc_cmdline(error);

/* BSDish */
return fu_kenv_get_string("kernel_options", error);
}
```
It looks for `/proc/cmdline` to see if this is a Linux system, and then assumes BSD-like if it doesn't find it. There's no third option, so it goes to a `kenv not supported` error as `fu_kenv_get_string` doesn't work on Windows. Adding an empty `/proc/cmdline` file to the Windows filesystem allows the update to proceed (but failed later on the process in this case, due to what I believe is a different problem that I am still investigating).

**Steps to Reproduce**
Attempt a firmware update on a Windows system. In my case, I was attempting to update a System76 Launch keyboard.

**Expected behavior**
I expected the firmware update to work, or at least proceed to the point where it's making the attempt.

**fwupd version information**
It's a Windows machine, no daemon. `fwupdtool` is version 1.7.7, installed from the downloadable `fwupd-1.7.7-setup-x86_64.exe` file.

**fwupd device information**
```
PS C:\Program Files (x86)\fwupd\bin> .\fwupdtool.exe get-devices
Loading…                 [***************************************]
WARNING: This package has not been validated, it may not work properly.
All Series
│
└─Launch Configurable Keyboard (launch 1):
      Device ID:          b43e0d166325bdb94f19c5b7971726f10d4ac65b
      Current version:    0.7.103-4011-g708ed5
      Vendor:             System76 (USB:0x3384)
      GUIDs:              b2cfb86e-dacd-5663-83e6-58059831ab6d
                          7549de75-189b-5fa6-911a-c9f5d127dc28 ← USB\VID_3384&PID_0001&REV_0001
                          af6272df-30f2-5067-ae05-201e69cd08a5 ← USB\VID_3384&PID_0001
      Device Flags:       • Updatable
                          • Unsigned Payload
```
**Additional questions**

- Operating system and version:
Windows 10 21H2

- Have you tried rebooting?
Yes

- Is this a regression?
No.

closed 2022-04-11 by hughsie

cwsystem76, 2022-04-11:
That was exceedingly quick. Thank you!

superm1, 2022-04-11:
There should be new Windows binaries made from CI if you want to verify it in advance of the next release.  Hopefully it's the only thing that needed fixing :)

cwsystem76, 2022-04-11:
No, there's definitely something else broken further along in the update process, but it's unclear what caused it. This was the only problem I could nail down definitively so the other issues will wait until I have more information on what went wrong. Thanks again.

superm1, 2022-04-11:
Please feel free to open a new issue with `--verbose` applied to the tool so we can see what is next.