fwupd/fwupd #4533: Xubuntu: Secure boot is enabled, but shim isn't installed to the EFI system partition

obel1x, 2022-04-17:
fwupd is complaining about missing shim, while looking in an automount-directory. Where is that set?

```
root@xubuntu-stick:/etc# fwupdmgr update
Devices with no available firmware updates: 
 • USB2.0 Hub
 • USB2.0 Hub
 • USB3.1 Hub
 • USB3.1 Hub
 • MZNLN256HAJQ-000L7
 • ThinkPad Hybrid USB-C with USB-A Dock
 • ThinkPad USB-C Dock Hybrid PD Controller
 • Thunderbolt host controller
 • UEFI Device Firmware
 • UEFI dbx
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade Embedded Controller from 0.1.12 to 0.1.19?                           ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ Lenovo ThinkPad T490/T590 Embedded Controller Firmware 1.19.The computer     ║
║ will be restarted automatically after updating EC completely. Do NOT turn    ║
║ off your computer or remove the AC adaptor while update is in progress.      ║
║                                                                              ║
║ No update for LVFS.                                                          ║
║                                                                              ║
║ 20N3S04300 must remain plugged into a power source for the duration of the   ║
║ update to avoid damage.                                                      ║
╚══════════════════════════════════════════════════════════════════════════════╝

Perform operation? [Y|n]: 
Downloading…             [***************************************]
Decompressing…           [***************************************] Less than one minute remaining…
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Authenticating…          [***************************************]
Restarting device…       [***************************************]
Writing…                 [***************************************]
Scheduling…              [***************************************]
Decompressing…           [***************************************]
Secure boot is enabled, but shim isn't installed to /media/root/989C-03BC/EFI/ubuntu/shimx64.efi

```
==============

```
root@xubuntu-stick:/etc# fwupdmgr --version
runtime   org.freedesktop.fwupd         1.7.5
compile   org.freedesktop.gusb          0.3.5
runtime   com.dell.libsmbios            2.4
runtime   org.kernel                    5.13.0-39-generic
compile   org.freedesktop.fwupd         1.7.5
runtime   org.freedesktop.gusb          0.3.5

```

```shell
root@xubuntu-stick:/etc# fwupdmgr get-devices --show-all-devices
20N3S04300
│
├─USB2.0 Hub:
│     Device ID:          301046452a49d84af6356d23e43a684b8f10660f
│     Summary:            USB 3.x hub
│     Current version:    55.43
│     Vendor:             VIA Labs, Inc. (USB:0x17EF)
│     Install Duration:   15 seconds
│     GUIDs:              fd59ee83-3328-5054-befc-4b2f800c274d ← USB\VID_17EF&PID_1028&REV_5543
│                         32588dcf-0447-538a-82fa-6730a0e11840 ← USB\VID_17EF&PID_1028
│                         1efdf95f-c3b6-5592-ab73-b7c0da7a547a ← USB\VID_17EF&PID_1028&HUB_1C
│                         bff55c76-fa0d-52a9-b9e3-e2174798a12c ← USB\VID_17EF&PID_1028&SPI_C840&REV_5543
│                         47d47071-a70b-57bf-a8c4-404e072b7c62 ← USB\VID_17EF&PID_1028&SPI_C840
│     Device Flags:       • Updatable
[... 836 lines cut ...]
├── initrd.img -> initrd.img-5.13.0-39-generic
├── initrd.img-5.13.0-37-generic
├── initrd.img-5.13.0-39-generic
├── initrd.img.old -> initrd.img-5.13.0-37-generic
├── memtest86+.bin
├── memtest86+.elf
├── memtest86+_multiboot.bin
├── System.map-5.13.0-37-generic
├── System.map-5.13.0-39-generic
├── vmlinuz -> vmlinuz-5.13.0-39-generic
├── vmlinuz-5.13.0-37-generic
├── vmlinuz-5.13.0-39-generic
└── vmlinuz.old -> vmlinuz-5.13.0-37-generic

5 directories, 582 files
```

**Additional questions**

- Operating system and version: xubuntu
- Have you tried rebooting? yes
- Is this a regression?
- Are you using an NVMe disk? no
- Is secure boot enabled? yes
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? dont know (no access to bios)

superm1, 2022-04-17:
It should be detecting your EFI system partition and then mounts it if it's not mounted.  Can you please manually mount it and share the output of tree from it?

You do have `shim-signed` installed, right?  If it possible you have multiple EFI system partitions maybe?

obel1x, 2022-04-17:
oh right, i see: that partition "989C-03BC" is the bootloader of windows on /dev/sda1, not having shim installed. My system runs from plugged in usb-stick and /boot/efi is mounted on /dev/sdb2 (UUID="9CAE-C17C"), which is the right bootloader having shim...

I cannot (should NOT) modify system-drive sda. It seems like the detection of the actual used bootloader-partition does not find the right one, used by the system.

Have a look at efimgr-output: the BootCurrent-id is different from what fwupdmgr is using...

Hers output of blkids
`root@xubuntu-stick:/boot/efi/EFI/ubuntu# blkid | grep fat
/dev/sda1: UUID="989C-03BC" BLOCK_SIZE="512" TYPE="vfat" PARTLABEL="EFI system partition" PARTUUID="debd0164-68f4-4081-84db-704a677ce0c5"
/dev/sdb2: UUID="9CAE-C17C" BLOCK_SIZE="512" TYPE="vfat" PARTUUID="8b25a3db-de26-4dad-8250-5c6825789ad9"
`
and 
`BootCurrent: 0003
BootOrder: 0003,0002,0000,0001,001B,001C,001D,001E,001F,0020,0021,0022,0023,0024,0012,0011
Boot0000* ubuntu	HD(2,GPT,f87eb52f-544f-43b7-aedf-bf98bb8dc49f,0x1000,0x100800)/File(\EFI\ubuntu\shimx64.efi)
Boot0001* Windows Boot Manager	HD(1,GPT,debd0164-68f4-4081-84db-704a677ce0c5,0x800,0xfa000)/File(\EFI\Microsoft\Boot\bootmgfw.efi)WINDOWS
Boot0002* ubuntu	HD(2,MBR,0x2a8e8ec9,0x800,0x18800)/File(\EFI\ubuntu\shimx64.efi)
Boot0003* ubuntu	HD(2,GPT,8b25a3db-de26-4dad-8250-5c6825789ad9,0x1000,0x100800)/File(\EFI\ubuntu\shimx64.efi)`

Yes, i am using shim-signed for boot and shimx64.efi, which are working fine

superm1, 2022-04-19:
That's right - we don't look at Boot variables to map the correct ESP.

It's pretty uncommon to have multiple ESP so that is definitely the cause here.

There is a configuration option in /etc/fwupd/daemon.conf. can that help you here?

superm1, 2022-04-19:
The option is actually in `/etc/fwupd/uefi_capsule.conf`.  It's this one:
```
# the EFI system partition (ESP) path used if UDisks is not available
# or if this partition is not mounted at /boot/efi, /boot, or /efi
#OverrideESPMountPoint=
```

You can manually mount your ESP and then put the mount point there to force it's usage.

obel1x, 2022-04-20:
well, /boot/efi is already mounted to the right partition.
```
horst@xubuntu-stick:/etc$ cat fstab | grep boot
# /boot/efi was on /dev/sda2 during installation
UUID=9CAE-C17C  /boot/efi       vfat    umask=0077      0       1

horst@xubuntu-stick:/etc$ df | grep boot
/dev/sdb2                      524272     5344    518928    2% /boot/efi
```
Shouldn't the right UUDI be used then?

superm1, 2022-04-21:
> well, /boot/efi is already mounted to the right partition.
> 
> ```
> horst@xubuntu-stick:/etc$ cat fstab | grep boot
> # /boot/efi was on /dev/sda2 during installation
> UUID=9CAE-C17C  /boot/efi       vfat    umask=0077      0       1
> 
> horst@xubuntu-stick:/etc$ df | grep boot
> /dev/sdb2                      524272     5344    518928    2% /boot/efi
> ```
> 
> Shouldn't the right UUDI be used then? Or what mount point should be taken for override?

Yeah just use /boot/efi in that conffile then.

superm1, 2022-04-21:
At first glance of those code I am surprised that it tried to use that ESP rather than complaining it found too many ESPs.
The reason is that we explicitly look for internal ESPs.  So we only found "one".  Your use case is (relatively) unusual.  Hopefully you can manage to get by the conffile change.

obel1x, 2022-04-23:
Yes, that worked, thank you. Even if this is uncommon - is there a reason for only looking for internal ESP even if mounted right external?
My usecase is uncommon, but with usb3 and larger sticks together with Internet+clouds this is a really fantastic way to have your Linux everywhere in the pocket. I would expect this to geht more common.
Maybe remove that limitation?

superm1, 2022-04-23:
The reason not to look at external storage by default is that it will trigger false positives in the more common case of internal only ESP.

Unless someone adds in code to look at the EFI nvram BootCurrent variable and match it to an ESP that wouldn't be safe.

superm1, 2022-04-25:
If you (or someone else) wants to improve the situation for multiple ESPs to examine the `BootCurrent` variable to better match the target ESP I'm happy to review a PR.  We can consider to drop the matching to internal devices only in that instance too.  Otherwise anyone else who encounters this the correct W/A is described above (to put the mount point of your ESP in the `uefi.conf` configuration file).

obel1x, 2022-10-19:
tried. as ubuntu did not ship 1.8.6, i installed via snap (1.8.6-43).
got:

```
# fwupd.fwupdmgr update
missing signed bootloader for secure boot: /snap/fwupd/4411/libexec/fwupd/efi/fwupdx64.efi.signed cannot be found
```

Thats true, there is not signed at /boot/efi/EFI/ubuntu - there is only fwupdx64.efi and shimx64.efi
Shouldn't shim be used to chainload the non-signed package by default?
Or maybe the location at that snap is still wrong? How to check what gets mountet in /snap/fwupd/4411/libexec/fwupd/efi ?

It looks, like it is not doing so when using snap. Configs do look same, or did i miss something?

superm1, 2022-10-19:
Looks like a snap specific bug....

I double checked the stable and edge channel snaps.  The files are present, but it seems like the location is wrong.

```
$ find /snap/fwupd -name *.efi*
/snap/fwupd/4412/usr/libexec/fwupd/efi/fwupdx64.efi
/snap/fwupd/4412/usr/libexec/fwupd/efi/fwupdx64.efi.signed
/snap/fwupd/4400/usr/libexec/fwupd/efi/fwupdx64.efi
/snap/fwupd/4400/usr/libexec/fwupd/efi/fwupdx64.efi.signed
```
There is an extra prefix in where they are reletive to what it's looking for...

superm1, 2022-10-19:
OK, there should be a new one published to the edge snap store shortly that should fix this.  Have a try with that.

closed 2023-02-19 by hughsie