fwupd/fwupd #4536: Snap package outdated: still at 1.7.6

PhilippWendler, 2022-04-19:
**Describe the bug**

As of 2022-04-19, the [Snap package](https://snapcraft.io/fwupd) is still at version 1.7.6 although version 1.7.7 of fwupd has been released 14 days ago.

**Steps to Reproduce**

Look at the official [snap package of fwupd](https://snapcraft.io/fwupd) in channel `latest/stable`.

**Expected behavior**

Each fwupd release should be released as a snap package in a timely manner.

---

Note that the `latest/edge` channel of the snap package has version 1.7.6-268-g0b937e7a from today, so publishing to the snap store still works in principle. Maybe the reason for the problem is that 1.7.7 was not released from the main branch.

Another side note: The channels `latest/candidate` and `latest/beta` of the snap package are at 1.6.2. Likely having "stable" and "edge" is enough as channels for fwupd, but in order to not confuse users the other channels could be removed?

superm1, 2022-04-19:
Caused by https://github.com/fwupd/fwupd/commit/1084d150fdb19a006d68e54f1f333a22fde29b41, which normally should be done at the time of 1.8.0 release, but seems to have been done early.  Let me see if I can manually promote a 1.7.7 build.

superm1, 2022-04-19:
Manually uploaded the artifact here: https://app.circleci.com/pipelines/github/fwupd/fwupd/9609/workflows/4ca3cdc1-2843-4146-9194-dafad8bd66d4/jobs/34916/artifacts
```
Revision 3996 of 'fwupd' created.
Track    Arch    Channel    Version              Revision
latest   amd64   stable     1.7.7                3996
                 candidate  1.6.2                3305
                 beta       1.6.2                3305
                 edge       1.7.6-268-g0b937e7a  3995
```

superm1, 2022-04-19:
> Likely having "stable" and "edge" is enough as channels for fwupd, but in order to not confuse users the other channels could be removed?

Does snap store let you remove channels?  I agree but I don't know how to do it.

closed 2022-04-19 by superm1

superm1, 2022-04-19:
Ah I found a button.  It's only in the "new UI".
```
$ snap info fwupd
name:      fwupd
summary:   Firmware updates for Linux
publisher: Richard Hughes
store-url: https://snapcraft.io/fwupd
contact:   https://github.com/fwupd/fwupd
license:   LGPL-2.1+
description: |
  The full fwupd firmware updating stack for Linux.  Can be used to perform updates on all supported
  devices via https://fwupd.org
snap-id: HpOj37PuyuaMUZY0NQhtwnp7oS5P8u5R
channels:
  latest/stable:    1.7.7               2022-04-19 (3996) 32MB classic
  latest/candidate: ↑
  latest/beta:      ↑
  latest/edge:      1.7.6-268-g0b937e7a 2022-04-19 (3995) 33MB classic
```

PhilippWendler, 2022-04-19:
Thanks for the fast response and fixing this! Looks good now and 1.7.7 via the snap package installs and works fine.