fwupd/fwupd #4547: Specified firmware is older than installed 'X.Y.Z < #####'

realbiz21, 2022-04-22:
I am trying to update the BIOS of a Lenovo T460s and getting an error about version incompatibilities. When overriding the error, upon reboot it says "Unsupported capsule."

I downloaded the zip file (n1cul41w.zip) from pcsupport.lenovo.com and extracted it.

I booted the computer with Fedora 36 Workstation downloaded on 04/22/2022. I am doing all of this offline with no network connection.

I then executed:

```
fwupdmgr install N1CET86W.cab`
```

I got the "Specified firmware is older than installed 'X.Y.Z < #####'" error back.

I tried to force with:

```
fwupdmgr install --allow-older N1CET86W.cab
```

fwupdmgr decompressed and extracted it and said I needed to reboot. But after rebooting I saw "Unsupported capsule".

**fwupdmgr version** in Fedora 36 Workstation LiveCD:

```
runtime   org.freedesktop.fwupd         1.7.5
runtime   com.dell.libsmbios            2.4
compile   org.freedesktop.gusb          0.3.10
runtime   org.kernel                    5.17.0-0.rc7.116.fc36.x86_64
compile   org.freedesktop.fwupd         1.7.5
runtime   org.freedesktop.gusb          0.3.10
```

**fwupdmgr devices output**:

```
(fwupdmgr:4408): FuMain-DEBUG: 18:24:36.134: FwupdDevice:
  DeviceId:             a45df35ac0e948ee180fe216a5f703f32dda163f
  Name:                 System Firmware
  Guid:                 230c8b18-8d9b-53ec-838b-6cfc0383493a ← main-system-firmware
  Guid:                 68b8d4d4-88dc-5c69-8013-67b942a77dd3 ← UEFI\RES_{42F29620-3D63-4F09-950B-9B7055570F28}
  Guid:                 42f29620-3d63-4f09-950b-9b7055570f28
  Summary:              UEFI ESRT device
  Plugin:               uefi_capsule
  Protocol:             org.uefi.capsule
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot|can-verify|usable-during-update
  Checksum:             SHA1(01c85b0155bca3ec143244bd75cde43cfa23b71e)
  Vendor:               Lenovo
  VendorId:             DMI:LENOVO
  Version:              0.1.53
  VersionLowest:        0.1.22
  VersionFormat:        triplet
  VersionRaw:           0x00010035
  VersionLowestRaw:     0x00010035
  Icon:                 computer
  Created:              2022-04-22
  UpdateState:          success
```

**fwupdmgr get-details** N1CET86W.cab:

```
Decompressing…: 0%
Idle…: 0%
XXX
│
└─System Firmware:
  │   Device ID:          a45df35ac0e948ee180fe216a5f703f32dda163f
  │   Description:
  │   Lenovo Thinkpad T460s System Firmware N1CET86W, version 1.54.
  │   Current version:    0.1.53
  │   GUID:               42f29620-3d63-4f09-950b-9b7055570f28
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Needs a reboot after installation
  │                       • Cryptographic hash verification is available
  │                       • Device is usable for the duration of the update
  │
  └─ThinkPad T460s:
        New version:      0.1.54
        Summary:          Lenovo ThinkPad T460s System Firmware
        License:          Proprietary
        Size:             9.4 MB
        Urgency:          High
        Description:
        Lenovo System Firmware Version 1.54
```

This issue is similar to #3903 which was resolved October 2021, against fwupd 1.5.x. This is why I am using Fedora 36 as it has fwupd version 1.7.x. Fedora 35 has version 1.6.x and also shows this issue.

hughsie, 2022-04-23:
Hmm, this is very confusing, I can't actually see what it wrong. Can you get the entire log of `sudo fwupdtool --verbose install N1CET86W.cab` please.

realbiz21, 2022-04-23:
@hughsie sorry, I captured the data wrong.

I've attached full traces:

- I booted the T460s in UEFI-only mode with Fedora 36 Workstation LiveUSB, without any network connection. fwupdmgr is 1.7.5 as shown in my first comment.
- [0-get-devices.txt](https://github.com/fwupd/fwupd/files/8547706/0-get-devices.txt): output of `fwupdmgr get-devices`. It shows current version as `65589` (instead of 0.1.53)
- [1-install.txt](https://github.com/fwupd/fwupd/files/8547707/1-install.txt): output of `fwupdtool --verbose install N1CET86W.cab`. This shows the error, `Specified firmware is older than installed '0.1.54 < 65589'`
- [2-install-allow-older.txt](https://github.com/fwupd/fwupd/files/8547708/2-install-allow-older.txt)
: output of `fwupdtool --verbose install --allow-older N1CET86W.cab`. This doesn't complain about versions and appears to allow installation.

Rebooting after `2-install-allow-older`, however, the UEFI updater doesn't run [or it either runs and quickly returns] and boots back into my Fedora 36 LiveUSB.

If I then reboot again, press F12 to select the boot device as LInuxFirmwareUpdater, it runs quickly [a quick flash], and nothing is seen on screen.

Side note: I think the discrepancy in Current Version that I had in my first comment and this comment is because I think I had network connected. So maybe when fwupdmgr has network it interprets the Current Version as a triplet. But I believe in _that_ scenario is when I would see Unsupported capsule after reboot from the LinuxFirmwareUpdater.

I'll stick to offline unless instructed otherwise, but I ultimately need to do this on multiple T460s which I do not want to supply a network connection to.

hughsie, 2022-04-25:
> It shows current version as 65589

This is expected. When we query the hardware it just tells us:

 * My ID is 42f29620-3d63-4f09-950b-9b7055570f28
 * My current version is 0x00010035
 * The lowest version I can downgrade to is 0x00010022

For Dell hardware we take that version and format it into the `0.1.2.3` dotted notation like you expect, for Lenovo we do `1.2.3` and for Intel we do something different again. This is called a version format. As the hardware doesn't tell us how to present the version we allow the LVFS metadata to "hint" about the presentation format. This is why you see the more familiar `0.1.53` on machines that had internet access; the metadata hinted *that GUID is a Lenovo update, use triplet format*.

Of course when you ask fwupd to compare the version of 65589 against 0.1.54 it gets very confused. In this case it'll assume both are semvers, and compare `0.1.54` against `65589.0.0` (simplifying slightly) and of course the latter is the larger number.

For offline use, you have two options:

 * Sync the /var/lib/fwupd/metadata/ directories from an online machine
 * *quirk* the hardware by dropping a file in `/var/lib/fwupd/quirks.d` or `/usr/share/fwupd/quirks.d` with the following content:

    whatever.quirk

    [42f29620-3d63-4f09-950b-9b7055570f28]
    VersionFormat = triplet

That should tell fwupd how to format the `65589` into `0.1.54` -- although you *might* need to restart fwupd manually (or reboot).

The `Unsupported capsule` is surprising, as at that point we've handed off to the firmware update EFI binary and am at the mercy of the system firmware. You might want to try fwupd-efi 1.3 just to make sure you're not hitting something fixed in the latest release.

realbiz21, 2022-04-26:
I used the quirk to force the GUID to triplet. `fwupdtool` no longer complained and appeared to do its thing.

However, I still got Unsupported capsule errors after reboot.

I then built `fwupd-efi` off of [master](https://github.com/fwupd/fwupd-efi/commit/639e7dd9122692a02c87aa3a661b0356ba352ec8), used my Fedora 36 Live USB and did:

- installed the quirk for VersionFormat
- replaced /usr/libexec/fwupd/efi/fwupdx64.efi with the version I built off master 
- `systemctl restart fwupd`
- `fwupdtool install N1CET86W.cab`

This still fails to install after the reboot, as I saw this on screen from LinuxFirmwareUpdater:

```
fwupd-efi version 1.4
WARNING: QueryCapsuleCapabilities failed, assuming EfiResetWarm: Unsupported
WARNING: Could not apply capsule update: Not Found
WARNING: Could not apply capsules: Not Found
Reset System
```
I saw a debugging flag in the fwupd-efi source, so I enabled that. Here's the full output from the EFI var that held the log:

```
../efi/fwupdate.c:554:efi_main(): fwupd-efi version 1.4
../efi/fwupdate.c:160:fwup_populate_update_table(): Found update fwupd-42f29620-3d63-4f09-950b-9b7055570f28-0
../efi/fwupdate.c:587:efi_main(): Adding new capsule
../efi/fwupdate.c:200:fwup_search_file(): Searching Device Path: HD(1,MBR,0x00000000)/\EFI\fedora\fw\fwupd-42f29620-3d63-4f09-950b-9b7055570f28.cap...
../efi/fwupdate.c:220:fwup_search_file(): Device Path prepared: HD(1,MBR,0x00000000)
../efi/fwupdate.c:230:fwup_search_file(): Device supporting SFSP: PciRoot(0)/Pci(0x17,0x0)/Sata(0x1,0x0,0x0)/HD(1,MBR,0x00000000)
../efi/fwupdate.c:233:fwup_search_file(): Comparing: HD(1,MBR,0x00000000) and PciRoot(0)/Pci(0x17,0x0)/Sata(0x1,0x0,0x0)/HD(1,MBR,0x00000000)
../efi/fwupdate.c:233:fwup_search_file(): Comparing: HD(1,MBR,0x00000000) and Pci(0x17,0x0)/Sata(0x1,0x0,0x0)/HD(1,MBR,0x00000000)
../efi/fwupdate.c:233:fwup_search_file(): Comparing: HD(1,MBR,0x00000000) and Sata(0x1,0x0,0x0)/HD(1,MBR,0x00000000)
../efi/fwupdate.c:233:fwup_search_file(): Comparing: HD(1,MBR,0x00000000) and HD(1,MBR,0x00000000)
../efi/fwupdate.c:242:fwup_search_file(): Match up! Returning \EFI\fedora\fw\fwupd-42f29620-3d63-4f09-950b-9b7055570f28.cap
../efi/fwupdate.c:302:fwup_open_file(): Found device
../efi/fwupdate.c:309:fwup_open_file(): Found volume
../efi/fwupdate.c:317:fwup_open_file(): Found file
../efi/fwupdate.c:417:fwup_add_update_capsule(): Read file; 9417312 bytes
../efi/fwupdate.c:418:fwup_add_update_capsule(): updates guid: 42F29620-3D63-4F09-950B-9B7055570F28
../efi/fwupdate.c:419:fwup_add_update_capsule(): File guid: 42F29620-3D63-4F09-950B-9B7055570F28
../efi/fwupdate.c:603:efi_main(): n_updates: 1
../efi/fwupdate.c:465:fwup_apply_capsules(): QueryCapsuleCapabilities failed, assuming EfiResetWarm: Unsupported
../efi/fwupdate.c:469:fwup_apply_capsules(): Capsules: 1
../efi/fwupdate.c:475:fwup_apply_capsules(): Could not apply capsule update: Not Found
../efi/fwupdate.c:618:efi_main(): Could not apply capsules: Not Found
../efi/fwupdate.c:621:efi_main(): Reset System
```

hughsie, 2022-04-27:
> Could not apply capsule update

Hmm. I've seen something like this before, when trying to update very old firmware to the latest one. In that case we could flash something less ancient using WinFlash "live" using Windows, which then updated the firmware enough so that subsequent UEFI capsule updates worked. The flaw in the update capsule design is that you have to ask the "old" firmware to update to the new firmware, and so if the old firmware has an issue (e.g. missing capsule support) then you SOL applying updates in fwupd. I think what's supposed to happen is the LVFS-provided firmware should have a firmware requirement for a version that's got a working UpdateCapsule RT service, but it appears in this case that's done.

What's confusing is that the old version isn't actually that old -- i.e .you're going from 0.1.53 to 0.1.54 -- and not 0.1.0 or older like I saw in the other report. @mrhpearson -- got any ideas? Are there any reports of 0.1.53 failing?

Also, I only see "Version 0.1.37 uploaded 3 years ago" on the LVFS and no newer versions than that. Is 0.1.53 _that_ old? Thanks.

realbiz21, 2022-04-27:
FWIW I tried the version on LVFS as well - same result (Unsupported).

from: https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN1CET.firmware

According to the LVFS, the current version I'm on (0.1.53) was only released 2021-11-21.

0.1.54 was on LVFS as of 2021-12-29 - only a one month difference.

I received these machines second-hand recently so I'm not sure how the previous versions were applied, or if that makes any difference. But they all had Windows on them when I got the machines so I doubt `fwupd` was used.

realbiz21, 2022-06-09:
@mrhpearson did you see the last question form @hughsie?

I have held off updating in case there is a new Lenovo BIOS to try debugging the issue, but otherwise I will upgrade using the ISO instead of fwupd.

mrhpearson, 2022-06-10:
Hi - I had missed that one, apologies.

I can flag this to the FW team (the team who have been doing this platforms updates have been really helpful) but one thought - can you confirm if you have secure boot enabled or not? There are some issues with shim that means you may need to disable secure boot for the updater to run. I'd like to rule it out as according to the reports there have been a number of succesful updates.

realbiz21, 2022-06-10:
@mrhpearson I do not have secure boot enabled. I am booting in legacy mode.

realbiz21, 2022-06-16:
As there is now a new BIOS out, I am going to upgrade using the ISO so that I am 1 version behind to not fall too far in security. I'll then try using fwupd to upgrade to the latest. If that works I'll close this issue since by now it seems clear the issue is in the BIOS update and not within fwupd code.

realbiz21, 2022-08-04:
BIOS update via fwupd did not succeed on the latest Lenovo T460s BIOS. Closing this issue as it's not an issue with fwupd but an issue with the Lenovo BIOS.

closed 2022-08-04 by realbiz21

devbranch-vadym, 2022-12-02:
Just wanted to highlight I've got the same issue with Thinkpad T14s Gen 3 (Intel):
```
Specified firmware is older than installed '0.1.30 < 655xx'
```

I've installed the firmware update using Lenovo's bootable ISO file and it worked fine, however, `fwupdmgr` still displays 65566 version, even after upgrade.

hughsie, 2022-12-03:
@devbranch-vadym that's a different issue, please file a new ticket.

dandv, 2023-04-05:
@hughsie As an end user, I don't know if I have the same problem or if it's a new one. Based on the feedback to @devbranch-vadym which likely caused them to [*not* file a new ticket](https://github.com/fwupd/fwupd/issues?q=is%3Aissue+is%3Aopen+commenter%3Adevbranch-vadym), I went ahead and filed #5686.