fwupd/fwupd #4550: fwupd-1.7.5 causes "systemd-modules-load: Failed to find module 'ipmi-devintf'" error

equaeghe, 2022-04-23:
**Describe the bug**
fwupd-1.7.5 causes systemd to try and load module `ipmi-devintf` even if it does not exist. This is caused by the mention of `ipmi-devintf` in `/lib/modules-load.d/fwupd-redfish.conf`.

**Steps to Reproduce**
Restart, look in log and notice issue.

**Expected behavior**
Restart, systemd is not led to try and load `ipmi-devintf`.

**fwupd version information**
```shell
fwupdmgr --version
compile   org.freedesktop.fwupd         1.7.5
runtime   org.kernel                    5.15.32-gentoo-r1-a
runtime   org.freedesktop.fwupd         1.7.5
```
Installed using Gentoo's package manager portage.

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices
20UDS0DK00                                                                                               
│                                                                                                        
├─Ryzen 7 PRO 4750U with Radeon Graphics:                                                                                                                                                                          
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027                                                                                                                                                 
│     Vendor:             AMD                                                                                                                                                                                      
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f ← cpu                                                                                                                                               
│                         997104da-ec11-5e04-9d94-ac5501fbf609 ← CPUID\PRO_0&FAM_17                      
│                         1f2e0c2f-e078-5fcb-a8de-f78eff9a470a ← CPUID\PRO_0&FAM_17&MOD_60               
│                         5fd5eb41-dd4a-5742-8448-861eaaa61cf8 ← CPUID\PRO_0&FAM_17&MOD_60&STP_1         
│     Device Flags:       • Internal device                                                                                                                                                                        
│                                                                                                        
├─SKHynix HFS001TD9TNI-L2B0B:                                                                            
│     Device ID:          71b677ca0f1bc2c5b804fa1d59e52064ce589293                                       
│     Summary:            NVM Express solid state drive                                                  
│     Current version:    11720C10                                                                       
│     Vendor:             SK hynix (NVME:0x1C5C)                                                         
│     Serial Number:      CJ0AN89731030AV5U                                                              
│     GUIDs:              424e8633-6c06-5c41-bd78-fbcb483dc0fb ← NVME\VEN_1C5C&DEV_1639&SUBSYS_1C5C1639&REV_00                                                                                                     
│                         b7603cf7-873e-59e8-b7aa-9d8a9877b6e7 ← NVME\VEN_1C5C&DEV_1639&SUBSYS_1C5C1639  
│                         af90743d-93b0-5a7f-b3aa-08db33ab1f82 ← NVME\VEN_1C5C&DEV_1639&REV_00           
│                         85ef24c5-fc0a-57a6-ab37-5711ac8acfbb ← NVME\VEN_1C5C&DEV_1639                  
│                         06f40517-864f-57be-9dad-d48bec1667e0 ← SKHynix_HFS001TD9TNI-L2B0B              
│     Device Flags:       • Internal device                                                              
│                         • Updatable                                                                    
│                         • System requires external power source                                        
│                         • Supported on remote server                                                                                                                                                             
│                         • Needs a reboot after installation                                            
│                         • Device is usable for the duration of the update                              
│                                                                                                        
├─System Firmware:                                                                                       
│ │   Device ID:          a45df35ac0e948ee180fe216a5f703f32dda163f                                                                                                                                                 
│ │   Summary:            UEFI ESRT device                                                               
│ │   Current version:    0.1.36                                                                         
│ │   Vendor:             Lenovo (DMI:LENOVO)                                                            
│ │   Update State:       Success                                                                        
│ │   Update Message:     The computer will be restarted automatically after updating BIOS completely. Do NOT turn off your computer or remove the AC adaptor while update is in progress.                         
│ │   Update Error:       Cannot install update when the lid is closed                                   
│ │   GUIDs:              01453b71-da0c-4832-9f4f-e378245339c7                                           
│ │                       230c8b18-8d9b-53ec-838b-6cfc0383493a ← main-system-firmware                    
│ │                       38ad8832-d322-5d73-aff7-77b8ae0f1a6f ← UEFI\RES_{01453B71-DA0C-4832-9F4F-E378245339C7}                                                                                                   
│ │   Device Flags:       • Internal device                                                              
│ │                       • System requires external power source                                                                                                                                                  
│ │                       • Supported on remote server                                                   
│ │                       • Needs a reboot after installation                                            
│ │                       • Cryptographic hash verification is available                                 
│ │                       • Device is usable for the duration of the update                              
│ │                       • Updatable                                                                                                                                                                              
[... 88 lines cut ...]
│                         • Device is usable for the duration of the update
│                         • Updatable
│                                                   
└─UEFI Device Firmware:                                                                                  
      Device ID:          f37fb01122dd62c773f4e84ec89737e059712d59
      Summary:            UEFI ESRT device                                                                                                                                                                         
      Current version:    24577            
      Minimum Version:    1                                                                              
      Vendor:             DMI:LENOVO                                                                     
      Update State:       Success                                                                        
      Update Error:       Cannot install update when the lid is closed
      GUIDs:              7d290ab6-bad9-4d27-bb17-642a3ef4cd98
                          29262650-a2b6-5264-aa9a-379f2ef3774d ← UEFI\RES_{7D290AB6-BAD9-4D27-BB17-642A3EF4CD98}
      Device Flags:       • Internal device                                                              
                          • System requires external power source
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update
                          • Updatable
```

**Additional questions**

- Operating system and version: *Gentoo Linux (rolling release; stable variant)*
- Have you tried rebooting? *yes*
- Is this a regression? *unknown*

hughsie, 2022-04-23:
I don't think there's a way to say "load ipmi-devintf only if it exists" and if we don't load it then the redfish plugin fails on some server systems.

equaeghe, 2022-04-23:
Is it fwupd's responsibility to load this? Normally, [kernel modules are loaded automatically by the kernel](https://www.freedesktop.org/software/systemd/man/modules-load.d.html#Description). If not, it seems the server administrator's job. fwupd's documentation can mention this. With the current approach, errors are being triggered on systems whose users shouldn't even be aware that redfish exists, i.e., the burden is placed in the wrong place.

superm1, 2022-04-25:
I think the problem here is that there isn't any module alias to know whether the underlying software/hardware necessary for redfish to work is there.  If we could get the kernel to add an appropriate module alias, then we can add a configurable behavior to either do that or not at install time with some hints from the kernel version you're using.

In your case if you don't have interest in redfish, just compile without `plugin_redfish`?  Wouldn't that cover the situation?

equaeghe, 2022-04-26:
Yes, I can override fwupd's `/lib/modules-load.d/fwupd-redfish.conf` with my own `/etc/modules-load.d/fwupd-redfish.conf -> /dev/null`. My distribution could also make the redfish plugin compilation conditional. However, I still think the burden is placed incorrectly: redfish users should configure loading the right module(s) instead of those not using redfish doing an effort to avoid an error.

closed 2022-06-06 by superm1