fwupd/fwupd #4551: Install the D-Bus interfaces XML file even if using `-Dintrospection=false`

lancer1268, 2022-04-24:
**Is your feature request related to a problem? Please describe.**
I want to access the D-Bus interfaces XML file in `$(datadir)/dbus-1/interfaces/` with the `-Dintrospection=false` option.

As described in the [D-Bus API Design Guidelines](https://dbus.freedesktop.org/doc/dbus-api-design.html)
> Interface files for public API should be installed to `$(datadir)/dbus-1/interfaces` so that other services can load them.

**Describe the solution you'd like**
Install the D-Bus interfaces XML file even if using `-Dintrospection=false`. Maybe revert [this commit](https://github.com/fwupd/fwupd/pull/4038/commits/e284b90cc102c642caaefcbec0cb08285243af1e).

**Describe alternatives you've considered**
Use another flag to enable the installation of the D-Bus XML file.

**Additional context**
I find the description of `introspection` in the `meson_options.txt`. It seems that the `introspection` here means the "GObject Introspection" instead of the "D-BUS Interface Introspection". Therefore, the D-Bus XML file should still be installed even if using `-Dintrospection=false`.
```
option('introspection', type : 'boolean', value : true, description : 'generate GObject Introspection data')
```

superm1, 2022-04-24:
Would you mind doing a PR to change this?

hughsie, 2022-04-24:
Ack, pr please.

lancer1268, 2022-04-24:
Ok, I can create PRs for this.

The `meson.build` for the `main` branch and `1_7_X` branch are different. Should I create one PR for each of them?

Note: My project currently uses the `1_7_X` branch.

superm1, 2022-04-24:
> Ok, I can create PRs for this.
> 
> The `meson.build` for the `main` branch and `1_7_X` branch are different. Should I create one PR for each of them?
> 
> Note: My project currently uses the `1_7_X` branch.

It should be done first in main. Then it should be backported/redone for 1_7_X.

lancer1268, 2022-04-24:
I see. Thank you.

hughsie, 2022-04-24:
Done! Can you do the same for 1_7_X pls

lancer1268, 2022-04-25:
Sure, but how can I do that for `1_7_X`? The presubmit check fails when I try to commit code to the `1_7_X` branch.
```
Don't commit to branch...................................................Failed
- hook id: no-commit-to-branch
- exit code: 1
```

superm1, 2022-04-25:
> Sure, but how can I do that for `1_7_X`? The presubmit check fails when I try to commit code to the `1_7_X` branch.
> 
> ```
> Don't commit to branch...................................................Failed
> - hook id: no-commit-to-branch
> - exit code: 1
> ```

Check out 1_7_X and then fork that to another branch.

closed 2022-04-25 by superm1