fwupd/fwupd #4560: Firmware update error of "failed to get device before update reload: ..." when it waits replug

chlee75, 2022-04-26:
**Describe the bug**
 Error message “failed to get device before update reload: failed to wait for detach replug: device c90bc6d52574e4f116fce36550e58ed589c31eb0 did not come back” sometimes happens while fwupd is waiting device replug with the RemoveDelay after updating firmware even though the device has been plugged to the computer within RemoveDelay.

fwupd daemon log  when this error happen : [fwupd.update_fail.log](https://github.com/fwupd/fwupd/files/8559864/fwupd.update_fail.log)

fwupdmgr error message 
$../build/src/fwupdmgr install --allow-older --allow-reinstall ./cab_file/HP-Thunderbolt_Dock_G4-V1.2.10.0.cab
Waiting…                 [****************************           ]failed to get device before update reload: failed to wait for detach replug: device c90bc6d52574e4f116fce36550e58ed589c31eb0 did not come back

**Steps to Reproduce**
Repeate updating HP G4 Dock firmware using $../build/src/fwupdmgr install --allow-older --allow-reinstall ./cab_file/HP-Thunderbolt_Dock_G4-V1.2.10.0.cab

**Expected behavior**
Error message “failed to get device before update reload: ..." should not happen because the device has been plugged to the computer within RemoveDelay after updating firmware.

**fwupd version information**
plugin name is ccgx and fwupd version is 1.8.0

**fwupd device information**
 HP Thunderbolt Dock G4

superm1, 2022-04-26:
Does this machine support pre-boot DMA protection?  Looking through log unless I'm reading it wrong It seems like no.
> [0m[32m03:04:11:0892 FuPluginAcpiDmar     [34mOemTableId: WHL     
[0m[32m03:04:11:0892 FuPluginAcpiDmar     [34mCreatorId: INTL
[0m[32m03:04:11:0892 FuPluginAcpiDmar     [34mFlags: 0x01

This means that the tunnels won't necessarily automatically create depending on the configured machine policy.  Is it possible that you only "authorized" the device once before you started and it didn't automatically authorize after being replugged?  While in this state, can you check the contents of `/sys/bus/thunderbolt/devices/` and see if the dock has `authorized` set to 1?  If it's set to 0, I would expect authorizing brings it back.

chlee75, 2022-04-26:
@superm1 
 Richard already knows this issue and he wants me to create this issue.

chlee75, 2022-04-27:
@hughsie 

I modified ccgx plugin to set FWUPD_DEVICE_FLAG_WAIT_FOR_REPLUG flag before sending trigger code to the the device to start firmware update to fix this issue in HP G4 Dock.  Device disconnect USB when it get trigger code.
See https://github.com/fwupd/fwupd/pull/4563

I have tested 2 days to update firmware 200 times and there was no "failed to get device before update reload:... " error message.
It seems that there was some timing between USB disconnection and  FWUPD_DEVICE_FLAG_WAIT_FOR_REPLUG flag.

chlee75, 2022-06-12:
I close this issue

closed 2022-06-12 by chlee75