fwupd/fwupd #4565: fwupdmgr get-devices no longer shows System Firmware, Intel Management Engine, or Embedded Controller. Lenovo T470s, X270

MountainX, 2022-04-27:
**Describe the bug**

`fwupdmgr get-devices` no longer shows System Firmware, Intel Management Engine, or Embedded Controller. I see this issue on multiple T470s and X270 devices.

**Steps to Reproduce**

On a T470s running Arch Linux, as root user:

    fwupdmgr get-devices
    fwupdmgr refresh
    fwupdmgr get-updates
    fwupdmgr update

Then review the list of updates / devices. Several important devices are missing, as noted above.

**Expected behavior**

Expected System Firmware, Intel Management Engine, and Embedded Controller to be listed in the system's updatable firmware devices, even if no current updates are available.

**fwupd version information**

```shell
# fwupdmgr --version
runtime   org.freedesktop.fwupd         1.7.7
compile   org.freedesktop.gusb          0.3.10
runtime   com.dell.libsmbios            2.4
runtime   org.kernel                    5.17.4-arch1-1
compile   com.hughsie.libjcat           0.1.10
compile   org.freedesktop.fwupd         1.7.7
runtime   org.freedesktop.gusb          0.3.10
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): pacman

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell

# Here's the output on this T470s up until a few weeks ago:

# fwupdmgr get-devices
20HGS6R400
│
├─Embedded Controller:
│     Device ID:          2292ae5236790b47884e37cf162dcf23bfcd1c60
│     Current version:    0.1.21
│     Minimum Version:    0.0.1
│     Vendor:             DMI:LENOVO
│     GUIDs:              e2acd8e4-a376-47bb-a316-4f1e62a8ca1f
│                         118d4b5a-8ef0-5dc6-b320-060dc79f2e36 ← UEFI\RES_{E2ACD8E4-A376-47BB-A316-4F1E62A8CA1F}
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─Intel Management Engine:
│     Device ID:          349bb341230b1a86e5effe7dfe4337e1590227bd
│     Current version:    184.77.3664
│     Minimum Version:    0.0.1
│     Vendor:             DMI:LENOVO
│     GUIDs:              e9124c4a-fdff-42e5-b2ad-f745db345953
│                         fa685a94-138c-531e-8bc2-5d61edfacba8 ← UEFI\RES_{E9124C4A-FDFF-42E5-B2AD-F745DB345953}
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
[... 201 lines cut ...]
  │     Vendor:           Intel Corporation (PCI:0x8086)
  │     GUID:             2c7b8bd2-3e77-5730-aaf6-596136f5254d ← IFD\NAME_BIOS
  │     Device Flags:     • Internal device
  │                       • Cryptographic hash verification is available
  │
  ├─Gigabit Ethernet:
  │     Device ID:        11188287b93230d58f85f059dfab93e1d59724bb
  │     Vendor:           Intel Corporation (PCI:0x8086)
  │     GUID:             30ee4fad-d1cd-522f-9a32-e53f4a4269cc ← IFD\NAME_GBE
  │     Device Flags:     • Internal device
  │                       • Cryptographic hash verification is available
  │
  └─Intel Management Engine:
        Device ID:        7ef8a531d2413174034556f12dff8aa3bb4a8c30
        Vendor:           Intel Corporation (PCI:0x8086)
        GUID:             614566e7-4d60-55f5-9cb9-48c5e488a705 ← IFD\NAME_ME
        Device Flags:     • Internal device
                          • Cryptographic hash verification is available
```

**Additional questions**

- Operating system and version:  Arch Linux
- Have you tried rebooting? Yes
- Is this a regression? I'm not sure

hughsie, 2022-04-28:
Can you attach the complete output of `sudo fwupdtool --plugins uefi-capsule get-devices -v` please.

MountainX, 2022-04-29:
> Can you attach the complete output of `sudo fwupdtool --plugins uefi-capsule get-devices -v` please.

Thank you. The output of that command allowed me to troubleshoot the issue. My uefi_capsule.conf had gotten misconfigured. Correcting it resolved my issue.

closed 2022-04-29 by MountainX

shuker85, 2022-06-16:
@MountainX hello what was your fix ?
I'm on t480 and having issues with not applying update for 
```
║ Upgrade Intel Management Engine from 184.86.3909 to 184.90.3987? 
```
[sudo fwupdtool --plugins uefi-capsule get-devices -v](https://github.com/fwupd/fwupd/files/8921675/fwupdtool.txt)

MountainX, 2022-06-21:
> @MountainX hello what was your fix ? I'm on t480 and having issues with not applying update for
> 
> ```
> ║ Upgrade Intel Management Engine from 184.86.3909 to 184.90.3987? 
> ```
> 

I don't have a fix for Intel Management Engine updates. This issue might be more relevant for you:

[Intel Management Engine fails on multiple ThinkPads · Issue #140 · fwupd/firmware-lenovo](https://github.com/fwupd/firmware-lenovo/issues/140#issuecomment-115867 "Intel Management Engine fails on multiple ThinkPads · Issue #140 · fwupd/firmware-lenovo")

The current issue, which I closed, was due to me having an incorrect path in one of the settings I had changed in uefi_capsule.conf.