fwupd/fwupd #4573: Thinkpad X1 Carbon Gen 9 Device ID: ae1abd099407b1d95698d69b7273f7fed5c6f35c Intel Management Engine 1.2.26.0 cant wake from sleep, turn off, massive issues

agoodm88, 2022-04-28:
**Describe the bug**
I attempted to complete 1.2.26.0 update on 25/04/22. Laptop rebooted and I went to bed after hearing fans speed up.  In the morning fans were still running at full speed while occasionally slowing down.  Laptop was roasting hot.   Unplugged/turned off/turned on update appeared to complete as usual.

Fast forward 2 days unplugged laptop. Laptop cant wake from sleep.  Cant turn off in this state.  Reset using hardware reset switch.  Backlight no longer working.  System freezes in POST.  On occasion system can be revived by repeatedly completing reset hole procedure.  Once booted system is stable.

**Steps to Reproduce**
Install 1.2.26.0 Intel Management Engine firmware update

**Expected behavior**
System should not be unusable

**fwupd version information**
alan@littlebrat:~$ fwupdmgr --version
runtime   org.freedesktop.fwupd         1.7.5
runtime   com.dell.libsmbios            2.4
compile   org.freedesktop.gusb          0.3.10
runtime   org.kernel                    5.17.0-051700-generic
compile   org.freedesktop.fwupd         1.7.5
runtime   org.freedesktop.gusb          0.3.10

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

From the distro

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
20XWCTO1WW
│
├─USB3.1 Hub:
│ │   Device ID:          7c5b6dadf8126e3820c86cb56d9e1777084a98a8
│ │   Summary:            USB 3.x hub
│ │   Current version:    13.24
│ │   Vendor:             VIA Labs, Inc. (USB:0x17EF)
│ │   Install Duration:   15 seconds
│ │   Update Error:       Device has been removed
│ │   GUIDs:              227d0f95-a764-598e-85de-3d2bd0a7372f
│ │                       e0c50668-5c2e-54df-8e17-c7c2472f3eaa
│ │                       c6381f16-fe6d-5c45-a4ad-b619af4f0f0e
│ │                       fc6cfb35-d4e9-5467-a014-8863de104969
│ │                       7ae3100f-9b01-571c-9fc2-20cd4544123a
│ │                       ef65710a-3fa0-5330-9a61-5ced5e976be1
│ │   Device Flags:       • Updatable
│ │                       • Cryptographic hash verification is available
│ │                       • Device stages updates
│ │                       • Device can recover flash failures
│ │ 
│ ├─USB2.0 Hub:
│ │ │   Device ID:        65a54fb6ce182f0e75edf0e43047d547a0d61f0e
│ │ │   Summary:          USB 3.x hub
│ │ │   Current version:  13.24
│ │ │   Vendor:           VIA Labs, Inc. (USB:0x17EF)
│ │ │   Install Duration: 15 seconds
│ │ │   GUIDs:            010262bc-c6fa-5843-bffd-42be9aa16e6a
│ │ │                     227d0f95-a764-598e-85de-3d2bd0a7372f
│ │ │                     4ec36768-1858-5e9b-9d35-40e6143c3cd4
│ │ │                     b8ce86b1-5d74-5876-8d02-a9fd61b0a9ee
│ │ │                     181168b9-f5c8-51c4-a8f2-958daab8af2d
│ │ │   Device Flags:     • Updatable
│ │ │                     • Supported on remote server
│ │ │                     • Cryptographic hash verification is available
│ │ │                     • Device stages updates
│ │ │                     • Device can recover flash failures
│ │ │ 
│ │ └─USB2.0 Hub:
│ │   │   Device ID:      7a2034b0a5bfbc758ec5bfdce77b0484bdc0b46f
│ │   │   Summary:        USB 3.x hub
│ │   │   Current version:13.23
│ │   │   Vendor:         VIA Labs, Inc. (USB:0x17EF)
[... 729 lines cut ...]
├── initrd.img-5.15.0-051500rc6-generic
├── initrd.img-5.17.0-051700-generic
├── initrd.img.old -> initrd.img-5.15.0-051500rc6-generic
├── lost+found  [error opening dir]
├── memtest86+.bin
├── memtest86+.elf
├── memtest86+_multiboot.bin
├── System.map-5.15.0-051500rc6-generic
├── System.map-5.17.0-051700-generic
├── vmlinuz -> vmlinuz-5.17.0-051700-generic
├── vmlinuz-5.15.0-051500rc6-generic
├── vmlinuz-5.17.0-051700-generic
└── vmlinuz.old -> vmlinuz-5.15.0-051500rc6-generic

6 directories, 302 files
```

**Additional questions**

- Operating system and version:
- Have you tried rebooting? - **duh**
- Is this a regression? **duh**
- Are you using an NVMe disk? **yes**
- Is secure boot enabled? **no**
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? **dunno**

agoodm88, 2022-04-28:
Thankfully I have premier support option.  This appears to be a known issue.  Lenovo are replacing my motherboard because of another issue.  If this doesnt happen tomorrow I will be attempting to reflash this 1.22.26 update somehow; apparently this can sometimes fix the issues.

mrhpearson, 2022-04-29:
Hi @agoodm88 

Did they say if the issue was with the HW or with the FW update? I hadn't heard of this particular issue - and if it's with the FW update then we need to pull it before others hit the problem...

Mark

agoodm88, 2022-04-29:
alan@littlebrat:~$ uname -r
5.17.0-051700-generic

superm1, 2022-04-29:
> alan@littlebrat:~$ uname -r
> 5.17.0-051700-generic

Okay no that won't be affected. Thanks for checking.

closed 2022-07-31 by stale[bot]