fwupd/fwupd #4578: [1.8.0] `libsqlite3-0.dll` missing from Windows installer

cwsystem76, 2022-04-30:
**Describe the bug**
The Windows build of `fwupd` version 1,8.0 does not start because it's missing the `libsqlite3-0.dll` library.

**Steps to Reproduce**
Install `fwupd` v1.8.0 for Windows from the release installer (https://github.com/fwupd/fwupd/releases/download/1.8.0/fwupd-1.8.0-setup-x86_64.exe) and run `"C:\Program Files (x86)\fwupd\bin\fwupdtool.exe"`. If run from Powershell in Windows Terminal, nothing happens. If run from `cmd.exe` then an error message pops up:

[image]

The file is not present in the `fwupd\bin` folder. It wasn't included with v1.7.7 either, so this appears to be a new dependency for v1.8.0 that does not have the appropriate file included in the installer. I am hesitant, for obvious reasons, to go looking for a copy of a specific DLL online, and this particular library is not part of the regular [sqlite distribution files for Windows](https://sqlite.org/download.html), so I can't test what happens if I add it to the `bin` folder.

I suspect that this library was available system-wide on the Windows machine used to test the build, but it's not part of a default Windows install.

**Expected behavior**
`fwupdtool` should run as expected.

**fwupd version information**
It's a Windows machine, no daemon. fwupdtool is version 1.8.0, installed from the downloadable fwupd-1.8.0-setup-x86_64.exe file.

**fwupd device information**
It does not get that far.

**Additional questions**

- Operating system and version:

Windows 10 21H2

- Have you tried rebooting?

Yes.

- Is this a regression?

I don't think so.

beautifulcoder, 2022-05-01:
I can confirm this same behavior with the latest v1.8.0 release. The previous v1.7.7 release does not have this problem.

superm1, 2022-05-01:
I bet this is a result of moving up to fedora 36 for the base.

hughsie, 2022-05-02:
> I bet this is a result of moving up to fedora 36 for the base.

My bet is that we removed the -Dsqlite=false when we added the autodeps stuff :) PR incoming.

closed 2022-05-02 by hughsie

cwsystem76, 2022-05-05:
@hughsie I was trying to upgrade the Launch with `fwupdtool` v1.7.7 on a Windows system, just for testing purposes because not all of our Launch customers are Linux users and having instructions to give Windows folks might be helpful, and I ran into that problem with `kenv` (https://github.com/fwupd/fwupd/issues/4516). Since that's fixed in v1.8.0, I was trying it again, and ran into the DLL thing. When v1.8.1 is released I will try again, and see if I can make it through a full firmware flash on the Launch, and if I can't, I'll try to figure out why and open another issue with the appropriate details.

As for ease of use, I can't really think of anything other than wrapping a GUI of some kind around `fwupdtool` for Windows users, since there's not going to be a system daemon. The CLI tool is a bit complex but it kind of has to be, since it does so many things. Maybe an interactive text UI would be useful? That would be easier to make work on both Windows and Linux, and could turn out to also be helpful for headless systems, remote administration and support, etc. Something along these lines is what I'm thinking about, a quick mockup based on interactive tools like `gdisk`:
```
# fwupd-interactive
fwupd v1.8.0 - system daemon not present
Command (? for help): list devices

[1] Launch Configurable Keyboard (launch 1) : current firmware version 0.7.103-4011-g708ed5 : updateable : no new updates available
[2] Patriot M.2 P300: current firmware version EDFM20.0 : updateable : not in LVFS

Command (? for help): show 1

There's been some informal talk of adding `fwupdtool` to our Windows build of the System76 Keyboard Configurator (https://github.com/pop-os/keyboard-configurator/) but that would be a _long_ way off, would be somewhat low priority, and would require a lot more testing than just one support guy messing around with a Launch on his gaming and home theater PC.

I _have_ noticed that the current Win32 build doesn't have an uninstaller, so I've just been manually deleting the `fwupd` folder instead. That would be a relatively simple quality of life thing for all Win32 users, since I think it's just a flag you can set when building the Windows installer?

superm1, 2022-05-05:
> We did ponder having a system daemon and just doing point-to-point DBus rather than relying on the bus. If we did p2p you oculd even use something like gnome-firmware in Windows(!).

"Windows service" I think is what you call it.  It does seem that dbus does run on Windows, maybe this is actually viable with point to point.

```
The Windows port from the windbus and dbus4win projects was merged into the freedesktop dbus development branch several years ago, and is released as part of dbus. Thanks to all the past and current contributors to that port.
It is known that the Windows port runs on Windows Vista, 8, 10 and server variants. Supported compilers/SDK are MSVC, 
mingw-w32/w64(gcc) and cygwin(gcc).
```

> That's not insane; maybe you can open a discussion and we can discuss there? Certainly that would be useful when dealing with redfish in fwupdtool -- as enumeration can take a couple of minutes.

I would argue that shouldn't go into fwupdtool, but fwupdmgr and we do this PtP thing instead.  Otherwise aren't you going to need to start threading?

hughsie, 2022-05-05:
> "Windows service" I think is what you call it

Ya, this is my problem -- I know very little about win32...

>  Otherwise aren't you going to need to start threading

Ohh, all these actions would be synchronous, otherwise it all gets a bit mad.

cwsystem76, 2022-05-05:
I'm generally hesitant to do testing on CI builds but I'm grabbing that one just to check the installer.

Okay, so `fwupd-1.8.1-setup-x86_64.exe` installs but does not add an entry to the list of installed programs in Windows 10, either in the original control panel "Uninstall or change a program" section or the new-style "Apps & features" list. It _does_ create a `Uninstall fwupd.exe` file in the install directory (`C:\Program Files (x86)\fwupd` in this case), but without the entry in the applications listings most Windows users won't even know it's there.

More importantly, something is wrong with the uninstaller. Running `Uninstall fwupd.exe` manually pops up a "This app can't run on your PC" error, similar to what happens if you try to run an ARM executable on an AMD64 system or vise versa. Running it from the command line results in this:
```
PS C:\Program Files (x86)\fwupd> & '.\Uninstall fwupd.exe'
ResourceUnavailable: Program 'Uninstall fwupd.exe' failed to run: An error occurred trying to start process 'C:\Program Files (x86)\fwupd\Uninstall fwupd.exe' with working directory 'C:\Program Files (x86)\fwupd'. The specified executable is not a valid application for this OS platform.At line:1 char:1
+ & '.\Uninstall fwupd.exe'
+ ~~~~~~~~~~~~~~~~~~~~~~~~~.
```
This is why I'd been manually deleting the `C:\Program Files (x86)\fwupd` folder, for some reason I was remembering it didn't have an uninstaller; it actually does, but it doesn't work. I'm going to open a new issue for this, if you think that's appropriate?

Re: porting `fwupdmgr` and the daemon to Windows, if that's something you guys think can be done without too many issues, that's your call. My thought was that an interactive text-mode tool would be able to call _either_ the daemon or `fwupdtool` depending on the system and what services are present, but this is definitely a software infrastructure decision.

hughsie, 2022-05-05:
> but does not add an entry to the list of installed programs in Windows 10

Does https://github.com/fwupd/fwupd/pull/4589 help? It makes it show up in `wine uninstaller` for me.

> This app can't run on your PC

No idea on that. Seems to work in wine, and can try real Windows later.

> either the daemon or fwupdtool

I'm hesitant to teach fwupdtool many more tricks, and I'm also hesitant adding in multiple processes, IPC and sockets into the mix. What I really want is someone to drive this work for me, someone who needs to use this in production. Hint hint.

cwsystem76, 2022-05-05:
That does add it to the list, yes:
[image]
However the uninstaller executable (now named `uninst.exe`) still does not work:
```
PS C:\Program Files (x86)\fwupd> .\uninst.exe
ResourceUnavailable: Program 'uninst.exe' failed to run: An error occurred trying to start process 'C:\Program Files (x86)\fwupd\uninst.exe' with working directory 'C:\Program Files (x86)\fwupd'. The specified executable is not a valid application for this OS platform.At line:1 char:1
+ .\uninst.exe
+ ~~~~~~~~~~~~.
```
This is literally the first time I've seen this error message on this PC so I'm pretty sure it's not a problem on my end. I can open up a new issue if you want.

And I'm going to be very up-front here re: win32 ports, we are absolutely not a Windows shop at all. I'm surprised we even have the Configurator for Windows and macOS, to be honest, given our focus on Linux development. If we have a working `fwupdtool` command line version for win32, having our Windows version of the Configurator bundle it with the installer and then call it for firmware updates is about as far as we're likely to go, and we might not even end up doing that.

My thought was about a completely separate text-based interactive tool (similar to `bluetoothctl` or `gdisk`), that would call out to the existing `fwupdtool` _or_ the daemon, similar to how `fwupdmgr` or GNOME Firmware calls out to the daemon, not adding any new functionality to `fwupdtool` itself. Something like that would be somewhat easier to maintain than a GUI like GNOME Firmware, and would run on a wider variety of platforms. Obviously it wouldn't be a trivial piece of development but you asked what would make things easier for Windows users, and that would seem to me to be a good middle ground between Windows people using `fwupdtool` alone or porting GNOME Firmware to Windows, and it wouldn't be work limited to the Windows platform either. If that doesn't fit in with your goals or resources, that's absolutely fine, it was just an idea.

hughsie, 2022-05-05:
> At line:1 char:1

Is it a script?! Can you upload it somewhere from a real windows machine? Built locally it's a `PE32 executable (GUI) Intel 80386 (stripped to external PDB), for MS Windows`

cwsystem76, 2022-05-05:
No, that's just output from PowerShell, instead of popping up the blue box with "This app can't run on your PC", which is what happens when you run it from `cmd`. It is definitely a binary file, running `certutil -dump uninst.exe` on it produces the following (and a lot more):
```
    000000  4d 5a 90 00 03 00 00 00  04 00 00 00 ff ff 00 00   MZ..............
    000010  b8 00 00 00 00 00 00 00  40 00 00 00 00 00 00 00   ........@.......
    000020  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   ................
    000030  00 00 00 00 00 00 00 00  00 00 00 00 80 00 00 00   ................
    000040  0e 1f ba 0e 00 b4 09 cd  21 b8 01 4c cd 21 54 68   ........!..L.!Th
    000050  69 73 20 70 72 6f 67 72  61 6d 20 63 61 6e 6e 6f   is program canno
    000060  74 20 62 65 20 72 75 6e  20 69 6e 20 44 4f 53 20   t be run in DOS
    000070  6d 6f 64 65 2e 0d 0d 0a  24 00 00 00 00 00 00 00   mode....$.......
    000080  50 45 00 00 4c 01 08 00  00 a6 e8 61 00 00 00 00   PE..L......a....
    000090  00 00 00 00 e0 00 0e 03  0b 01 02 25 00 a0 00 00   ...........%....
```
VirusTotal says it's a valid EXE (and hilariously one of the AV vendors thinks it's malicious: https://www.virustotal.com/gui/file/65423203713cfdbbda1ea6320a5848daeecc7ab52c21ee72a164c2dd90d135a8/details) and Sysinternals `sigcheck` says it's a 32-bit Windows app, so I have no idea why it's not loading since I have piles of other 32-bit stuff that runs fine. I only have the one Windows machine right now to test with, though.

Here's the `uninst.exe` as it ends up after running the installer:
[uninst.zip](https://github.com/fwupd/fwupd/files/8633302/uninst.zip)

hughsie, 2022-05-05:
> 32-bit Windows app

Hmm, I wonder if it's because we're using nsis in 32 bit mode, and everything else in 64 bit mode. Will investigate why there's no mingw64-nsis in Fedora...

hughsie, 2022-05-05:
Please.

cwsystem76, 2022-05-05:
Done: https://github.com/fwupd/fwupd/issues/4591