fwupd/fwupd #4583: Reverse Engineering the IK Uno Drum update method - appears to be 'DFU'

mungewell, 2022-05-04:
For a while I have been working on a project to reverse engineer the IK Multimedia UNO Drum.
https://github.com/mungewell/uno-drum-utils

I have the device well understood for normal operation, and have also figured out how IK encode their 'drum packs' - thus allowing users to upload their own samples to the device... not something that IK official supports. However this last part is reliant on IK's tools and since they ONLY allow registered users to access these, thus it is impossible for 2nd hand users to control the hardware they own.

I would like to understand if it is possible to use FWUPD to perform the upload, which is done over USB DFU.

Does this, or the info below, make any sense to trained eye? Is this somewhat similar to how FWUPD works on some devices??
Thanks for any help you can give.

There is a 'magic midi sysex packet' which places the hardware into updater mode.
```
$ hexdump -C UNO_Drum_Update_Mode.syx 
00000000  f0 00 21 1a 02 02 11 00  42 4f 4f 54 20 55 4e 4f  |..!.....BOOT UNO|
00000010  00 f7                                             |..|
00000012
```

At which point it reboots and presents on USB as:
[updater_lsusb_vv.txt](https://github.com/fwupd/fwupd/files/8617887/updater_lsusb_vv.txt)

The file which will be uploaded (for audio samples) starts:
```
$ hexdump -C Uno_Drum_lib.dfu | head
00000000  44 66 75 53 65 01 e9 ab  07 00 01 54 61 72 67 65  |DfuSe......Targe|
00000010  74 01 01 00 00 00 50 43  4d 20 4c 69 62 72 61 72  |t.....PCM Librar|
00000020  79 20 00 db 00 c9 c0 1e  0c d0 00 01 00 00 8e 00  |y ..............|
00000030  00 00 31 53 50 53 a6 6a  63 28 3d 95 d2 11 b5 d6  |..1SPS.jc(=.....|
00000040  00 c0 4f d9 18 d0 61 00  00 00 20 00 00 00 00 11  |..O...a... .....|
00000050  10 00 00 50 00 00 00 14  00 1f 50 e0 4f d0 20 ea  |...P......P.O. .|
00000060  3a 69 10 a2 d8 08 00 2b  30 30 9d 3a 00 2e 80 1c  |:i.....+00.:....|
00000070  ff cd a8 78 48 be 43 b5  fd f8 09 1c 1c 60 d0 26  |...xH.C......`.&|
00000080  00 01 00 26 00 ef be 11  00 00 00 b1 a7 44 ff b5  |...&.........D..|
00000090  f5 d2 01 99 fe 47 67 25  cd d4 01 99 fe 47 67 25  |.....Gg%.....Gg%|
```

The contents of the (official) installer directory has the following files:
```
$ ls
'UNO Drum Sound Content Installer.exe'   dfu-prefix.exe   dfu-util-static.exe   libusb-1.0.dll
 Uno_Drum_lib.dfu                        dfu-suffix.exe   dfu-util.exe
```

mungewell, 2022-05-04:
There is a video tutorial on how to build the custom sound packs here:
https://www.youtube.com/watch?v=YkdraaaeCWo

hughsie, 2022-05-04:
> Is this somewhat similar to how FWUPD works on some devices

Yes, this is a bit like how the first generation Jabra devices work. e.g. we have to send a "_go to appIDLE_" packet which switches them into programmer mode. e.g. this code: https://github.com/fwupd/fwupd/blob/main/plugins/jabra/fu-jabra-device.c#L98

Then when in the DFU mode we can upload firmware using `fwupdtool --plugins dfu install-blob firmware.bin` -- I'm almost certain this will work for fwupd too, as the DfuSe header looks pretty normal. e.g. try `fwupdtool firmware-parse Uno_Drum_lib.dfu dfuse`. You might also need a quirk for https://github.com/fwupd/fwupd/blob/main/plugins/dfu/dfu.quirk before fwupd works properly.

Maybe that helps?

mungewell, 2022-05-05:
Thank you for the suggestions. My laptop is running an older dist with an outdated 'fwupd', but I was able to confirm the file with a bootable USB dist.

```
# fwupdtool firmware-parse  /cdrom/Uno_Drum_lib.dfu dfuse
Loading                 [***************************************]
FuDfuseFirmware:
Flags:                  has-checksum|has-vid-pid
Vid:                    0x1963
Pid:                    0x48
Release:                0x3
Version:                0x11a
  FuFirmwareImage:
  ID:                   PCM Library
  Index:                0x1
    FuChunk:
      Index:            0x0
      Page:             0x0
      Address:          0x8040000
      DataSz:           0x7aac4
```

With this, and the USB log I'll note the `Version : 1.1a`. I saw note that this likely is a ST variation of DFU, which (apparently) is quite different from normal DFU.

For the quirks I did see one mentioned against an ST part `DfuForceVersion = 011a`, so maybe this would/might be needed. As for others??? The device will already be in DFU mode (from Midi SysEx), is there something I need to `quirk` to tell the system that?

I should be clear that (probably) only myself and maybe 5 other people on earth are interested in getting this going, so there really isn't a point patching this into  fwupd.... I am just figuring out a manual method.

hughsie, 2022-05-05:
> I saw note that this likely is a ST variation of DFU

Yup, different, but not unusual.

> is there something I need to quirk to tell the system that

No, fwupd should automatically "probe" anything with a DFU descriptor automatically.

> isn't a point patching this into fwupd

If you did want to add a plugin I wouldn't be against it :) It'll mean that millions of people are able to just switch the device into DFU mode without compiling and installing anything.

mungewell, 2022-05-06:
For those following along... I have had success, but using a different tool (`dfu-util` on Ubuntu).
[installed.txt](https://github.com/fwupd/fwupd/files/8637477/installed.txt)

I couldn't see a way to register the device's VID:PID with fwupd, and the suggested `fwupdtool --plugins dfu install-blob empty.dfu` didn't find the device.

HOWEVER, I am now thinking that this may be related to the PID embedded in the file. After moving to try `dfu-util` I realized that the file PID (as from official files) did not match the device when it was in boot loader mode. It seems that the official tool does not care about this....

_I'll have to have another go with `fwupdtool` tomorrow._

hughsie, 2022-05-09:
What does `fwupdtool --plugins dfu get-devices --verbose` show when the device is in DFU mode?

mungewell, 2022-05-10:
I tried with the DFU with PID set as 0x0049, but still reported that it couldn't ind the device.

Full log of 'get-devices' here:
[get_devices.txt](https://github.com/fwupd/fwupd/files/8657133/get_devices.txt)

hughsie, 2022-05-10:
> DFU version 0x0000 invalid, v1.1 assumed

This is fishy, but I think it's okay.

> using 70835e1192ddb447c964de775916a4cd505ec3b3 for usb:03:00:01:02

So this means fwupd found a device, but it wasn't marked as upgradable.

Can you do `fwupdtool --plugins dfu get-devices --verbose --show-all` to confirm that?

mungewell, 2022-05-11:
I still get the same thing....
[lsusb_vv.txt](https://github.com/fwupd/fwupd/files/8665704/lsusb_vv.txt)
[get_devices_all.txt](https://github.com/fwupd/fwupd/files/8665705/get_devices_all.txt)

I'm not convinced that "usb:03:00:01:02" is the IK device, maybe something else in the system.

I'll stress that I have not done **anything** to adjust fwupd, I am just booting a live CD, then switching the device into Bootloader mode with a 'magic' Midi/SysEx packet.

Also thank you for you help. I don't want to take too much of your time, if this is going to be a deep problem we can 'forget it'., As mentioned I was able to get 'dfu-util' to flash the device....

hughsie, 2022-05-11:
> usb:03:00:01:02

This looks to be the Genesys hub I think.

Are you able to try fwupd 1.7.x or 1.8.x? 1.5.11 is super old.

closed 2022-07-31 by stale[bot]