fwupd/fwupd #4591: [1.8.0] Windows uninstaller does not run

cwsystem76, 2022-05-05:
**Describe the bug**
The Windows build of fwupd version 1,8.0 creates an uninstaller (renamed `uninst.exe` viaPIR https://github.com/fwupd/fwupd/pull/4589), but it does not actually work. Running it from PowerShell results in:
```
PS C:\Program Files (x86)\fwupd> .\uninst.exe
ResourceUnavailable: Program 'uninst.exe' failed to run: An error occurred trying to start process 'C:\Program Files (x86)\fwupd\uninst.exe' with working directory 'C:\Program Files (x86)\fwupd'. The specified executable is not a valid application for this OS platform.At line:1 char:1
+ .\uninst.exe
+ ~~~~~~~~~~~~.
```
Running it from the Explorer shell or via `cmd.exe` results in this:

[image]

This error is typical when there's a mismatch between the system architecture and the executable, but everything says that this is a standard 32-bit Windows EXE file (including [VirusTotal](https://www.virustotal.com/gui/file/65423203713cfdbbda1ea6320a5848daeecc7ab52c21ee72a164c2dd90d135a8/details)), and other 32-bit Windows apps run fine on this same system.

This was originally brought up in closed issue https://github.com/fwupd/fwupd/issues/4578 where the thought is that this has something to do with using a 32-bit installer while the rest of the build is 64-bit.

**Steps to Reproduce**
Install from one of the amd64 installers (like [this one](https://app.circleci.com/pipelines/github/fwupd/fwupd/9865/workflows/ca5c5f7e-5727-4445-a725-6958691d87fa/jobs/35759/artifacts) from the CI sytstem) and try to run the `C:\Program Files (x86)\fwupd\uninst.exe` executable.

**Expected behavior**
It should run the uninstaller.

**fwupd version information**
It's a Windows machine, no daemon. fwupdtool is version `1.8.1` (actually `1.8.0` with a PR), installed from the downloadable [fwupd-1.8.1-setup-x86_64.exe](https://output.circle-artifacts.com/output/job/c39e776d-0b4b-46fe-910e-1fc09afd2330/artifacts/0/dist/setup/fwupd-1.8.1-setup-x86_64.exe) file.

**fwupd device information**
n/a

**Additional questions**

- Operating system and version:

Windows 10 21H2

- Have you tried rebooting?

Yes.

- Is this a regression?

I don't think it's ever worked.

cwsystem76, 2022-05-07:
Looks like this will be the fix: [Switch the Windows installer from NSIS to MSI](https://github.com/fwupd/fwupd/pull/4597)

superm1, 2022-05-09:
I still don't see the CI binary running unfortunately.
[image]

Copying in the missing DLL, then I see

[image]

Which I'm sure that was working before, maybe the move to Fedora36 and the new way the CA bundle was built is causing it.

hughsie, 2022-05-10:
> Which I'm sure that was working before

It used to be as easy as copying `ca-bundle.crt` into the bindir. Can you strace and see where fwupdtool is looking to load the bundle from?

> the new way the CA bundle was built

Oh, this changed? Got any links?

hughsie, 2022-05-10:
In wine I get this:

    08:36:45:0886 FuCommon             running 'curl.exe https://cdn.fwupd.org/downloads/firmware.xml.gz.jcat --output C:\users\hughsie\AppData\Local\M
icrosoft\Windows\INetCache\fwupd\firmware.xml.gz.jcat --cacert Z:\home\hughsie\Code\fwupd-win32\build-win32\src\dist\bin\..\bin\ca-bundle.crt'

i.e. we specified the ca-cert manually (and it exists) -- I wonder if curl.exe in windows != curl.exe in linux...

cwsystem76, 2022-05-28:
This problem appears to have been fixed with the switch to a different installer mechanism for the v1.8.1 release. There's no longer an `uninst.exe`  (since it's no longer using NSIS) and the uninstall process now works as expected. I'll leave it up to the devs to close this if there's nothing else they need this issue open for.

closed 2022-05-28 by superm1