fwupd/fwupd #4611: [Colorhug] failed to wait for detach replug

guilhem, 2022-05-10:
**Describe the bug**

Fail to update colorhug from 1.2.3 to 1.2.6

<details>

```shell
❯ sudo fwupdtool --plugins colorhug --verbose update
11:13:16:0468 FuDebug              Verbose debugging enabled (on console 1)
11:13:16:0468 FuMain               locked /run/lock/fwupdtool
11:13:16:0470 FuMain               Failed to stop daemon: failed to find fwupd.service: GDBus.Error:org.freedesktop.systemd1.NoSuchUnit: Unit fwupd.service not loaded.
11:13:16:0470 FuConfig             trying to load config values from /etc/fwupd/daemon.conf
11:13:16:0470 FuCommon             reading /etc/fwupd/daemon.conf with 1931 bytes
11:13:16:0470 FuConfig             trying to load config values from /var/etc/fwupd/daemon.conf
11:13:16:0470 FuConfig             using autodetected max archive size 18,4 EB
11:13:16:0471 GLib-GIO             _g_io_module_get_default: Found default implementation local (GLocalVfs) for ‘gio-vfs’
11:13:16:0471 FuRemoteList         loading /usr/share/fwupd/metainfo
11:13:16:0471 XbSilo               attempting to load /var/cache/fwupd/metainfo.xmlb
11:13:16:0471 XbSilo               file: 7e264eaa-9fee-5a2c-6d45-db448384846b, current:7e264eaa-9fee-5a2c-6d45-db448384846b, cached: (null)
11:13:16:0471 XbSilo               loading silo with file contents
11:13:16:0471 FuRemoteList         loading remote from /etc/fwupd/remotes.d/dell-esrt.conf
11:13:16:0471 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor-directory.conf
11:13:16:0471 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor.conf
11:13:16:0471 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs.conf
11:13:16:0471 FuRemoteList         skipping invalid file /etc/fwupd/remotes.d/lvfs-testing.conf.dpkg-old
11:13:16:0471 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs-testing.conf
11:13:16:0472 FuRemoteList         skipping invalid file /var/lib/fwupd/remotes.d/lvfs
11:13:16:0472 FuRemoteList         ignoring unfound remote fwupd
11:13:16:0472 FuRemoteList         ordering lvfs-testing=lvfs+1
11:13:16:0472 Jcat                 ignoring GPG-KEY-Hughski-Limited as not PKCS-7 certificate
11:13:16:0472 Jcat                 ignoring GPG-KEY-Linux-Foundation-Firmware as not PKCS-7 certificate
11:13:16:0472 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
11:13:16:0472 Jcat                 trying to load certificate from /etc/pki/fwupd/LVFS-CA.pem
11:13:16:0472 Jcat                 reading /etc/pki/fwupd/LVFS-CA.pem with 1679 bytes
11:13:16:0472 Jcat                 loaded 1 certificates
11:13:16:0472 Jcat                 ignoring GPG-KEY-Linux-Foundation-Metadata as not PKCS-7 certificate
11:13:16:0472 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
11:13:16:0472 Jcat                 trying to load certificate from /etc/pki/fwupd-metadata/LVFS-CA.pem
11:13:16:0472 Jcat                 reading /etc/pki/fwupd-metadata/LVFS-CA.pem with 1679 bytes
11:13:16:0472 Jcat                 loaded 1 certificates
11:13:16:0472 Jcat                 reading /var/lib/fwupd/pki/secret.key with 2459 bytes
11:13:16:0472 Jcat                 reading /var/lib/fwupd/pki/client.pem with 1383 bytes
11:13:16:0472 FuEngine             client certificate now exists: nothing to do
11:13:16:0472 FuHistory            trying to open database '/var/lib/fwupd/pending.db'
11:13:16:0472 FuHistory            got schema version of 7
11:13:16:0472 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d
11:13:16:0476 FuQuirks             loading quirks from /var/lib/fwupd/quirks.d
11:13:16:0476 XbSilo               attempting to load /var/cache/fwupd/quirks.xmlb
11:13:16:0476 XbSilo               file: 20fb6a7b-d84f-8916-6070-be4b061499ad, current:20fb6a7b-d84f-8916-6070-be4b061499ad, cached: (null)
11:13:16:0476 XbSilo               loading silo with file contents
11:13:16:0476 FuSmbios             trying to read /sys/class/dmi/id
11:13:16:0476 FuHwids              loading HwId overrides from /etc/fwupd/daemon.conf
11:13:16:0476 FuHwids              not loading HwId overrides from /var/lib/fwupd/daemon.conf
11:13:16:0476 FuHwids              smbios property Manufacturer=Dell Inc.
11:13:16:0476 FuHwids              smbios property EnclosureKind=1f
11:13:16:0476 FuHwids              smbios property Family=XPS
11:13:16:0476 FuHwids              smbios property ProductName=XPS 13 9310 2-in-1
11:13:16:0476 FuHwids              smbios property ProductSku=09FF
11:13:16:0476 FuHwids              smbios property BiosVendor=Dell Inc.
11:13:16:0476 FuHwids              smbios property BiosVersion=2.9.0
11:13:16:0476 FuHwids              smbios property BiosMajorRelease=02
11:13:16:0476 FuHwids              smbios property BiosMinorRelease=09
11:13:16:0476 FuHwids              smbios property FirmwareMajorRelease=ff
11:13:16:0476 FuHwids              smbios property FirmwareMinorRelease=ff
11:13:16:0476 FuHwids              smbios property BaseboardManufacturer=Dell Inc.
11:13:16:0476 FuHwids              smbios property BaseboardProduct=0PH1FK
11:13:16:0480 FuEngine             ignoring: Error opening directory “/var/lib/fwupd/local.d”: No such file or directory
11:13:16:0480 FuEngine             ignoring: Error opening directory “/usr/share/fwupd/local.d”: No such file or directory
11:13:16:0480 XbSilo               attempting to load /var/cache/fwupd/metadata.xmlb
11:13:16:0480 XbSilo               file: 0de7a6a9-6cf4-0cf2-099d-4f213d49c392, current:0de7a6a9-6cf4-0cf2-099d-4f213d49c392, cached: (null)
[... 381 lines cut ...]
│                         • Updatable
│   
├─Unifying Receiver:
│     Device ID:          26a3ab629ab8dab530d4e5cd7d18aa8ac60a8ac5
│     Summary:            Miniaturised USB wireless receiver
│     Current version:    RQR24.11_B0036
│     Bootloader Version: BOT03.02_B0009
│     Vendor:             HIDRAW:0x046D|USB:0x046D
│     Install Duration:   30 seconds
│     GUIDs:              cc4cbfa9-bf9d-540b-b92b-172ce31013c1
│                         fcf55bf5-767b-51ce-9c17-f6f538c4ee9f
│                         279ed287-3607-549e-bacc-f873bb9838c4
│     Device Flags:       • Updatable
│   
└─Wacom AES Device:
      Device ID:          dbbf40284cb602275c039924b932efc748233670
      Current version:    3006.00
      Vendor:             HIDRAW:0x056A
      GUIDs:              f2e31e0f-ab3b-5cd2-8a2f-462ada7ed50a
                          5f02cd21-7586-5650-95e9-388fdf0b6e63
                          a6531f36-4cd5-5878-a8f4-9efc1b7702b3
      Device Flags:       • Internal device
                          • Updatable
```
</details>

hughsie, 2022-05-10:
I can't reproduce with 1.8.1:

    [hughsie@fedora build (main %)]$ sudo ./src/fwupdtool --plugins colorhug --verbose update
    11:26:10:0083 FuDebug              Verbose debugging enabled (on console 1)
    ...
    11:26:20:0277 FuEngine             Updating ColorHug2 took 5.978851 seconds

> Device is blinking red / green.

This means it's in bootloader -- we asked it to detach and waited up to 10 seconds for it to come back. Is the runtime mode listed in `lsusb`? if you do `fwupdtool attach` does it go directly to runtime? Can you reproduce this behaviour every time?

hughsie, 2022-05-10:
> ran update

Can you try `fwupdmgr update` instead? `fwupdtool` is only really designed as a power user tool.

joakim-tjernlund, 2022-05-11:
I recall seeing the msg:
```
failed to get device before update: failed to wait for detach replug: device 1bcca71b3480ec5662e5da8a742c4c805275ffbf did not come back
```
when trying to update a Bluetooth Mouse FW (or FW to its USB to BT plug)

hughsie, 2022-05-11:
@guilhem ahh, I was testing with a ColorHug2! Does https://github.com/fwupd/fwupd/pull/4616 fix the failure for you?

guilhem, 2022-05-11:
> @guilhem ahh, I was testing with a ColorHug2! Does #4616 fix the failure for you?

@hughsie same behavior :/

<details>

```shell
~/sources/fwupd wip/hughsie/colorhug
❯ sudo ./build/src/fwupdtool --plugins colorhug --verbose update
10:19:16:0251 FuDebug              Verbose debugging enabled (on console 1)
10:19:16:0252 FuMain               locked /run/lock/fwupdtool
10:19:16:0255 FuMain               Failed to stop daemon: failed to find fwupd.service: GDBus.Error:org.freedesktop.systemd1.NoSuchUnit: Unit fwupd.service not loaded.
10:19:16:0255 FuConfig             trying to load config values from /usr/local/etc/fwupd/daemon.conf
10:19:16:0255 FuCommon             reading /usr/local/etc/fwupd/daemon.conf with 2080 bytes
10:19:16:0255 FuConfig             trying to load config values from /var/local/etc/fwupd/daemon.conf
10:19:16:0255 FuConfig             using autodetected max archive size 18,4 EB
10:19:16:0256 GLib-GIO             _g_io_module_get_default: Found default implementation local (GLocalVfs) for ‘gio-vfs’
10:19:16:0256 FuRemoteList         loading /usr/local/share/fwupd/metainfo
10:19:16:0256 XbSilo               attempting to load /var/local/cache/fwupd/metainfo.xmlb
10:19:16:0256 XbSilo               file: bca6e538-90b2-fa1e-736e-9a2c313b60b1, current:bca6e538-90b2-fa1e-736e-9a2c313b60b1, cached: (null)
10:19:16:0256 XbSilo               loading silo with file contents
10:19:16:0256 FuRemoteList         loading remote from /usr/local/etc/fwupd/remotes.d/lvfs-testing.conf
10:19:16:0256 FuRemoteList         loading remote from /usr/local/etc/fwupd/remotes.d/lvfs.conf
10:19:16:0256 FuRemoteList         loading remote from /usr/local/etc/fwupd/remotes.d/vendor.conf
10:19:16:0256 FuRemoteList         loading remote from /usr/local/etc/fwupd/remotes.d/vendor-directory.conf
10:19:16:0256 FuRemoteList         loading remote from /usr/local/etc/fwupd/remotes.d/fwupd-tests.conf
10:19:16:0256 FuRemoteList         ordering lvfs-testing=lvfs+1
10:19:16:0256 FuRemoteList         ignoring unfound remote fwupd
10:19:16:0256 Jcat                 ignoring GPG-KEY-Linux-Foundation-Firmware as not PKCS-7 certificate
10:19:16:0256 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
10:19:16:0256 Jcat                 reading /usr/local/etc/pki/fwupd/LVFS-CA.pem with 1679 bytes
10:19:16:0256 Jcat                 loaded 1 certificates
10:19:16:0256 Jcat                 ignoring GPG-KEY-Linux-Foundation-Metadata as not PKCS-7 certificate
10:19:16:0256 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
10:19:16:0256 Jcat                 reading /usr/local/etc/pki/fwupd-metadata/LVFS-CA.pem with 1679 bytes
10:19:16:0256 Jcat                 loaded 1 certificates
10:19:16:0256 Jcat                 reading /var/local/lib/fwupd/pki/secret.key with 2459 bytes
10:19:16:0256 Jcat                 reading /var/local/lib/fwupd/pki/client.pem with 1383 bytes
10:19:16:0256 FuEngine             client certificate now exists: nothing to do
10:19:16:0256 FuHistory            trying to open database '/var/local/lib/fwupd/pending.db'
10:19:16:0257 FuHistory            got schema version of 7
10:19:16:0258 FuPlugin             load(/usr/local/lib/x86_64-linux-gnu/fwupd-plugins-6/libfu_plugin_ccgx.so)
10:19:16:0258 FuPlugin             load(/usr/local/lib/x86_64-linux-gnu/fwupd-plugins-6/libfu_plugin_corsair.so)
10:19:16:0258 FuPlugin             load(/usr/local/lib/x86_64-linux-gnu/fwupd-plugins-6/libfu_plugin_dell_dock.so)
10:19:16:0259 FuPlugin             load(/usr/local/lib/x86_64-linux-gnu/fwupd-plugins-6/libfu_plugin_dfu.so)
10:19:16:0259 FuPlugin             load(/usr/local/lib/x86_64-linux-gnu/fwupd-plugins-6/libfu_plugin_fastboot.so)
10:19:16:0260 FuPlugin             load(/usr/local/lib/x86_64-linux-gnu/fwupd-plugins-6/libfu_plugin_genesys.so)
10:19:16:0260 FuPlugin             load(/usr/local/lib/x86_64-linux-gnu/fwupd-plugins-6/libfu_plugin_jabra.so)
10:19:16:0261 FuPlugin             load(/usr/local/lib/x86_64-linux-gnu/fwupd-plugins-6/libfu_plugin_pci_bcr.so)
10:19:16:0262 FuPlugin             load(/usr/local/lib/x86_64-linux-gnu/fwupd-plugins-6/libfu_plugin_redfish.so)
10:19:16:0262 FuPlugin             load(/usr/local/lib/x86_64-linux-gnu/fwupd-plugins-6/libfu_plugin_rts54hid.so)
10:19:16:0262 FuPlugin             load(/usr/local/lib/x86_64-linux-gnu/fwupd-plugins-6/libfu_plugin_rts54hub.so)
10:19:16:0262 FuPlugin             load(/usr/local/lib/x86_64-linux-gnu/fwupd-plugins-6/libfu_plugin_steelseries.so)
10:19:16:0262 FuPlugin             load(/usr/local/lib/x86_64-linux-gnu/fwupd-plugins-6/libfu_plugin_synaptics_cxaudio.so)
10:19:16:0263 FuPlugin             load(/usr/local/lib/x86_64-linux-gnu/fwupd-plugins-6/libfu_plugin_vli.so)
10:19:16:0263 FuQuirks             loading quirks from /usr/local/share/fwupd/quirks.d
10:19:16:0266 FuQuirks             loading quirks from /var/local/lib/fwupd/quirks.d
10:19:16:0266 XbSilo               attempting to load /var/local/cache/fwupd/quirks.xmlb
10:19:16:0266 XbSilo               file: 261b98aa-f2a1-3028-c474-00d93ab49a38, current:261b98aa-f2a1-3028-c474-00d93ab49a38, cached: (null)
10:19:16:0266 XbSilo               loading silo with file contents
10:19:16:0266 FuSmbios             trying to read /sys/class/dmi/id
10:19:16:0266 FuHwids              loading HwId overrides from /usr/local/etc/fwupd/daemon.conf
[... cut ...]
10:19:19:0688 FuHistory            modifying device ColorHug [1bcca71b3480ec5662e5da8a742c4c805275ffbf]
10:19:19:0695 FuDeviceList         waiting 10000ms for ColorHug device removal
10:19:29:0669 FuHistory            modifying device ColorHug [1bcca71b3480ec5662e5da8a742c4c805275ffbf]
10:19:29:0679 FuHistory            modifying device ColorHug [1bcca71b3480ec5662e5da8a742c4c805275ffbf]
10:19:29:0687 FuIdle               uninhibiting: update
failed to get device before update: failed to wait for detach replug: device 1bcca71b3480ec5662e5da8a742c4c805275ffbf did not come back
```

</details>

hughsie, 2022-05-11:
> detach(colorhug)

Is the device flashing green and red? If so, what does lsusb say at this point?

guilhem, 2022-05-11:
> Is the device flashing green and red? If so, what does lsusb say at this point?

Yes, blinking green/red:

<details>

```
> lsusb -v
Bus 003 Device 123: ID 04d8:f8da Microchip Technology, Inc. Hughski Ltd. ColorHug
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x04d8 Microchip Technology, Inc.
  idProduct          0xf8da Hughski Ltd. ColorHug
  bcdDevice            0.02
  iManufacturer           1 Hughski Ltd.
  iProduct                2 ColorHug (bootloader)
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0029
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xc0
      Self Powered
    MaxPower              300mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      29
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
Device Status:     0x0001
  Self Powered
```

</details>

hughsie, 2022-05-11:
If it's `0x04d8,0xf8da` then I know exactly what's happened!

hughsie, 2022-05-11:
Can you update wip/hughsie/colorhug and try again please.

guilhem, 2022-05-11:
> Can you update wip/hughsie/colorhug and try again please.

:tada: it works!

<details>

```
❯ sudo ./build/src/fwupdtool --plugins colorhug --verbose update
11:29:40:0081 FuDebug              Verbose debugging enabled (on console 1)
11:29:40:0081 FuMain               locked /run/lock/fwupdtool
11:29:40:0084 FuMain               Failed to stop daemon: failed to find fwupd.service: GDBus.Error:org.freedesktop.systemd1.NoSuchUnit: Unit fwupd.service not loaded.
11:29:40:0084 FuConfig             trying to load config values from /usr/local/etc/fwupd/daemon.conf
11:29:40:0084 FuCommon             reading /usr/local/etc/fwupd/daemon.conf with 2080 bytes
11:29:40:0084 FuConfig             trying to load config values from /var/local/etc/fwupd/daemon.conf
11:29:40:0084 FuConfig             using autodetected max archive size 18,4 EB
11:29:40:0084 GLib-GIO             _g_io_module_get_default: Found default implementation local (GLocalVfs) for ‘gio-vfs’
11:29:40:0085 FuRemoteList         loading /usr/local/share/fwupd/metainfo
11:29:40:0085 XbSilo               attempting to load /var/local/cache/fwupd/metainfo.xmlb
11:29:40:0085 XbSilo               file: bca6e538-90b2-fa1e-736e-9a2c313b60b1, current:602f7f24-ac6b-2565-e829-0fbaa24d14e7, cached: (null)
11:29:40:0087 FuRemoteList         loading remote from /usr/local/etc/fwupd/remotes.d/lvfs-testing.conf
11:29:40:0087 FuRemoteList         loading remote from /usr/local/etc/fwupd/remotes.d/lvfs.conf
11:29:40:0087 FuRemoteList         loading remote from /usr/local/etc/fwupd/remotes.d/vendor.conf
11:29:40:0087 FuRemoteList         loading remote from /usr/local/etc/fwupd/remotes.d/vendor-directory.conf
11:29:40:0087 FuRemoteList         loading remote from /usr/local/etc/fwupd/remotes.d/fwupd-tests.conf
11:29:40:0087 FuRemoteList         ordering lvfs-testing=lvfs+1
11:29:40:0087 FuRemoteList         ignoring unfound remote fwupd
11:29:40:0087 Jcat                 ignoring GPG-KEY-Linux-Foundation-Firmware as not PKCS-7 certificate
11:29:40:0087 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
11:29:40:0087 Jcat                 reading /usr/local/etc/pki/fwupd/LVFS-CA.pem with 1679 bytes
11:29:40:0088 Jcat                 loaded 1 certificates
11:29:40:0088 Jcat                 ignoring GPG-KEY-Linux-Foundation-Metadata as not PKCS-7 certificate
11:29:40:0088 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
11:29:40:0088 Jcat                 reading /usr/local/etc/pki/fwupd-metadata/LVFS-CA.pem with 1679 bytes
11:29:40:0088 Jcat                 loaded 1 certificates
11:29:40:0088 Jcat                 reading /var/local/lib/fwupd/pki/secret.key with 2459 bytes
11:29:40:0088 Jcat                 reading /var/local/lib/fwupd/pki/client.pem with 1383 bytes
11:29:40:0088 FuEngine             client certificate now exists: nothing to do
11:29:40:0088 FuHistory            trying to open database '/var/local/lib/fwupd/pending.db'
11:29:40:0088 FuHistory            got schema version of 7
11:29:40:0089 FuPlugin             load(/usr/local/lib/x86_64-linux-gnu/fwupd-plugins-6/libfu_plugin_ccgx.so)
11:29:40:0089 FuPlugin             load(/usr/local/lib/x86_64-linux-gnu/fwupd-plugins-6/libfu_plugin_corsair.so)
11:29:40:0090 FuPlugin             load(/usr/local/lib/x86_64-linux-gnu/fwupd-plugins-6/libfu_plugin_dell_dock.so)
11:29:40:0090 FuPlugin             load(/usr/local/lib/x86_64-linux-gnu/fwupd-plugins-6/libfu_plugin_dfu.so)
11:29:40:0090 FuPlugin             load(/usr/local/lib/x86_64-linux-gnu/fwupd-plugins-6/libfu_plugin_fastboot.so)
11:29:40:0091 FuPlugin             load(/usr/local/lib/x86_64-linux-gnu/fwupd-plugins-6/libfu_plugin_genesys.so)
11:29:40:0091 FuPlugin             load(/usr/local/lib/x86_64-linux-gnu/fwupd-plugins-6/libfu_plugin_jabra.so)
11:29:40:0092 FuPlugin             load(/usr/local/lib/x86_64-linux-gnu/fwupd-plugins-6/libfu_plugin_pci_bcr.so)
11:29:40:0093 FuPlugin             load(/usr/local/lib/x86_64-linux-gnu/fwupd-plugins-6/libfu_plugin_redfish.so)
11:29:40:0093 FuPlugin             load(/usr/local/lib/x86_64-linux-gnu/fwupd-plugins-6/libfu_plugin_rts54hid.so)
11:29:40:0093 FuPlugin             load(/usr/local/lib/x86_64-linux-gnu/fwupd-plugins-6/libfu_plugin_rts54hub.so)
11:29:40:0093 FuPlugin             load(/usr/local/lib/x86_64-linux-gnu/fwupd-plugins-6/libfu_plugin_steelseries.so)
11:29:40:0093 FuPlugin             load(/usr/local/lib/x86_64-linux-gnu/fwupd-plugins-6/libfu_plugin_synaptics_cxaudio.so)
11:29:40:0094 FuPlugin             load(/usr/local/lib/x86_64-linux-gnu/fwupd-plugins-6/libfu_plugin_vli.so)
11:29:40:0094 FuQuirks             loading quirks from /usr/local/share/fwupd/quirks.d
11:29:40:0097 FuQuirks             loading quirks from /var/local/lib/fwupd/quirks.d
11:29:40:0097 XbSilo               attempting to load /var/local/cache/fwupd/quirks.xmlb
11:29:40:0097 XbSilo               file: 261b98aa-f2a1-3028-c474-00d93ab49a38, current:5a3e2a97-6d08-0136-e265-c5d627caeb73, cached: (null)
11:29:40:0111 FuSmbios             trying to read /sys/class/dmi/id
11:29:40:0111 FuHwids              loading HwId overrides from /usr/local/etc/fwupd/daemon.conf
11:29:40:0111 FuHwids              not loading HwId overrides from /var/local/lib/fwupd/daemon.conf
11:29:40:0111 FuHwids              smbios property Manufacturer=Dell Inc.
11:29:40:0111 FuHwids              smbios property EnclosureKind=1f
[... cut ...]
      GUIDs:              40338ceb-b966-4eae-adae-9c32edfcc484
                          afdcc391-6c33-5914-b4d2-b4dd71fe9c5a
                          6bc5ff27-d631-5660-9991-6d24954c6f90 ← USB\VID_273F&PID_1001
                          4841a9e4-e5c8-5107-a83e-d6c6d9c21248 ← USB\VID_273F&PID_1001&REV_0002
      Device Flags:       • Updatable
                          • Supported on remote server
                          • Device can recover flash failures
                          • Unsigned Payload
```

guilhem, 2022-05-11:
I see that I was maybe a colorhug too early adopter :sweat_smile:

hughsie, 2022-05-11:
Great, you've got one of the early ColorHug devices there! I still remember soldering them (by hand!) in my living room. You can also know that fwupd grew out of the ColorHug project, so although Hughski Limited is no more, it grew into something that's now delivering 2 million updates a month! :)

closed 2022-05-11 by hughsie

guilhem, 2022-05-11:
> Great, you've got one of the early ColorHug devices there! I still remember soldering them (by hand!) in my living room. You can also know that fwupd grew out of the ColorHug project, so although Hughski Limited is no more, it grew into something that's now delivering 2 million updates a month! :)

I know, I'm following you since this days ;)

I tried to update my colorhug when I see your tweet about sacrificing a colorhug to test.
I was wondering how I can help with mine ;)