fwupd/fwupd #4628: Thunderbolt: Implement a generic fwupd plugin for Intel Thunderbolt 4 Controller ,Goshen ridge reference docking station designs

krankun, 2022-05-13:
**Is your feature request related to a problem? Please describe.**
As a user I should be able to update the GR nvm for all the docks based on Intel GR reference design for all vendor docks.

**Describe the solution you'd like**
plugin any Goshen ridge dock to USB4 supported chromebook should update the GR NVM for the dock.

**Describe alternatives you've considered**
Not any at this point , as this issue is created to scope and findout the missing peices of the implementation to have  full support for the problem described.

**Additional context**
Idea is to have a generic plugin as much as possible so that all the intel vendors can reuse the same thunderbolt GR plugin.

krankun, 2022-05-13:
cc: @bleungatchromium @djcampello @superm1 @hughsie @westeri 
Hi Benson and Daniel ,

I will be starting this work from Intel side, Feng Li would also come in and join us here soon, Please add details if I miss any.
Currently we need to identify gaps and know what needs to be implemented for this feature. Assuming Mario and Dell team already did some heavy lift here on the topic.

Thank you!

superm1, 2022-05-13:
In general it seems that this should be split out from dell-dock in it's handling for GR to its own plugin.

I suggest once you have something working @CragW can help change Dell dock plugin to work closely with the separated one.

RajmohanMani, 2022-05-18:
@rajmohanmani

RajmohanMani, 2022-05-18:
/cc @rajmohanmani

krankun, 2022-05-19:
@CragW  do we have a .cab file for the dell dock , I want to check the dell plugin probe for nvm update to see what is available, also could you drop me the Dell USB4 docking station model for our usage/tests.

superm1, 2022-05-20:
https://fwupd.org/lvfs/devices/com.dell.wd198ceeeffd.firmware

Wd22tb4

krankun, 2022-05-20:
building fwupd tot on linux: 
./contrib/setup script cannot install meson , could you please share the pip install command to install meson 0.60.0 ?
Build type: native build
meson.build:1:0: ERROR: Meson version is 0.56.2 but project requires >=0.60.0

superm1, 2022-05-20:
`pip3 install meson --upgrade` (or similar - I'm not in front of pip)

hughsie, 2022-05-21:
Or if you're running Fedora 35, you can do `sudo dnf update meson --releasever 36`

krankun, 2022-05-22:
am using linux ubuntu , after distro upgrade to 21.10 , "pip3 install meson --upgrade" has worked and now able to compile code, now looking on to configuring and running fwupd daemon on ubuntu/linux

krankun, 2022-05-26:
I have verified the GR devices can update the GR controller nvm using fwupd plugin , 

localhost ~ # fwupdmgr install /home/GR_NVM_38.cab --allow-reinstall
Waiting?                 [***************************************] Less than one minute remaining?
Successfully installed firmware

After debug looks like the plugin invoked for reference GR device ( which is non dell) 
https://github.com/fwupd/fwupd/blob/main/plugins/dell-dock/fu-dell-dock-usb-usb4.c

As a first step should we make a generic plugin now example "intel-goshenridge" and refactor the code hosted at dell plugin to make it generic ?

hughsie, 2022-05-26:
Is it Goshen ridge specific? Or could it be intel-usb4?

superm1, 2022-05-26:
That's surprising to me that dell-dock plugin invoked here, but I guess it makes life a lot easier to just move code around.

krankun, 2022-05-26:
> Is it Goshen ridge specific? Or could it be intel-usb4?

I have a reference intel USB4 dock which has GR controller on it.

krankun, 2022-05-26:
> That's surprising to me that dell-dock plugin invoked here, but I guess it makes life a lot easier to just move code around.

I think so yes , we were also surprised , I was informed by @westeri  that he tried it some time ago and this has worked.

superm1, 2022-05-26:
The question being posed is if this is really "intel-goshen-ridge" or "intel-usb4" meaning is the protocol going to be generic to the follow on Intel controller from GR or specific to GR?  It's just naming nomenclature.  If it's not yet known, then I say make it intel-goshen-ridge and rename if it need be.

krankun, 2022-05-26:
> The question being posed is if this is really "intel-goshen-ridge" or "intel-usb4" meaning is the protocol going to be generic to the follow on Intel controller from GR or specific to GR? It's just [...]

Sure, I guess its intel-usb4 @westeri or @RajmohanMani  can you please confirm.

RajmohanMani, 2022-05-26:
Like Mario suggested, we can keep it as intel-goshen-ridge and rename it as needed.

Hi Mika, Thoughts here?

Kranthi,
What is the common criteria that is used to load the Dell Dock plugin for Goshen Ridge as well as Dell Dock? Is it USB VID/PID? If so, can you list the PID that we see with Goshen Ridge?

krankun, 2022-05-26:
> Like Mario suggested, we can keep it as intel-goshen-ridge and rename it as needed.
>
> Hi Mika, Thoughts here?
>
> Kranthi, What is the common criteria that is used to load the Dell Dock plugin for [...]

below is what we get from get-devices.

localhost ~ # fwupdmgr get-devices
Decompressing?           [-                                      ]
?
??Intel Thunderbolt generic model name:
?     Device ID:          ae37dac43e4aab74d22c3dc0bf70256630dd782a
?     Current version:    38.00
?     Vendor:             Intel Thunderbolt generic vendor name (THUNDERBOLT:0x8087, TBT:0x8087)
?     Update State:       Success
?     GUIDs:              4371c7e1-ae4e-57a5-900a-a76eca89d19a ? THUNDERBOLT\VEN_8087&DEV_1234
?                         2ef89a74-303c-5990-8bb3-903d3c92a12d ? THUNDERBOLT\VEN_8087&DEV_1234&REV_00
?                         2590556b-dcca-595c-8c71-c9c2247556b3 ? TBT-80871234
?                         3875adf9-b265-5673-814f-a0a2ca6167ae ? TBT-80871234-controller1-1
?     Device Flags:       ? Updatable
?                         ? System requires external power source
?                         ? Device stages updates
?                         ? Signed Payload
?

RajmohanMani, 2022-05-26:
Thanks Kranthi. So, the device ID is listed as 1234.
So, it sounds like the Dell USB4 dock also has the VEN and DEV id as the Goshen Ridge.

krankun, 2022-05-26:
here is for the Kensington USB4 dock:

localhost ~ # fwupdmgr get-devices
Decompressing?           [-                                      ]
?
??TB4 Docking Station:
?     Device ID:          4d41fbd8a348ab95749140a79eb4b0eb814bd1db
?     Current version:    27.00
?     Vendor:             Kensington (THUNDERBOLT:0x001B, TBT:0x001B)
?     GUIDs:              4aaf1db7-8fa8-59f1-88de-434da219f6ae ? THUNDERBOLT\VEN_001B&DEV_0009
?                         d16eb944-16b6-5dea-a91d-69475e7709ef ? THUNDERBOLT\VEN_001B&DEV_0009&REV_00
?                         c15c13f5-0869-52f4-8813-b0a729b43a3b ? TBT-001b0009
?                         d4f094fe-57b9-5b3c-8bde-1eabbc48aaa7 ? TBT-001b0009-controller1-1
?     Device Flags:       ? Updatable
?                         ? System requires external power source
?                         ? Device stages updates
?                         ? Signed Payload

superm1, 2022-05-26:
> Thanks Kranthi. So, the device ID is listed as 1234.
> So, it sounds like the Dell USB4 dock also has the VEN and DEV id as the Goshen Ridge.

No don't believe that's the case.

superm1, 2022-05-26:
Should be using this quirk:
https://github.com/fwupd/fwupd/blob/main/plugins/dell-dock/dell-dock.quirk#L208

CragW, 2022-05-27:
The usb4 in dell-dock plugin uses specific [VID/PID](https://github.com/fwupd/fwupd/blob/main/plugins/dell-dock/fu-dell-dock-common.h#L36) for device [probing](https://github.com/fwupd/fwupd/blob/main/plugins/dell-dock/fu-plugin-dell-dock.c#L165).

In your device list it looks like `thunderbolt` plugin was invoked, not the usb protocol written in `dell-dock`. Worth to `inhibit` that protocol in your configuration file and test it again. My expectation is that this time the device won't be added/updated.

krankun, 2022-05-31:
@CragW is there a way i can disable dell plugin completely and try out, I also think this could be a thunderbolt plugin which updates the GR NVM , am not sure how to inhibit protocol in config , here is the config file we used to update gr nvm,
[firmware.metainfo.txt](https://github.com/fwupd/fwupd/files/8809876/firmware.metainfo.txt)

CragW, 2022-06-01:
We can append `dell_dock` into `DisabledPlugins` line at `/etc/fwupd/daemon.conf`file, then the plugin will be prevented to probing devices. Presumably you still can update peripharal's nvm with `thunderbolt` plugin as long as the thunderbolt interface presents physically, in order to test updating nvm with `usb` interface you'd probably want to disable `thunderbolt` instead.

krankun, 2022-06-02:
DisabledPlugins --> dell_dock //still works
DisabledPlugins -->thunderbolt // doesn't work "No supported devices as expected, also fwupdmgr get-devices doesn't list any devices.

so looks like GR nvm update support is already available in thunderbolt plugin of fwupd

superm1, 2022-06-02:
> DisabledPlugins --> dell_dock //still works
> DisabledPlugins -->thunderbolt // doesn't work "No supported devices as expected, also fwupdmgr get-devices doesn't list any devices.
> 
> so looks like GR nvm update support is already available in thunderbolt plugin of fwupd
> 

It relies upon having USB4 or thunderbolt host. If you want to update GR from USB3 you'll need to port it out of the Dell dock plug-in code.

krankun, 2022-06-10:
My apologies for delayed response , as per our current plans its enough to have fwupd support for TBT3/USB4 host.
Since the currently implementation of the thunderbolt plugin already has that support there will be further development for this feature.

superm1, 2022-09-13:
📕via https://github.com/fwupd/fwupd/commit/5ebc3b6ad91f3e800973adb7f4223af548d0742c

closed 2022-09-13 by superm1