fwupd/fwupd #4629: List updates that require AC power even if power isn't connected

nullr0ute, 2022-05-14:
**Is your feature request related to a problem? Please describe.**
When running "fwupdmgr get-updates" it doesn't list available updates that need a AC power connect. The get-updates option doesn't actually apply updates so there's not a risk of updating while low power.

**Describe the solution you'd like**
It's useful to be able to see all updates that are available.

**Describe alternatives you've considered**
None

**Additional context**

Output without power connected to a Lenovo X1 gen9
````
# fwupdmgr get-updates
Devices with no available firmware updates: 
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • WDC WDS100T2B0C
 • Prometheus IOTA Config
 • UEFI dbx
Devices with the latest available firmware version:
 • Prometheus
Devices not currently updatable:
 • Embedded Controller — Cannot install update when not on AC power
 • Intel Management Engine — Cannot install update when not on AC power
 • System Firmware — Cannot install update when not on AC power
 • UEFI Device Firmware — Cannot install update when not on AC power
No updates available
````

With power attached:
````
# fwupdmgr get-updates
Devices with no available firmware updates: 
 • Prometheus IOTA Config
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI dbx
 • WDC WDS100T2B0C
Devices with the latest available firmware version:
 • Prometheus
20XXS3HC03
│
├─Embedded Controller:
│ │   Device ID:          5bfc5b91c18ef8b751d3d052af77b69ebf738038
│ │   Summary:            UEFI ESRT device
│ │   Current version:    0.1.31
│ │   Minimum Version:    0.1.31
│ │   Vendor:             Lenovo (DMI:LENOVO)
│ │   Update State:       Success
│ │   GUID:               61b65ccc-0116-4b62-80ed-ec5f089ae523
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
[... 137 lines cut ...]
  │     Lenovo Intel Thunderbolt Retimer Firmware 1.2.26.0
  │     
  │     The computer will be restarted automatically after updating device completely . Do NOT turn off your computer or remove the AC adaptor while update is in progress.
  │     
  │     Fixed some issues related to Modern standby.
  │     
  │     Fixed a connection issue.
  │   
  └─ThinkPad X1 Carbon Gen 9 / X1 Yoga Gen 6:
        New version:      1.2.24.0
        Remote ID:        lvfs
        Release ID:       11817
        Summary:          Lenovo ThinkPad X1 Carbon Gen 9 / X1 Yoga Gen 6 Retimer Firmware
        Licence:          Proprietary
        Size:             285.4 kB
        Created:          2022-02-02
        Vendor:           Lenovo
        Release Flags:    • Is upgrade
        Description:      
        Lenovo Intel Thunderbolt Retimer Firmware 1.2.24.0
        
        The computer will be restarted automatically after updating device completely . Do NOT turn off your computer or remove the AC adaptor while update is in progress.
        
        Fixed some bugs regarding connection issues.
````

closed 2022-05-20 by hughsie

nullr0ute, 2022-05-20:
Awesome, thanks :heart:

hughsie, 2022-05-20:
@nullr0ute I'll do an upstream release in a week or so and get it into F36; the translators have some work to do now. :)

nullr0ute, 2022-05-20:
There is absolutely no rush from my PoV, I saw it the other day and just filed it in a drive by RFE as I felt it would be useful, didn't expect a PR in 4 days