fwupd/fwupd #4630: How to obtain a working 'shim' on Ubuntu 20.04 and systems based on that version of Ubuntu?

LinuxOnTheDesktop, 2022-05-15:
**Describe the question**
A ThinkPad of mine cannot process a firmware update. The cause seems to be the 'shim' regression described [here](https://github.com/rhboot/shim/issues/194) and [here](https://github.com/rhboot/shim/issues/194). What can I do? I have not managed to find a way of obtaining or compiling the relevant code.

**fwupd version information**

```shell
$ fwupdmgr --version
runtime   org.freedesktop.fwupd         1.7.5
compile   org.freedesktop.gusb          0.3.4
runtime   com.dell.libsmbios            2.4
runtime   org.kernel                    5.13.0-41-generic
compile   org.freedesktop.fwupd         1.7.5
runtime   org.freedesktop.gusb          0.3.4
$ flatpak list | grep fw
fwupd	org.freedesktop.fwupd	1.8.0	stable	system
```

The ThinkPad in question runs Linux Mint 20.3, which is based on Ubuntu 20.04 (and a version of Mint that has a new Ubuntu base is not due out for a while).

I think that others (not just me) might be helped by an answer to this question.

hughsie, 2022-05-15:
I think you need to file this on the mint bugtracker, there's literally nothing we can do.

closed 2022-05-15 by hughsie

LinuxOnTheDesktop, 2022-05-15:
But, ah, there is [this workaround](https://github.com/fwupd/fwupd/issues/1797#issuecomment-702263720), it seems.