fwupd/fwupd #4631: Thinkpad X1 G9 firmware update failing QueryCapsuleCapability failed

timrichardson, 2022-05-15:
I can't update my Thinkpad with bion 1.52. On the reboot, I get some error messages, one that looks significant is 

QueryCapsuleCapability failed

This is Fedora 36

After re-downloading the update but not rebooting immediately, I see this:
```
efibootmgr -v
BootNext: 0002
BootCurrent: 0001
Timeout: 0 seconds
BootOrder: 0001,001A,0000,001B,001C,001D,001E,001F,0020,0021,0022,0002
Boot0000* Windows Boot Manager	HD(1,GPT,cdc5bfef-e901-4033-a1da-a8bfb3986549,0x800,0x82000)/File(\EFI\Microsoft\Boot\bootmgfw.efi)WINDOWS.........x...B.C.D.O.B.J.E.C.T.=.{.9.d.e.a.8.6.2.c.-.5.c.d.d.-.4.e.7.0.-.a.c.c.1.-.f.3.2.b.3.4.4.d.4.7.9.5.}...4................
Boot0001* Fedora	HD(1,GPT,cdc5bfef-e901-4033-a1da-a8bfb3986549,0x800,0x82000)/File(\EFI\fedora\shimx64.efi)
Boot0002* Linux-Firmware-Updater	HD(1,GPT,cdc5bfef-e901-4033-a1da-a8bfb3986549,0x800,0x82000)/File(\EFI\fedora\fwupdx64.efi)
Boot0010  Setup	FvFile(721c8b66-426c-4e86-8e99-3457c46ab0b9)
Boot0011  Boot Menu	FvFile(126a762d-5758-4fca-8531-201a7f57f850)
Boot0012  Diagnostic Splash Screen	FvFile(a7d8d9a6-6ab0-4aeb-ad9d-163e59a7a380)
Boot0013  Lenovo Diagnostics	FvFile(3f7e615b-0d45-4f80-88dc-26b234958560)
Boot0014  Asset Information	FvFile(da465b87-a26f-4c12-b78a-0361428fa026)
Boot0015  Regulatory Information	FvFile(478c92a0-2622-42b7-a65d-5894169e4d24)
Boot0016  ThinkShield secure wipe	FvFile(3593a0d5-bd52-43a0-808e-cbff5ece2477)
Boot0017  Startup Interrupt Menu	FvFile(f46ee6f4-4785-43a3-923d-7f786c3c8479)
Boot0018  Rescue and Recovery	FvFile(665d3f60-ad3e-4cad-8e26-db46eee9f1b5)
Boot0019  MEBx Hot Key	FvFile(ac6fd56a-3d41-4efd-a1b9-870293811a28)
Boot001A* USB CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,86701296aa5a7848b66cd49dd3ba6a55)
Boot001B* USB FDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,6ff015a28830b543a8b8641009461e49)
Boot001C* NVMe0	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,001c199932d94c4eae9aa0b6e98eb8a400)
Boot001D* USB HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,33e821aaaf33bc4789bd419f88c50803)
Boot001E* PXE BOOT	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,78a84aaf2b2afc4ea79cf5cc8f3d3803)
Boot001F* LENOVO CLOUD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,ad38ccbbf7edf04d959cf42aa74d3650)/Uri(https://download.lenovo.com/pccbbs/cdeploy/efi/boot.efi)
Boot0020* ON-PREMISE	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,ad38ccbbf7edf04d959cf42aa74d3650)/Uri()
Boot0021  Other CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a35400)
Boot0022  Other HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f600)
Boot0023* IDER BOOT CDROM	PciRoot(0x0)/Pci(0x14,0x0)/USB(11,1)
Boot0024* IDER BOOT Floppy	PciRoot(0x0)/Pci(0x14,0x0)/USB(11,0)
Boot0025* ATA HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f6)
Boot0026* ATAPI CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a354)
[tim@fedora ~]$ 
```

hughsie, 2022-05-15:
We need some of the details that were asked in the issue template.

timrichardson, 2022-05-15:
Please note: in Fedora, I staged the update again. Then I attempted to boot into an external drive with Ubuntu, using the external drive as the boot device. This time, the capsule installed. So that's good and interesting, but I can't reproduce the problem.
Is there a dummy capsule that can be used to do pretend firmware updates? 

The diagnostics below come from Fedora.

**fwupd version information**
Please provide the version of the daemon and client.

```

[tim@fedora ~]$ fwupmgr --version
bash: fwupmgr: command not found...
[tim@fedora ~]$ fwupmgr --version
bash: fwupmgr: command not found...
[tim@fedora ~]$ sudo su
[sudo] password for tim: 
[root@fedora tim]# fwupdmgr --version
runtime   org.freedesktop.fwupd         1.7.7
runtime   com.dell.libsmbios            2.4
compile   org.freedesktop.gusb          0.3.10
compile   com.hughsie.libjcat           0.1.10
runtime   org.kernel                    5.17.7-300.fc36.x86_64
compile   org.freedesktop.fwupd         1.7.7
runtime   org.freedesktop.gusb          0.3.10

```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

dnf

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices
```

root@fedora tim]# fwupdmgr get-devices --show-all-devices
20XXS02700
│
├─Unknown Device:
│     Device ID:          63223cc2d1b0b8d4c45ef8cedcb1f45315d3ff20
│     GUID:               ca5c154c-a46a-59d8-bf77-0ca40522cc85 ← GPIO\ID_INT34C5:00
│   
├─11th Gen Intel Core™ i7-1165G7 @ 2.80GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x000000a4
│     Vendor:             Intel
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f ← cpu
│                         30249f37-d140-5d3e-9319-186b1bd5cac3 ← CPUID\PRO_0&FAM_06
│                         2347ad97-b545-5bae-b6b3-b70a9553cbba ← CPUID\PRO_0&FAM_06&MOD_8C
│                         1df99df5-22f4-5505-98cc-22dcc53b132a ← CPUID\PRO_0&FAM_06&MOD_8C&STP_1
│     Device Flags:       • Internal device
│   
├─Embedded Controller:
│     Device ID:          5bfc5b91c18ef8b751d3d052af77b69ebf738038
│     Summary:            UEFI ESRT device
│     Current version:    0.1.32
│     Minimum Version:    0.1.32
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Success
│     GUID:               61b65ccc-0116-4b62-80ed-ec5f089ae523
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
[... 784 lines cut ...]
├── vmlinuz-5.17.5-300.fc36.x86_64
├── vmlinuz-5.17.6-300.fc36.x86_64
└── vmlinuz-5.17.7-300.fc36.x86_64

60 directories, 259 files

**Additional questions**

- Operating system and version:
- Fedora 36

- Have you tried rebooting?
- yes

- Is this a regression?
 It used to work.

- Are you using an NVMe disk?
- Yes

- Is secure boot enabled?
- No

- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS?
- Boot Order Lock is not turned on

hughsie, 2022-05-15:
> Please note: in Fedora, I staged the update again

I suspect fwupd-efi is newer in Fedora than Ubuntu, i.e. contains this fix: https://github.com/fwupd/fwupd-efi/commit/13492e3563da1a25fb2ee7ac7e6a210b211bdccc

timrichardson, 2022-05-15:
I don't think I was clear. It consistently failed in Fedora. It was only
after I tried to boot to Ubuntu that it worked.
But the update which worked was staged in Fedora, because I only ever ran
the update in Fedora.
The external drive has its own EFI partition and I booted from it, but I
never actually got into Ubuntu ... the firmware update took over the boot.
But this time, it worked.
I do not understand the interaction of me trying to boot from an external
drive and the pending update capsule. To me it looks like it doesn't work
at all in Fedora, and somehow trying to boot to Ubuntu fixed it.

-- 

*Tim Richardson CPA, Director, BA, BSc, MAcc*
GrowthPath. Finance transformation for SMEs via Cloud ERP, advanced
reporting, CRM

Mobile: +61 423 091 732 Office/Reception: +61 3 8678 1850.
Book call: https://vyte.in/growthpath/15
Timezone is Melbourne AU.
See this link for international time planning:
https://www.timeanddate.com/worldclock/meeting.html?year=2020&month=5&day=16&p1=152

GrowthPath Pty Ltd
ABN 18100392326
Xero Gold Partner. Dear Inventory, Zoho Analytics and Cin7 Implementation
Partner. Custom integration specialists.

<http://www.growthpath.com.au/>

timrichardson, 2022-05-15:
Or is it possible that from an earlier boot of the external drive, I also
had a pending update installed, so that when I booted off that, it worked?
In that case, it looks the like newer version in Fedora has introduced a
regression,

-- 

*Tim Richardson CPA, Director, BA, BSc, MAcc*
GrowthPath. Finance transformation for SMEs via Cloud ERP, advanced
reporting, CRM

Mobile: +61 423 091 732 Office/Reception: +61 3 8678 1850.
Book call: https://vyte.in/growthpath/15
Timezone is Melbourne AU.
See this link for international time planning:
https://www.timeanddate.com/worldclock/meeting.html?year=2020&month=5&day=16&p1=152

GrowthPath Pty Ltd
ABN 18100392326
Xero Gold Partner. Dear Inventory, Zoho Analytics and Cin7 Implementation
Partner. Custom integration specialists.

<http://www.growthpath.com.au/>

timrichardson, 2022-05-15:
this is ubuntu version info in case it is relevant: 

root@ubuntu:/home/tim# !1
fwupdmgr --version
runtime   org.freedesktop.fwupd         1.7.5
runtime   com.dell.libsmbios            2.4
compile   org.freedesktop.gusb          0.3.10
runtime   org.kernel                    5.15.0-27-generic
compile   org.freedesktop.fwupd         1.7.5
runtime   org.freedesktop.gusb          0.3.10

closed 2023-02-19 by hughsie