fwupd/fwupd #4648: FWUpdMgr Not Seeing Logitech G305 Mouse/Receiver

ourichermath, 2022-05-19:
**Describe the bug**
I've got a Logitech G305 mouse paired with its Lightspeed Receiver (according to Solaar).  But, when I run fwupdmgr --get-devices, I don't see anything regarding them.

**Steps to Reproduce**
- Have a Logitech G305 mouse/receiver plugged into a USB port
- In a terminal window, run fwupdmgr --get-devices

**Expected behavior**
I'd expect something looking like the Logitech mouse/receiver that shows up in Solaar to appear with fwupdmgr.

**fwupd version information**

```fwupdmgr --version                                                                                        bash  18:56:15 
runtime   org.freedesktop.fwupd         1.7.7
runtime   com.dell.libsmbios            2.4
compile   org.freedesktop.gusb          0.3.9
compile   com.hughsie.libjcat           0.1.9
runtime   org.kernel                    5.17.7-200.fc35.x86_64
compile   org.freedesktop.fwupd         1.7.7
runtime   org.freedesktop.gusb          0.3.9
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):  I'm sorry, but I don't think I did.  I think it came installed on Fedora 35.  But, I really don't know.

**fwupd device information**

```fwupdmgr get-devices --show-all-devices                                                                    bash  18:56:12 
System Product Name
│
├─Unknown Device:
│     Device ID:          f685512aa07369c9e77742acef941d779d31e766
│     GUID:               37b440a9-2473-5087-a39b-db84f32a8ed8
│   
├─Unknown Device:
│     Device ID:          a29de8124928f7dc31db026069ac1d601a56056f
│     GUID:               2d9ac22e-0ffe-5691-9e21-f9fecce8761c
│   
├─KINGSTON SKC3000D2048G:
│     Device ID:          71b677ca0f1bc2c5b804fa1d59e52064ce589293
│     Summary:            NVM Express solid state drive
│     Current version:    EIFK31.6
│     Vendor:             Kingston Technology Company, Inc. (NVME:0x2646)
│     GUIDs:              b184d815-dbe0-5c0f-81f2-0b9501d80313
│                         e33407c5-cd9a-5688-a7a5-76a1a02b211f
│                         fece2d79-2905-5eef-a350-42f09e13e81b
│                         a3b32ef4-275f-5e41-badb-43f398f5a799
│                         8db91c1f-712d-5fc3-b251-b7659a22be2e
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs shutdown after installation
│                         • Device is usable for the duration of the update
│                         • Signed Payload
│   
├─Navi 21 [Radeon RX 6800/6800 XT / 6900 XT]:
│     Device ID:          d05d536fa7bd6811a229e6b6e896497fa3cf1084
│     Current version:    c1
│     Vendor:             Advanced Micro Devices, Inc. [AMD/ATI] (PCI:0x1002)
│     GUIDs:              2f861538-7081-5074-b1af-d217e66dc293
│                         db63c13d-a766-5edb-a62c-28a196b428e9
│                         8bd768bf-f7cf-5068-ac9c-c57c876916d9
│                         b0b03a9b-014b-5e71-92f2-aaa8fa2c7738
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
├─Ryzen 5 5600X 6-Core Processor:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Vendor:             AMD
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f
│                         22f9ecf4-588d-5c0a-8326-6ebff3655c6d
│                         af6af62f-513f-5d1b-8c80-23c44eafbe19
│                         36d37879-a735-577e-835a-e9858a7c0c4d
│     Device Flags:       • Internal device
│   
├─System Firmware:
│ │   Device ID:          a45df35ac0e948ee180fe216a5f703f32dda163f
│ │   Summary:            UEFI ESRT device
│ │   Current version:    9731
│ │   Minimum Version:    9731
│ │   Vendor:             ASUS (DMI:American Megatrends Inc.)
│ │   Update State:       Success
│ │   GUIDs:              910a1b7b-350d-507a-b4fd-5512b632202f
│ │                       230c8b18-8d9b-53ec-838b-6cfc0383493a
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Needs a reboot after installation
│ │                       • Cryptographic hash verification is available
│ │                       • Device is usable for the duration of the update
│ │ 
│ └─UEFI dbx:
│       Device ID:        362301da643102b9f38477387e2193e57abaa590
│       Summary:          UEFI revocation database
│       Current version:  83
│       Minimum Version:  83
│       Vendor:           UEFI:Linux Foundation
│       Install Duration: 1 second
│       GUIDs:            fda6234b-adcb-5105-8515-9af647d29775
│                         f8ff0d50-c757-5dc3-951a-39d86e16f419
│                         c6682ade-b5ec-57c4-b687-676351208742
│                         f8ba2887-9411-5c36-9cee-88995bb39731
│                         7d5759e5-9aa0-5f0c-abd6-7439bb11b9f6
│                         0c7691e1-b6f2-5d71-bc9c-aabee364c916
│       Device Flags:     • Internal device
│                         • Updatable
│                         • Needs a reboot after installation
│                         • Only version upgrades are allowed
│                         • Signed Payload
│     
├─TPM:
│     Device ID:          c6a80ac3a22083423992a3cb15018989f37834d6
│     Summary:            TPM 2.0 Device
│     Current version:    3.77.0.5
│     Vendor:             AMD (TPM:AMD)
│     GUIDs:              ff71992e-52f7-5eea-94ef-883e56e034c6
│                         9305de1c-1e12-5665-81c4-37f8e51219b8
│                         78a291ae-b499-5b0f-8f1d-74e1fefd0b1c
│                         65a3fced-b423-563f-8098-bf5c329fc063
│                         5e704f0d-83cb-5364-8384-f46d725a23b8
│     Device Flags:       • Internal device
│
├─UEFI Device Firmware:
│     Device ID:          349bb341230b1a86e5effe7dfe4337e1590227bd
│     Summary:            UEFI ESRT device
│     Current version:    335609894
│     Vendor:             DMI:American Megatrends Inc.
│     Update State:       Success
│     GUID:               f766f6e6-b43d-4acd-a4bd-80ff2f0af5cc
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│
└─UEFI Device Firmware:
Device ID:          2292ae5236790b47884e37cf162dcf23bfcd1c60
Summary:            UEFI ESRT device
Current version:    196608
Vendor:             DMI:American Megatrends Inc.
Update State:       Success
GUID:               f5536e63-e4c0-4e0d-84d4-e8e152b1ba65
Device Flags:       • Internal device
• Updatable
• System requires external power source
• Needs a reboot after installation
• Device is usable for the duration of the update
```

**Additional questions**

The receiver does show up with lsusb:
```
lsusb
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 005 Device 002: ID 08bb:2704 Texas Instruments PCM2704 16-bit stereo audio DAC
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 002: ID 2516:014d Cooler Master Co., Ltd. ARES
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 046d:c53f Logitech, Inc. USB Receiver
Bus 001 Device 004: ID 1b1c:1b39 Corsair Corsair Gaming K65 RGB RAPIDFIRE Keyboard
Bus 001 Device 003: ID 0b05:1939 ASUSTek Computer, Inc. AURA LED Controller
Bus 001 Device 002: ID 8087:0029 Intel Corp. AX200 Bluetooth
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

- Operating system and version:

```
Operating System: Fedora Linux 35
KDE Plasma Version: 5.24.4
KDE Frameworks Version: 5.91.0
Qt Version: 5.15.2
Kernel Version: 5.17.7-200.fc35.x86_64 (64-bit)
Graphics Platform: Wayland
Processors: 12 × AMD Ryzen 5 5600X 6-Core Processor
Memory: 15.6 GiB of RAM
Graphics Processor: AMD Radeon RX 6800 XT
```

- Have you tried rebooting?  Yes.  Many times.
- Is this a regression?  I don't know.

closed 2022-05-19 by hughsie

paulmenzel, 2023-02-13:
I ran into the same problem with Logitech’s Unifying Receiver (046d:c52b).

Is Logitech aware of this problem?

Would it be possible to list the device in `fwupdmgr get-devices` and denote somehow, that it’s unsupported?

hughsie, 2023-02-13:
> Would it be possible to list the device in fwupdmgr get-devices and denote somehow, that it’s unsupported

Sure, we could certainly enumerate them and show that as not UPDATABLE. Do you know of the method solaar uses to get the version numbers of Lightspeed devices? Maybe it's  small tweak to the existing code? i.e. add it to logitech_hidpp.quirk and see what `fwupdtool --plugins logitech_hidpp get-devices --verbose` says.

paulmenzel, 2023-02-13:
> I create a dedicated issue. I have no idea, how solaar is doing it. I can try the rest in the next weeks.

Just as an update, that my issue turned out to be a different issue, so is not related to the suggestion here. Sorry about that.

pinkflames, 2023-06-19:
Sorry for poking an old and closed issue but it would be great, if the device was listed as not updatable, since otherwise people (like me) might end up scourging the internet to figure out if their fwupd installation is missing a non-obvous plugin or some kernel configuration option or maybe have a broken system with bad fwupd integration...