fwupd/fwupd #4649: There is a  “Decompressing” progress bar when the first time to open Gnome-firmware software.

gaby-xie, 2022-05-19:
**Describe the bug**
There is a  “Decompressing” progress bar when the first time to open Gnome-firmware software.

**Steps to Reproduce**
1) Install Fedora 36
2)Update Fedora 36
   $sudo dnf update -y 
3)Install gnome-firmware
   $sudo dnf install gnome-firmware
4)Enable Testing State.
  $sudo dnf copr enable rhughes/fwupd 
  $sudo dnf update fwupd
  $sudo /sbin/shutdown -r now 
  $sudo /usr/bin/fwupdmgr --force refresh 
  $/usr/bin/fwupdmgr enable-remote lvfs-testing
5)Run gnome-firmware and test firmware update.
6)Open the gnome-firmware software by click Firmware icon after dock FW update successful. =>Find the “Decompressing” progress bar issue.
7)Then we launch gnome-firmware in terminal, can also duplicate this issue.

**Expected behavior**
No “Decompressing” progress bar show in gnome-firmware interface.

**fwupd version information**
fwupd version is 1.8.1.

gaby-xie, 2022-05-19:
[image]

chlee75, 2022-05-19:
@hughsie 
Please check Gaby's issue. It seems to be a simple gnome-firmware UI issue and "Decompressing ..." message disappears when pressing "Refresh Metadata" menu in the gnome-firmware.

hughsie, 2022-05-19:
@gaby-xie hey, thanks for reporting this. Can you get me the output from `gnome-firmware --verbose` please when you can see the problem. i.e. install the firmware, but make sure gnome-firmware is closed and then open up gnome firmware on the console with the verbose flag, and then copy and paste the verbose log into this bug. I want to see if this is a fwupd bug where the status is being left incorrectly in the decompressing state, or whether the front-end client is not actually getting the status value at startup. Thanks.

gaby-xie, 2022-05-20:
[image]

gaby-xie, 2022-05-20:
@hughsie Please check the verbose log info. Thanks!

chlee75, 2022-05-20:
@gaby-xie 
Could you share the log files to Richard?
You can save logs in two ways.

First, Use Mouse dragging and Right Click. 
See below.
https://www.howtogeek.com/440558/how-to-copy-and-paste-text-at-linuxs-bash-shell/#:~:text=Ctrl%2BShift%2BC%20and%20Ctrl%2BShift%2BV&text=If%20you%20highlight%20text%20in,text%20into%20a%20clipboard%20buffer.&text=You%20can%20use%20Ctrl%2BShift,or%20into%20another%20terminal%20window.

Second, use script command.
1) Start logging with script command with log file name
  $script gnome-firmware.log  
  $gnome-firmware --verbose
 2) Testing gnome-firmware and quit gnome-firmware when you met the issue
 3) Exit logging with exit command in the terminal and you can find gnome-firmware.log file
  $exit

gaby-xie, 2022-05-23:
@chlee75 
Thanks!
Update gnome-firmware.log to you.

[gnome-firmware.log](https://github.com/fwupd/fwupd/files/8751268/gnome-firmware.log)

chlee75, 2022-05-24:
@hughsie 
Can HP and I test it in the testing state or should we build the latest fwupd to test it?

@gaby-xie 
Thanks for sharing the log files. Richard already seems to fix it.

hughsie, 2022-05-24:
I've built new packages of both into https://copr.fedorainfracloud.org/coprs/rhughes/fwupd/

gaby-xie, 2022-05-25:
@chlee75 
Thanks!
I enable testing state fwupd today morning, but still can duplicate this issue. Thanks!
[gnome-firmware.log](https://github.com/fwupd/fwupd/files/8767552/gnome-firmware.log)

chlee75, 2022-05-25:
@hughsie 

The “Decompressing” message is still displayed on the progress bar. And this message disappears when I press "Refresh Metadata" button in the menu of gnome-firmware.
This issue happens in both HP G4 and G5 dock.
Please check this issue again.

I attached gnome-firmware log files and video.

The “Decompressing” message log.
[gnome-firmware-g5.log](https://github.com/fwupd/fwupd/files/8767955/gnome-firmware-g5.log)

The “Decompressing” message disappear after pressing "Refresh meta"
[gnome-firmware-g5-refresh_meta.log](https://github.com/fwupd/fwupd/files/8767956/gnome-firmware-g5-refresh_meta.log)

gnome-firmware video.
[decompressing-gnome-firmware.zip](https://github.com/fwupd/fwupd/files/8768006/decompressing-gnome-firmware.zip)

chlee75, 2022-05-25:
@hughsie 

> I'm struggling to debug this

I appreciate it.

>can you do 100% explicit instructions

I describe 100% reproduction steps for this issue and attached video clip.

Reproduction Steps

0. Pre-condition 
  Linux version is Fedora 36
  fwupd 1.7.7 is running on your computer
  Connect HP G5 Dock to your computer (During testing, keep the dock connected to your computer)

1. Update fwupd for lvfs-tesing and reboot system
 $sudo dnf copr enable rhughes/fwupd
 $sudo dnf update fwupd
 $shutdown -r now

2. Enable lvfs-tesing
 $fwupdmgr --force refresh
 $fwupdmgr enable-remote lvfs-testing

3. Update Firmware 
 Run gnome-firmware
 Select HP G5 Dock and Update firmware 
 Close gnome-firmware when firmware update is complete
 Run gnome-firmware again. "Decompressing..." is always displayed whenever you run the gnome-firmware

Video Clip
[decompressing-gnome-firmware-reproduction.zip](https://github.com/fwupd/fwupd/files/8775196/decompressing-gnome-firmware-reproduction.zip)

hughsie, 2022-05-26:
That was super, thanks. I've managed to reproduce it myself, and am just testing a fix.

closed 2022-05-26 by hughsie

chlee75, 2022-05-27:
@gaby-xie , @syabbashp 

I checked this issue has been fixed with https://github.com/fwupd/fwupd/pull/4678.
Richard also has updated it to both the fwupd main and testing state package.
You can also check it.

gaby-xie, 2022-05-27:
Yes, this issue had been fixed. Thanks!