fwupd/fwupd #4664: Error when opening the Ubuntu Software or the Flatpak Software app

Overcraft90, 2022-05-23:
**Describe the bug**
Once the apps Ubuntu Software or Flatpak Software are launched for the first time after booting into the system an error message is displayed as blow:
 --FLATPAK--
[image]
--UBUNTU--
[image]

**Steps to Reproduce**
Click on the icon for Ubuntu Software or Flatpak Software on the Deck.
[image]

**Expected behavior**
The application should start without prompting any error message (as used to be previously).

**fwupd version information**

```shell
Failed to connect to daemon: Error calling StartServiceByName for org.freedesktop.fwupd: Unit snap.fwupd.fwupd.service not found.
```
Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
Failed to connect to daemon: Error calling StartServiceByName for org.freedesktop.fwupd: Unit snap.fwupd.fwupd.service not found.
```

**Additional questions**

- Operating system and version: Ubuntu 22.04 "JAMMY JELLYFISH".
- Have you tried rebooting? Yes.
- Is this a regression? No.

superm1, 2022-05-23:
Did you install the snap at some point?  22.04 has a pretty recent version of fwupd, you should be able to just use the packaged one.

Overcraft90, 2022-05-23:
I'm not sure what you mean by "the snap". I have both Ubuntu Software and Flatpak — I installed this one since it is faster and apps are more stable and up to date.

Should I remove Ubuntu Software and simply keep Flatpak? What I've done is installing `fwupd` in the Flatpak app; now, I'm prompted with this message when I open Ubuntu Software:

[image]

No more messages when I open the Flatpak app, though. Let me know what to do, thanks!

closed 2022-05-24 by superm1