fwupd/fwupd #4677: login notice 'see fwupdmgr get-upgrades for more info, then I run it and get `Failed to create '/boot/EFI/arch/fw': Read-only file system`

ElijahLynn, 2022-05-25:
**Describe the bug**
I'm getting login notice on Gnome 42 to `fwupdmgr get-upgrades` for more info, then I run it and get `Failed to create '/boot/EFI/arch/fw': Read-only file system`

**Steps to Reproduce**
Steps to reproduce the behavior.

**Expected behavior**
A clear and concise description of what you expected to happen.

**fwupd version information**
Please provide the version of the daemon and client.

```shell
fwupdmgr --version
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices
```

**System UEFI configuration**
Please provide the output of the following commands:

```shell
efibootmgr -v
```

```shell
efivar -l | grep fw
```

```shell
tree /boot
```

**Additional questions**

- Operating system and version:
- Have you tried rebooting?
- Is this a regression?
- Are you using an NVMe disk?
- Is secure boot enabled?
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS?

hughsie, 2022-05-25:
We need the info requested in the issue template please

ElijahLynn, 2022-05-25:
> We need the info requested in the issue template please

sorry, still populating, I edited the title and hit the wrong button so it created the issue before I was done. Info is coming, thx!

ElijahLynn, 2022-05-25:
K, it is all populated now. Is that helpful? Anything else?

superm1, 2022-05-25:
Is your ESP actually read only?  How are you mounting it?  `/etc/fstab`?

ElijahLynn, 2022-05-26:
> Is your ESP actually read only? How are you mounting it? `/etc/fstab`?

`/boot` appears to be `rw`. 

```shell
❯ cat /etc/fstab
# Static information about the filesystems.
# See fstab(5) for details.

# <file system> <dir> <type> <options> <dump> <pass>
# /dev/nvme0n1p2 LABEL=ROOT
UUID=e989b9f4-cd65-48ca-bb5b-4ec6a00e61c9       /               ext4            rw,relatime     0 1

# /dev/nvme0n1p1 LABEL=EFI
UUID=295A-5841          /boot           vfat            rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,utf8,errors=remount-ro       0 2

/swapfile none swap defaults 0 0

❯ ls -alh /boot/
total 52M
drwxr-xr-x  2 root root 4.0K Dec 31  1969 ./
drwxr-xr-x 20 root root 4.0K Dec  9 10:03 ../
-rwxr-xr-x  1 root root  33M May 10 16:50 initramfs-linux-fallback.img*
-rwxr-xr-x  1 root root 8.8M May 10 16:50 initramfs-linux.img*
-rwxr-xr-x  1 root root  10M May 10 16:50 vmlinuz-linux*
```

ElijahLynn, 2022-05-26:
I just updated the issue with this step to show that it was running as root:

```sh
❯ sudo fwupdmgr upgrade 349bb341230b1a86e5effe7dfe4337e1590227bd

[sudo] password for elijah: 
Devices with no available firmware updates: 
 • Embedded Controller
 • Jabra SPEAK 410 USB
 • ThinkPad Thunderbolt 3 Dock
 • Thunderbolt host controller
 • UEFI dbx
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade Intel Management Engine from 184.50.3460 to 184.86.3909?             ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ Lenovo ThinkPad X1 Yoga 3rd Corporate ME Firmware                            ║
║                                                                              ║
║ Version 11.8.86.3909 (LVFS: 184.86.3909)                                     ║
║                                                                              ║
║ Fix Release for Intel Hotfix.                                                ║
║                                                                              ║
║ CVES provided in issue list.                                                 ║
║                                                                              ║
║ 20LD0017US must remain plugged into a power source for the duration of the   ║
║ update to avoid damage.                                                      ║
╚══════════════════════════════════════════════════════════════════════════════╝

Perform operation? [Y|n]: 
Downloading…             [***************************************]
Downloading…             [***************************************]
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Decompressing…           [                                       ]Failed to create '/boot/EFI/arch/fw': Read-only file system
```

superm1, 2022-05-26:
You seem to be missing the regular EFI directories on your ESP.  Is this really a bootable ESP?

ElijahLynn, 2022-05-26:
> You seem to be missing the regular EFI directories on your ESP. Is this really a bootable ESP?

I honestly don't even know what an ESP is. I'm just an end Gnome user so pretty high up in userland.

superm1, 2022-05-26:
I don't see how a system can boot this.  An ESP needs an EFI bootloader.  If you don't have one, then you won't be able to boot anything.  This must not be where you store your EFI bootloader.

ElijahLynn, 2022-05-26:
K, I'm reading through this now, which seems to be a similar setup > https://unix.stackexchange.com/questions/672744/move-boot-to-esp-currently-boot-efi

fwiw, I think I followed https://wiki.archlinux.org/title/EFI_system_partition to install Arch.

ElijahLynn, 2022-05-26:
> [mount](https://wiki.archlinux.org/title/Mount) ESP to /boot. This is the preferred method when directly booting an [EFISTUB](https://wiki.archlinux.org/title/EFISTUB) kernel from UEFI or booting it via a boot manager like [systemd-boot](https://wiki.archlinux.org/title/Systemd-boot).