fwupd/fwupd #4684: 1.8.1: test suite is failing

kloczek, 2022-05-27:
Meson status:
```console
fwupd 1.8.1

  User defined options
    auto_features                 : enabled
    bindir                        : /usr/bin
    buildtype                     : plain
    datadir                       : /usr/share
    includedir                    : /usr/include
    infodir                       : /usr/share/info
    libdir                        : /usr/lib64
    libexecdir                    : /usr/libexec
    localedir                     : /usr/share/locale
    localstatedir                 : /var
    mandir                        : /usr/share/man
    prefix                        : /usr
    sbindir                       : /usr/sbin
    sharedstatedir                : /var/lib
    sysconfdir                    : /etc
    wrap_mode                     : nodownload
    bluez                         : enabled
    build                         : all
    consolekit                    : enabled
    curl                          : enabled
    docs                          : gtkdoc
    firmware-packager             : true
    gudev                         : enabled
    gusb                          : enabled
    introspection                 : enabled
    libarchive                    : enabled
    lvfs                          : true
    lzma                          : enabled
    man                           : true
    plugin_amt                    : enabled
    plugin_dell                   : enabled
    plugin_dummy                  : true
    plugin_emmc                   : enabled
    plugin_flashrom               : enabled
    plugin_gpio                   : enabled
    plugin_intel_spi              : true
    plugin_logitech_bulkcontroller: enabled
    plugin_modem_manager          : enabled
    plugin_msr                    : enabled
    plugin_nvme                   : enabled
    plugin_parade_lspcon          : enabled
    plugin_platform_integrity     : true
    plugin_powerd                 : enabled
    plugin_realtek_mst            : enabled
    plugin_redfish                : enabled
    plugin_synaptics_rmi          : enabled
    plugin_thunderbolt            : enabled
    plugin_tpm                    : enabled
    plugin_uefi_capsule           : enabled
    plugin_uefi_capsule_splash    : true
    plugin_uefi_pk                : enabled
    polkit                        : enabled
    soup_session_compat           : true
    supported_build               : enabled
    systemd                       : enabled
    tests                         : true
```
I'm running my builds inside LXC zones and I'm not sure is it the cause. Nevertheless test suite is failing
```console
+ cd fwupd-1.8.1
+ /usr/bin/meson test -C x86_64-redhat-linux-gnu --num-processes 48 --print-errorlogs
ninja: Entering directory `/home/tkloczko/rpmbuild/BUILD/fwupd-1.8.1/x86_64-redhat-linux-gnu'
ninja: no work to do.
 1/35 fwupd-exported-api                     OK               0.04s
 2/35 fwupd-self-test                        OK               0.04s
[... 62 lines cut ...]
ok 2 /redfish/common
# FuPluginRedfish-DEBUG: using 1.2.3 for P50 v1.2.3 PROD
# FuPluginRedfish-DEBUG: using 1.2.3 for P50 1.2.3 DEV
ok 3 /redfish/common{version}
ok 4 /redfish/common{lenovo}
Bail out! FuPluginRedfish:ERROR:../plugins/redfish/fu-self-test.c:220:fu_test_redfish_network_mac_addr_func: assertion failed (error == NULL): missing 00:13:F7:29:C2:D8: failed to construct proxy for org.freedesktop.NetworkManager: Could not connect: Connection refused (g-io-error-quark, 39)
stderr:
**
FuPluginRedfish:ERROR:../plugins/redfish/fu-self-test.c:220:fu_test_redfish_network_mac_addr_func: assertion failed (error == NULL): missing 00:13:F7:29:C2:D8: failed to construct proxy for org.freedesktop.NetworkManager: Could not connect: Connection refused (g-io-error-quark, 39)
――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――

35/35 thunderbolt-self-test                  OK               3.40s

Summary of Failures:

32/35 linux-swap-self-test           FAIL             0.23s   killed by signal 6 SIGABRT
34/35 redfish-self-test              FAIL             0.33s   killed by signal 6 SIGABRT

Ok:                 33
Expected Fail:      0
Fail:               2
Unexpected Pass:    0
Skipped:            0
Timeout:            0
```

kloczek, 2022-05-27:
If dbus session would be requires I can run pytest over dbus-session command ..

kloczek, 2022-05-27:
Tested with dbus-session and result is exactly the same.

kloczek, 2022-05-27:
I'm using my own distribution and I have no ability to replicate that on Fedora.
What I've tested was in my normal devel LXC zone where I have dbus and NM.
Final build will be stripped those bits because NM and dbus are not on list of BuildRequires and are not part of the base system used to assembly per package build env.

superm1, 2022-06-01:
> @superm1 -- opinion please -- do you think the self tests should work without a working DBus daemon?

I think anything dependent on it should be skipped.  We do this for plenty of tests already where things like a TSS emulator isn't there for example.

hughsie, 2022-06-01:
@kloczek if you can add the g_error_matches() with the right values, we can g_test_skip() the right ones. Could you do a PR for that perhaps?

kloczek, 2022-07-31:
> @kloczek if you can add the g_error_matches() with the right values, we can g_test_skip() the right ones. Could you do a PR for that perhaps?

Sorry I'm not so familiar with fwupd code to be able prepare PR for that.
Please let me know if it is anything else which I can try to diagnose that issue.

closed 2022-08-22 by hughsie