fwupd/fwupd #4686: ESP not detected

Samisafool, 2022-05-30:
**Describe the bug**
ESP is not successfully detected when /boot is not ESP and /boot and ESP (/efi) are separate partitions

**Steps to Reproduce**
- ESP on /efi
- /boot is seperate partition - without separate partition this bug does not appear
- `fwupdmgr update`
- It warns you:
```
WARNING: UEFI ESP partition not detected or configured
  See https://github.com/fwupd/fwupd/wiki/PluginFlag:esp-not-found for more information.
Devices with no available firmware updates: 
```

**Expected behavior**
It would find the ESP

**fwupd version information**
Please provide the version of the daemon and client.

```shell
$ fwupdmgr --version
runtime   org.freedesktop.fwupd         1.8.1
runtime   org.freedesktop.fwupd-efi     1.3
compile   org.freedesktop.gusb          0.3.10
runtime   com.dell.libsmbios            2.4
runtime   org.kernel                    5.18.0-zen1-1-zen
runtime   com.hughsie.libjcat           0.1.11
compile   com.hughsie.libjcat           0.1.11
compile   org.freedesktop.fwupd         1.8.1
runtime   org.freedesktop.gusb          0.3.10
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): pacman

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
$ fwupdmgr get-devices --show-all-devices
81VV
│
├─Unknown Device:
│     Device ID:          b600a576418a18bd2254d06d97d2bd8f58e47356
│     GUID:               1c1a9219-4eb7-52d5-ad47-fb517af3f656 ← GPIO\ID_INT3455:00
│   
├─Core™ i5-1035G1 CPU @ 1.00GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x000000b0
│     Vendor:             Intel
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f ← cpu
│                         30249f37-d140-5d3e-9319-186b1bd5cac3 ← CPUID\PRO_0&FAM_06
│                         d70b62bc-d9c6-5156-bd86-4f7440ccb34d ← CPUID\PRO_0&FAM_06&MOD_7E
│                         95b2821b-aba1-51cc-ac93-bf7518dbceb0 ← CPUID\PRO_0&FAM_06&MOD_7E&STP_5
│     Device Flags:       • Internal device
│   
├─Ice Lake-LP LPC Controller:
│ │   Device ID:          71b31258b13a4b2793e529856a190f8fb02ad151
│ │   Current version:    30
│ │   Vendor:             Intel Corporation (PCI:0x8086)
│ │   GUIDs:              0da3f89f-aa5b-53ff-b7bf-b9d79a587bb1 ← PCI\VEN_8086&DEV_3482
│ │                       eb1e0dbd-914e-51cd-8704-edea53b7610a ← PCI\VEN_8086&DEV_3482&REV_30
│ │                       016039eb-f821-5eab-8dda-453e63fdcebf ← PCI\VEN_8086&DEV_3482&SUBSYS_17AA3801
│ │                       36816870-b8fc-5d0e-9ba7-3c5d212429bd ← PCI\VEN_8086&DEV_3482&SUBSYS_17AA3801&REV_30
│ │                       2d80f689-0b5e-5c4b-b6df-bd767f6e9f05 ← INTEL_SPI_CHIPSET\ID_PCH300
│ │   Device Flags:       • Internal device
│ │                       • Cryptographic hash verification is available
│ │ 
│ ├─BIOS:
│ │     Device ID:        ff7dbf2f6e354a5727c6ce1c466230f38bd26ff0
[... 161 lines cut ...]
```

```shell
$ tree /boot
/boot
├── amd-ucode.img
├── initramfs-linux-lts-fallback.img
├── initramfs-linux-lts.img
├── initramfs-linux-zen-fallback.img
├── initramfs-linux-zen.img
├── intel-ucode.img
├── vmlinuz-linux-lts
└── vmlinuz-linux-zen

0 directories, 8 files
```

**Additional questions**

- Operating system and version: Arch Linux
- Have you tried rebooting? Yes
- Is this a regression? Unknown
- Are you using an NVMe disk? Yes
- Is secure boot enabled? No
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? I don't have that option but this is Lenovo, yes

superm1, 2022-06-01:
Do you have udisks installed?  If not - can you please install it?  Does it keep happening?
If it keeps happening, lets see the output of `udisksctl dump`

Samisafool, 2022-07-31:
I have already figured it out - it was due to /efi not being marked with efi flag. Have since changed my setup too.

closed 2022-07-31 by Samisafool

mars4science, 2023-11-14:
@superm1 Why fwupd needs to check to the partition? Is it safety check only or are there some files to be used? If files, how to point it to the files? I have `EFI/boot` folder with `bootx64.efi  grubx64.efi  mmx64.efi` - are they what fwupd needs?