fwupd/fwupd #4730: Probing Genesys Hub/Dock causes USB disconnect w/ fwupd 1.8.1

codesalatdev, 2022-06-10:
**Describe the bug**
Every time fwupd gets a list of devices and reaches a Genesys Hub, devices connected to it get disconnected for around ~2 seconds. This makes peripherals completely unresponsive.

**Steps to Reproduce**
- Have a Genesys Hub/Dock
- Install/Upgrade to fwupd 1.8.1
- Reboot or (re-)start fwupd service manually

**Expected behavior**
fwupd probes the Genesys Hub/Dock without interrupting operation of connected devices.

**fwupd version information**
Please provide the version of the daemon and client.

```shell
runtime   org.freedesktop.fwupd         1.8.1
runtime   org.freedesktop.fwupd-efi     1.3
runtime   com.dell.libsmbios            2.4
compile   org.freedesktop.gusb          0.3.10
compile   com.hughsie.libjcat           0.1.10
runtime   org.kernel                    5.17.12-300.fc36.x86_64
compile   org.freedesktop.fwupd         1.8.1
runtime   org.freedesktop.gusb          0.3.10
```

Installed via DNF, shipped with Distro.

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```
WARNING: UEFI capsule updates not available or enabled in firmware setup
  See https://github.com/fwupd/fwupd/wiki/PluginFlag:capsules-unsupported for more information.
Z97-HD3
│
├─Core™ i7-4790 CPU @ 3.60GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x00000028
│     Vendor:             Intel
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f ← cpu
│                         30249f37-d140-5d3e-9319-186b1bd5cac3 ← CPUID\PRO_0&FAM_06
│                         dd2cdaeb-2b08-599d-aaec-5a8374add8d6 ← CPUID\PRO_0&FAM_06&MOD_3C
│                         112c1ba3-2c29-5ab2-bd72-ea65bd10dd67 ← CPUID\PRO_0&FAM_06&MOD_3C&STP_3
│     Device Flags:       • Internal device
│   
├─GP104 [GeForce GTX 1070 Ti]:
│     Device ID:          ce4c74a5188d5b9cdb1e72ed32dad2d313c1c999
│     Current version:    a1
│     Vendor:             NVIDIA Corporation (PCI:0x10DE)
│     GUIDs:              83d9aece-e3d0-507b-9ab6-b655677c57d0 ← PCI\VEN_10DE&DEV_1B82
│                         656a95f5-8a2a-5d6d-b647-e81f4bcf9f80 ← PCI\VEN_10DE&DEV_1B82&REV_A1
│                         cdb70969-09e3-5411-b7ae-3a591137c130 ← PCI\VEN_10DE&DEV_1B82&SUBSYS_19DA2435
│                         63c30daf-6f13-5a66-a1ab-0918613142fa ← PCI\VEN_10DE&DEV_1B82&SUBSYS_19DA2435&REV_A1
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
├─Hitachi HDT721032SLA360:
│     Device ID:          f219ac034743df43be49a235e41f128dd47d855f
│     Summary:            ATA drive
│     Current version:    ST2OA31B
│     Vendor:             Western Digital (ATA:0x101C, OUI:000cca)
│     Serial Number:      STF204ML0XJJBP
│     GUIDs:              01328342-7517-5799-b46f-d2a83699d70f ← IDE\Hitachi_HDT721032SLA360_________________ST2OA31B
│                         b3ae9f41-f660-548d-81e9-68443642a42c ← IDE\0Hitachi_HDT721032SLA360_________________
│                         5f909e86-3a7f-5557-99b5-b8eb942667d9 ← Hitachi HDT721032SLA360
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
[... 687 lines cut ...]
  Vendor:               Western Digital
  VendorId:             ATA:0x101C
  VendorId:             OUI:0014ee
  Version:              01.01A01
  VersionFormat:        plain
  Icon:                 drive-harddisk
  Created:              2022-06-10

09:15:55:0920 FuPlugin             destroy(synaptics_mst)
09:15:55:0920 FuPlugin             destroy(logind)
09:15:55:0920 FuPlugin             destroy(pci_mei)
09:15:55:0920 FuPlugin             destroy(upower)
09:15:55:0920 FuPlugin             destroy(modem_manager)
09:15:55:0921 FuPlugin             destroy(linux_lockdown)
09:15:55:0921 FuPlugin             destroy(linux_swap)
09:15:55:0921 FuPlugin             destroy(dell)
09:15:55:0921 FuPlugin             destroy(flashrom)
09:15:55:0921 FuPlugin             destroy(linux_tainted)
09:15:55:0921 FuPlugin             destroy(gpio)
09:15:55:0921 FuPlugin             destroy(tpm)
09:15:55:0921 FuPlugin             destroy(redfish)
09:15:55:0921 FuPlugin             destroy(uefi_capsule)
```

Original bug report on RedHat's Bugzilla: https://bugzilla.redhat.com/show_bug.cgi?id=2092210

gportay, 2022-06-10:
I will give a try

gportay, 2022-06-10:
Indeed, I can see the devices connected to the hub get re-enumerated.

I will see what I can do to get rid of that.

gportay, 2022-06-10:
It happens after claiming the interface at `_open()` and `_close()`.

gportay, 2022-06-11:
@codesalatdev: do you mind testing #4737 if that works for you?

gportay, 2022-06-11:
With these changes I do not see re-enumeration in `dmesg` anymore after running `fwupdtool get-devices`.

hughsie, 2022-06-11:
I can build a fedora rpm if that helps.

codesalatdev, 2022-06-11:
@gportay will do, but our timezones seem to be quite different so I'll report back within 24h 😁

@hughsie please only do if you have spare time on your hands. I have plenty of it tomorrow and the compiling instructions on the wiki look simple enough

hughsie, 2022-06-12:
Builds of that branch here: https://people.freedesktop.org/~hughsient/fedora/32/x86_64/ -- just grab the fwupd packages, the others might be unsuitable.

codesalatdev, 2022-06-12:
Had to use the libjcat rpm from your site @hughsie since the newest version on F36 seems to be 0.1.10 (https://packages.fedoraproject.org/pkgs/libjcat/libjcat/) and fwupd requires 0.1.11.

The build fixed the issue, I can no longer observe a disconnect on manual service start nor after boot.

Thanks a lot for the quick responses and your work :smiley:

Can be closed if none of you have anything else to add

gportay, 2022-06-12:
Thanks you very much for your report @codesalatdev; that must have been not easy to find out why your devices on your hub get re-enumerated :)

closed 2022-06-12 by gportay