fwupd/fwupd #4733: ME update for Lenovo Thinkpad offered to Intel NUC

tomkv, 2022-06-10:
**Describe the bug**

fwupd offers the following firmware update:

```
$ fwupdmgr get-upgrades
Devices with no available firmware updates:
 • SSD 860 EVO 500GB
 • System Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI dbx
NUC11TNHi3
│
└─Intel Management Engine:
  │   Device ID:          2292ae5236790b47884e37cf162dcf23bfcd1c60
  │   Summary:            UEFI ESRT device
  │   Current version:    0.35.1898
  │   Vendor:             Lenovo (DMI:Intel Corp.)
  │   Update State:       Success
  │   GUID:               19aef08f-2701-4db3-b90b-d4bc1412ade7
  │                       950e547f-0090-5a43-a74f-497902f980a9
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Device is usable for the duration of the update
  │
  └─ThinkPad L13 Gen 2:
        New version:      240.35.1951
        Remote ID:        lvfs
        Release ID:       13426
        Summary:          Lenovo ThinkPad L13 Gen 2 Consumer ME Firmware
        License:          Proprietary
        Size:             4,1 MB
        Created:          2016-07-08
        Urgency:          High
        Details:          https://pcsupport.lenovo.com/de/en/search?query=R1FRN18W
        Vendor:           Lenovo
        Release Flags:    • Is upgrade
        Description:
        description (en_US): Lenovo ThinkPad L13 gen2 Consumer ME Firmware
```
Problem is, that this is not a Lenovo device. This is Intel NUC (NUC11TNHi3 - DMI: Intel(R) Client Systems NUC11TNHi3/NUC11TNBi3, BIOS TNTGL357.0064.2022.0217.1550 02/17/2022)

**Steps to Reproduce**

- Installation from scratch on the above-mentioned Intel NUC machine
- Installation of gnome-software from the apt-package (Ubuntu by default uses snap version)
- The update is offered there.

**Expected behavior**

Updates not intended for device should not be shown as available to install.

**fwupd version information**

```
runtime   org.freedesktop.fwupd         1.7.5
runtime   com.dell.libsmbios            2.4
compile   org.freedesktop.gusb          0.3.10
runtime   org.kernel                    5.15.0-37-generic
compile   org.freedesktop.fwupd         1.7.5
runtime   org.freedesktop.gusb          0.3.10
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
[... 778 lines cut ...]
├── initrd.img -> initrd.img-5.15.0-37-generic
├── initrd.img.old -> initrd.img-5.15.0-33-generic
├── initrd.img-5.15.0-33-generic
├── initrd.img-5.15.0-37-generic
├── memtest86+.bin
├── memtest86+.elf
├── memtest86+_multiboot.bin
├── System.map-5.15.0-33-generic
├── System.map-5.15.0-37-generic
├── vmlinuz -> vmlinuz-5.15.0-37-generic
├── vmlinuz.old -> vmlinuz-5.15.0-33-generic
├── vmlinuz-5.15.0-33-generic
└── vmlinuz-5.15.0-37-generic

12 directories, 597 files
```

**Additional questions**

- Operating system and version: Ubuntu 22.04
- Have you tried rebooting? Yes
- Is this a regression? 
- Are you using an NVMe disk? No
- Is secure boot enabled? Yes
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? No

hughsie, 2022-06-10:
We just pushed a fix for this; can you wait 4 hours, then do "fwupdmgr refresh --force" and then "fwupdmgr get-updates" please

tomkv, 2022-06-11:
The update is no longer offered, thank you!

closed 2022-06-11 by tomkv