fwupd/fwupd #4734: Firmware update fails on Thinkpad X1 Carbon Gen 10

mripard, 2022-06-11:
**Describe the bug**

When updating either the system firmware or the Embedded Controller on a Thinkpad X1 Carbon Gen 10, after a reboot the firmware isn't updated.

The display does blank at boot, but nothing is printed so it's hard to tell if there's an error or something.

**Steps to Reproduce**

Trigger a firmware update through either the UI or fwupdmgr update, reboot.

**Expected behavior**

The firmware should have been updated.

**fwupd version information**
Please provide the version of the daemon and client.

```shell
fwupdmgr --version 
runtime   org.freedesktop.fwupd         1.8.1
runtime   org.freedesktop.fwupd-efi     1.3
compile   org.freedesktop.gusb          0.3.10
runtime   com.dell.libsmbios            2.4
runtime   org.kernel                    5.17.12-300.fc36.x86_64
compile   com.hughsie.libjcat           0.1.10
compile   org.freedesktop.fwupd         1.8.1
runtime   org.freedesktop.gusb          0.3.10

```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

It was installed by the Fedora 36 Silverblue installer.

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices
21CB006QFR
│
├─ELAN067C:00 04F3:31F9:
│     Device ID:          a945346f09c11090062c7a76ccd870fdb2663199
│     Summary:            Touchpad
│     Current version:    0x0005
│     Bootloader Version: 0x0003
│     Vendor:             HIDRAW:0x04F3
│     GUIDs:              61c9c175-eb11-59e4-bdd0-6dd9a611105f ← HIDRAW\VEN_04F3&DEV_31F9
│                         29518b68-207a-5d25-b083-c972bc8223ff ← HIDRAW\VEN_04F3&DEV_31F9&REV_00
│                         819fddf1-8ece-5c1e-9c5f-888d1e6a8791 ← HIDRAW\VEN_04F3&DEV_31F9&MOD_012E
│                         4b808d27-c833-50cc-8177-9a401f5c08fe ← ELANTP\ICTYPE_14
│                         a846d7d1-c2c9-5d9e-9f10-78b71168ae07 ← ELANTP\ICTYPE_14&MOD_012E
│     Device Flags:       • Internal device
│                         • Updatable
│   
├─Unknown Device:
│     Device ID:          6b5a06645e27c4e94b6e67d5185c46d11ec9ad67
│     GUID:               c96751f4-16e8-59ea-975d-69d3dac88d7d ← GPIO\ID_INTC1055:00
│   
├─12th Gen Intel Core™ i5-1240P:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x0000041c
│     Vendor:             Intel
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f ← cpu
│                         30249f37-d140-5d3e-9319-186b1bd5cac3 ← CPUID\PRO_0&FAM_06
│                         ab855c04-4ff6-54af-8a8a-d8193daa0cd8 ← CPUID\PRO_0&FAM_06&MOD_9A
│                         3ebbde86-d03e-549a-a8fd-02ebf9aa537a ← CPUID\PRO_0&FAM_06&MOD_9A&STP_3
│     Device Flags:       • Internal device
│   
├─Alder Lake-P Integrated Graphics Controller:
[... 398 lines cut ...]
├── loader.1
│   ├── entries
│   │   ├── ostree-1-fedora.conf
│   │   └── ostree-2-fedora.conf
│   └── grub.cfg
├── lost+found  [error opening dir]
└── ostree
    ├── fedora-bf67205e2f3c265fb381dde505da28aa6c830bd5a068c13034e0d20384262942
    │   ├── initramfs-5.17.12-300.fc36.x86_64.img
    │   └── vmlinuz-5.17.12-300.fc36.x86_64
    └── fedora-f198cfcbbb9c6e39690e4d72207017c592c5245afc3b2748407ff2db7b6ba600
        ├── initramfs-5.17.9-300.fc36.x86_64.img
        └── vmlinuz-5.17.9-300.fc36.x86_64

10 directories, 7 files
```

**Additional questions**

- Operating system and version: Fedora 36 Silverblue
- Have you tried rebooting? Yes
- Is this a regression? No
- Are you using an NVMe disk? Yes
- Is secure boot enabled? Yes
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? Not sure, I'll check

mripard, 2022-06-11:
Boot order lock is disabled in the UEFI

hughsie, 2022-06-11:
Does it work with secure boot turned off?

mripard, 2022-06-11:
It does, thanks for the suggestion

hughsie, 2022-06-11:
This is a known Fedora bug in the shim component. Fix in progress.

mripard, 2022-07-25:
If the fix was supposed to be deployed in the late june update of shim-x64, the issue is still present

mripard, 2022-09-23:
This is still occurring on current Fedora. Would you have a link to that shim bug tracker issue? Thanks!

mripard, 2023-10-11:
This still happens with Fedora 39 beta. Could you reopen this issue?

mripard, 2023-12-11:
For reference, this was due to Silverblue not updating `/boot/` but `/usr/lib/ostree-boot/` when updating the shim. See https://github.com/coreos/rpm-ostree/pull/969/commits/dc0e8db5bd92e1f478a0763d1a02b48e57022b59

Copying over the content of `/usr/lib/ostree-boot/` to `/boot/` by hand using `sudo rsync -havP /usr/lib/ostree-boot/ /boot/` fixed it.

Thanks for your help :)