fwupd/fwupd #4741: Non-Lenovo hardware receiving update notifications for Lenovo ME firmware (but does not install)

cwsystem76, 2022-06-13:
**Describe the bug**

Multiple non-Lenovo systems are receiving firmware update notifications through LVFS for _Lenovo ThinkPad L3 gen2 Consumer ME Firmware_, version `240.35.1951`. Attempting to install the firmware update does nothing, but the notification remains.

I believe but am not 100% sure that this is the firmware package in question, and the upload date matches when we started seeing the issue: https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadR1FRN.firmware

**Steps to Reproduce**

On two System76 Meerkat systems (based on the 11th-gen Intel NUC platform) and one Tuxedo laptop (https://chat.pop-os.org/pop-os/channels/support/8nr99qb5eifh7kxamjafgg5puy), all running Pop!OS v22.04, the firmware update system (which uses `fwupd` v1.8.0 as a foundation) reports a Lenovo firmware update. I expect that we will see more of this over the course of the week.

**Expected behavior**

Lenovo firmware should not show up on non-Lenovo hardware. I think that Lenovo may have uploaded an update to LVFS that's too "wide", so to speak.

**fwupd version information**

```shell
ckw@flatline:~$ fwupdmgr --version
runtime   org.freedesktop.fwupd         1.8.0
runtime   org.freedesktop.fwupd-efi     1.0
runtime   com.dell.libsmbios            2.4
compile   org.freedesktop.gusb          0.3.10
runtime   org.kernel                    5.17.5-76051705-generic
compile   com.hughsie.libjcat           0.1.9
compile   org.freedesktop.fwupd         1.8.0
runtime   org.freedesktop.gusb          0.3.10
```

[Pop!OS has its own `fwupd` package](https://github.com/pop-os/fwupd) that allows us to use a newer version than what Ubuntu ships, but it's just a clone of the main v1.8.0 release at the moment. This package is installed in Pop!OS by default.

**fwupd device information**

We don't have the output for the devices just yet. I'm asking a customer with one of our Meerkats to provide this.

**Additional questions**

- Operating system and version:

Pop!OS v22.04

- Have you tried rebooting?

Yes.

- Is this a regression?

Don't think so.

hughsie, 2022-06-13:
Dupe of https://github.com/fwupd/fwupd/issues/4733 perhaps?

jackpot51, 2022-06-13:
It looks like it. Is a newer fwupd version than 1.8.0 required to solve this, or should refreshing the repository data fix it?

jackpot51, 2022-06-13:
Ok, good to hear. @cwsystem76 can you test if `fwupdmgr refresh --force` fixes this?

cwsystem76, 2022-06-13:
I'm asking a customer to do so right now, and I'll let the guy in the Mattermost w/ the Tuxedo laptop know as well.

cwsystem76, 2022-06-13:
> Per the following commands, I no longer see the firmware update for Lenovo listed:

Looks like this worked out for a customer. I'll leave it up to @jackpot51 or someone else to close this, but I think we're all good here.

closed 2022-06-13 by hughsie