fwupd/fwupd #4754: Failed to create '/boot/efi/EFI/arch    /fw': Invalid argument

binarynoise, 2022-06-17:
**Describe the bug**
`fwupdmgr update` fails because it can't create the folder `/boot/efi/EFI/arch    /fw` (note the 4 additional spaces).
I do have udisks2 installed, I tried reinstalling that and fwupd.
The real question is where it gets the additional spaces.

**Steps to Reproduce**
run `fwupdmgr update`

**Expected behavior**
it would create the folder without the spaces

**fwupd version information**
```shell
fwupdmgr --version
runtime   org.freedesktop.fwupd         1.8.1
runtime   org.freedesktop.fwupd-efi     1.3
compile   org.freedesktop.gusb          0.3.10
runtime   com.dell.libsmbios            2.4
runtime   org.kernel                    5.18.3-arch1-1
runtime   com.hughsie.libjcat           0.1.11
compile   com.hughsie.libjcat           0.1.11
compile   org.freedesktop.fwupd         1.8.1
runtime   org.freedesktop.gusb          0.3.10
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
pacman

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices
Precision 7510
│
├─CM236 Chipset LPC/eSPI Controller:
│ │   Gerätekennung:      71b31258b13a4b2793e529856a190f8fb02ad151
│ │   Aktuelle Version:   31
│ │   Anbieter:           Intel Corporation (PCI:0x8086)
│ │   GUIDs:              c60968d9-9a1d-5ae3-97b5-c0d3fadb56ae ← PCI\VEN_8086&DEV_A150
│ │                       d3f0e5be-48ac-5e84-b953-25e291652927 ← PCI\VEN_8086&DEV_A150&REV_31
│ │                       1b63b95c-00ea-535e-a9fe-c5780ba749a7 ← PCI\VEN_8086&DEV_A150&SUBSYS_102806D9
│ │                       2b4b9923-533f-54a1-b747-1776eeef22af ← PCI\VEN_8086&DEV_A150&SUBSYS_102806D9&REV_31
│ │                       cfd2b4c2-2913-584f-95d4-58fca2e8596d ← INTEL_SPI_CHIPSET\ID_PCH100
│ │   Device Flags:       • Internes Gerät
│ │                       • Kryptografische Hash-Verifizierung ist verfügbar
│ │ 
│ ├─BIOS:
│ │     Gerätekennung:    ff7dbf2f6e354a5727c6ce1c466230f38bd26ff0
│ │     Anbieter:         Intel Corporation (PCI:0x8086)
│ │     GUID:             2c7b8bd2-3e77-5730-aaf6-596136f5254d ← IFD\NAME_BIOS
│ │     Device Flags:     • Internes Gerät
│ │                       • Kryptografische Hash-Verifizierung ist verfügbar
│ │   
│ ├─Gigabit Ethernet:
│ │     Gerätekennung:    11188287b93230d58f85f059dfab93e1d59724bb
│ │     Anbieter:         Intel Corporation (PCI:0x8086)
│ │     GUID:             30ee4fad-d1cd-522f-9a32-e53f4a4269cc ← IFD\NAME_GBE
│ │     Device Flags:     • Internes Gerät
│ │                       • Kryptografische Hash-Verifizierung ist verfügbar
│ │   
│ └─Intel Management Engine:
│       Gerätekennung:    7ef8a531d2413174034556f12dff8aa3bb4a8c30
│       Anbieter:         Intel Corporation (PCI:0x8086)
│       GUID:             614566e7-4d60-55f5-9cb9-48c5e488a705 ← IFD\NAME_ME
│       Device Flags:     • Internes Gerät
│                         • Gerät ist gesperrt
│                         • Kryptografische Hash-Verifizierung ist verfügbar
│     
├─Core™ i7-6820HQ CPU @ 2.70GHz:
[... 131 lines cut ...]
│   │   ├── starfield
│   │   │   └── [ ... ]
│   │   └── Vimix
│   │       └── [ ... ]
│   └── x86_64-efi
│       └── [ long list of .mod files ]
├── initramfs-linux-fallback.img
├── initramfs-linux.img
├── initramfs-linux-lts-fallback.img
├── initramfs-linux-lts.img
├── intel-ucode.img
├── vmlinuz-linux
└── vmlinuz-linux-lts

13 directories, 387 files
```

**Additional questions**

- Operating system and version: Reborn OS (Arch Linux)
- Have you tried rebooting? Yes, several times
- Is this a regression? 
- Are you using an NVMe disk? No, normal SSD
- Is secure boot enabled? No
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? No

binarynoise, 2022-06-17:
What you also might find interesting / ask for

```shell
sudo /usr/bin/fwupdtool esp-list --verbose
17:57:58:0905 FuDebug              Verbose debugging enabled (on console 1)
17:57:58:0920 FuCommon             device /org/freedesktop/UDisks2/block_devices/sda3, type: 0fc63daf-8483-4772-8e79-3d69d8477de4, internal: 1, fs: ext4
17:57:58:0922 FuCommon             device /org/freedesktop/UDisks2/block_devices/sda1, type: c12a7328-f81f-11d2-ba4b-00a0c93ec93b, internal: 1, fs: vfat
17:57:58:0925 FuCommon             device /org/freedesktop/UDisks2/block_devices/sda4, type: 0fc63daf-8483-4772-8e79-3d69d8477de4, internal: 1, fs: ext4
17:57:58:0927 FuCommon             device /org/freedesktop/UDisks2/block
[fstab.txt](https://github.com/fwupd/fwupd/files/8929735/fstab.txt)
_devices/sda2, type: 0657fd6d-a4ab-43c4-84e5-0933c84b4f4f, internal: 1, fs: swap
Ausgewählter Datenträger: /org/freedesktop/UDisks2/block_devices/sda1
/boot/efi/EFI/ArchLinux/grubx64.efi
```

[fstab.txt](https://github.com/fwupd/fwupd/files/8929737/fstab.txt)
[udisk2 dump.txt](https://github.com/fwupd/fwupd/files/8929709/udisk2.dump.txt)

`/etc/fwupd/uefi_capsule.conf` only contains `[uefi_capsule]` and comments

hughsie, 2022-06-17:
What does /etc/os-release say? i.e. are the extra spaces there?

binarynoise, 2022-06-18:
Yes, they are there. I removed them and it works now. Thanks for the hint!

From that I get a feature request: trim the ID or use the directory that is already there or just create your own in /boot/efi/EFI. 
From my short experience with fwupd it seems it doesn't really matter how the directory is named.

superm1, 2022-06-18:
Any idea where the trailing spaces came from?

binarynoise, 2022-06-19:
Probably because I at some point edited that file and didn't notice the trailing spaces. Nano should have warned me.

superm1, 2022-06-20:
> just create your own in /boot/efi/EFI

The directory intentionally re-uses the OS provided one because the OS also provides shim which trusts the signatures used for fwupd.  This means that an Ubuntu signed fwupdx64.efi wouldn't be bootable by a shim from Fedora and vice versa.

The only sane way we could consume our own directory is if we had fwupx64.efi signed by Microsoft directly, or shipped our own shim and our own signed fwupdx64.efi.  I don't expect either of those to really fly with the distro ecosystem.

closed 2022-06-20 by hughsie