fwupd/fwupd #4774: Switch to "native" fwupd support for libsmbios

superm1, 2022-06-26:
**Is your feature request related to a problem? Please describe.**
Currently `plugin_dell` has a dependency on libsmbios.  This used to have a lot more purposes than it does today.  Today it's only really used for:

1. [Reading information](https://github.com/fwupd/fwupd/blob/main/plugins/dell/fu-dell-smi.c#L127) from certain types of legacy docks 
2. Reading [TPM information](https://github.com/fwupd/fwupd/blob/main/plugins/dell/fu-plugin-dell.c#L695).

IMO these are weak reasons to have a dependency on a library.  We've re-implemented library code for plenty of other plugins to avoid dependencies.  Because fwupd is a supported package in major distros those distros pull in libsmbios by default too.  I don't think that makes a lot of sense.

Furthermore the kernel has an [API](https://github.com/torvalds/linux/blob/master/Documentation/ABI/testing/dell-smbios-wmi) for performing these communications now on modern systems.  This is what libsmbios uses internally.

Granted the libsmbios library supports both a WMI and a legacy interface, all the systems that support type-C docks and have TPMs also support the WMI interface.

**Describe the solution you'd like**
* Add a native implementation for the communication with the kernel SMBIOS WMI API.
* Drop the dependency on libsmbios in fwupd.
* Update packaging to drop libsmbios
* Notify distros of this change so they can drop libsmbios from the supported packages (unless there are other reasons to keep it)

**Describe alternatives you've considered**
Keep things as is.  I think that's perfectly fine, it just leaves a much larger surface area for fwupd to be attacked; particularly as it's code that directly can access BIOS APIs.  I think the less library code we depend on the better.

hughsie, 2022-06-27:
>  I think the less library code we depend on the better

Agree.

superm1, 2022-06-28:
BTW - I forgot to share that there is sample code available with the kernel to demonstrate the native API.  You should be able to use this to build the fwupd code.
https://github.com/torvalds/linux/blob/master/tools/wmi/dell-smbios-example.c

superm1, 2022-08-24:
Any progress on this?  FYI some of the API just changed function names from `bios_attr` to `bios_settings` to clarify things, so if you started you will probably want to rebase on `main` again.

CragW, 2022-08-25:
I've just noticed that firmware-attributes not working on some platforms, very likely have to use `dell-smbios` for some time until issue gets resolved. I'll start working on it.

superm1, 2022-08-25:
That's unfortunate to hear.  Is it trending firmware bug or kernel bug?  I guess the easiest way to distinguish will be to check on Windows.

In the interim, can you set things up as a preferred to FwupdBiosSettings and fallback to dell-smbios?  Then we will at least be better prepared to remove the smbios support when the time is right.

hughsie, 2023-09-23:
Lets close this; I'm no longer building the libsmbios code in Fedora and nobody seems to have noticed.

superm1, 2023-09-23:
> Lets close this; I'm no longer building the libsmbios code in Fedora and nobody seems to have noticed.

What about upstream changes for it too though?

hughsie, 2023-09-25:
> What about upstream changes for it too though?

I don't think the code is that expensive to maintain (famous last words I'm sure) -- got any ideas? We could flip the default upstream I suppose.

superm1, 2023-09-25:
IMO I think it's a pretty big attack surface to keep in place "dead" code that only gets enabled for some distros that decide they need it for `$reasons`.  It's a whole plugin and related infrastructure.

I'd say the options are:
1. Rip most of it out.  I think the only "development" should be to port dell-esrt to use firmware-attributes API and to mark Dell TPM's as upgradable.
2. Rip it **all** out (including dell-esrt plugin)
3. Default to it enabled upstream, leave it as is.

hughsie, 2023-09-26:
> to mark Dell TPM's as upgradable

Is any of that hardware going to get updates? I suspect it's all very much EOL now, right?

superm1, 2023-09-26:
I'm out of the loop now but can't the newer systems still take UEFI capsule updates?

hughsie, 2023-09-26:
My gut instinct here is we should probably port the dell-esrt "enable capsule updates" feature to the sysfs interface. The libsmbios functionality in the dell plugin is only being used as a fallback in case the FU_HWIDS_KEY_PRODUCT_SKU isn't set. Is the sysinfo_get_dell_system_id() value available in sysfs anywhere?

superm1, 2023-09-26:
> The libsmbios functionality in the dell plugin is only being used as a fallback in case the FU_HWIDS_KEY_PRODUCT_SKU isn't set

The history with that is that it existed before the standardized HWID started to exist.  I think we can safely tear that out now.

I don't believe that's the only consumer for libsmbios.  The TPM stuff also  runs `fu_dell_execute_simple_smi` to get TPM information.  The TPM switching stuff I think we don't need anymore, but the things for marking a TPM updatable and count the flashes left, that part I'm not so sure about.

superm1, 2023-09-26:
I've pushed a few cleanups to a branch for review that tear out some of the stuff I know to be dead.  I think we're **a lot closer** to being able to drop libsmbios with those.

superm1, 2023-09-26:
The remaining things (in my mind):
1. Decide whether to implement native `/dev/wmi/dell-smbios` support or use firmware-attributes API for remaining callers (TPM flashes left and dell-esrt plugin)
2. Decide whether to keep TPM flashes left.  I lean on no.  Without mode switching, it's astronomically unlikely that someone runs firmware update enough times to need it.
3. If TPM flashes left goes; decide if there is ANYTHING left having for the Dell plugin.  IE|: should it just fold into a quirk for TPM plugin?
4. dell-esrt plugin: even if we switch to native `/dev/wmi/dell-smbios` for this I think it should be fallback code.  The primary way it's used should be firmware attributes API.

superm1, 2023-09-26:
Here's a stab at replacing the dell-esrt plugin: https://github.com/fwupd/fwupd/pull/6215

superm1, 2023-09-26:
Here's a stab at tearing out TPM flashes left / ownership checking: https://github.com/fwupd/fwupd/pull/6216

closed 2023-09-27 by superm1

hughsie, 2023-09-27:
Many thanks @superm1 -- it's much appreciated.

commit 4737fd15 (2023-09-27): Drop all references to libsmbios

This means there is no point to having a plugin_dell anymore.  Move
that check to plugin_uefi_capsule instead.

Fixes: #4774
commit 49181f6a (2023-09-27): Drop all references to libsmbios

This means there is no point to having a plugin_dell anymore.  Move
that check to plugin_uefi_capsule instead.

Fixes: #4774
commit e72f8749 (2023-09-27): Drop all references to libsmbios

This means there is no point to having a plugin_dell anymore.  Move
that check to plugin_uefi_capsule instead.

Fixes: #4774
commit b178a31b (2023-09-27): Drop all references to libsmbios

This means there is no point to having a plugin_dell anymore.  Move
that check to plugin_uefi_capsule instead.

Fixes: #4774