fwupd/fwupd #4823: HP Dev One doesn't update firmware on restart, just boots operating system

maxamillion, 2022-07-13:
**Describe the bug**
A clear and concise description of what the bug is.
```
$ fwupdmgr upgrade
Devices with no available firmware updates:
 • HP HD Camera
 • PC711 HFS001TDE9X073N
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade System Firmware from 251920384 to 251985920?                         ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This release contains the following changes:                                 ║
║                                                                              ║
║ • Correct erroneous 90b fan error reporting                                  ║
║                                                                              ║
║ Dev One Notebook PC must remain plugged into a power source for the          ║
║ duration of the update to avoid damage.                                      ║
╚══════════════════════════════════════════════════════════════════════════════╝

Perform operation? [Y|n]: Y
Downloading…             [***************************************]
Downloading…             [***************************************]
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Waiting…                 [************************************** ]Successfully installed firmware
Waiting…                 [***************************************]

An update requires a reboot to complete. Restart now? [y|N]: y

```

on reboot, it just boots the operating system

**Steps to Reproduce**
Steps to reproduce the behavior.

listed above

**Expected behavior**
A clear and concise description of what you expected to happen.
reboot into firmware update process

**fwupd version information**
Please provide the version of the daemon and client.

```shell
$ fwupdmgr --version
runtime   org.freedesktop.fwupd         1.8.1
runtime   org.freedesktop.fwupd-efi     1.3
compile   org.freedesktop.gusb          0.3.10
runtime   com.dell.libsmbios            2.4
runtime   org.kernel                    5.18.10-200.fc36.x86_64
compile   com.hughsie.libjcat           0.1.10
compile   org.freedesktop.fwupd         1.8.1
runtime   org.freedesktop.gusb          0.3.10
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
$ fwupdmgr get-devices --show-all
Dev One Notebook PC
│
├─Unknown Device:
│     Device ID:          f685512aa07369c9e77742acef941d779d31e766
│     GUID:               37b440a9-2473-5087-a39b-db84f32a8ed8 ← GPIO\ID_AMDI0030:00
│
├─HP HD Camera:
│     Device ID:          3fa281ddf80d8a06b8ee5d8beb48d38ee95a9627
│     Current version:    0.2
│     Vendor:             Quanta (USB:0x0408)
│     Serial Number:      0001
│     GUIDs:              186b213b-1287-57a1-a95e-ca33e0598d74 ← USB\VID_0408&PID_5484
│                         8f5a0c1c-a6dc-5cb2-a247-bbf3245cec7f ← USB\VID_0408&PID_5484&REV_0002
│     Device Flags:       • Updatable
│
├─PC711 HFS001TDE9X073N:
│     Device ID:          03281da317dccd2b18de2bd1cc70a782df40ed7e
│     Summary:            NVM Express solid state drive
│     Current version:    HPS0
│     Vendor:             SK hynix (NVME:0x1C5C)
│     Serial Number:      ASB3N553811003E42
│     GUIDs:              0ee57466-ca9d-5f10-a005-ec6bacf6fb82 ← NVME\VEN_1C5C&DEV_174A
│                         4406b8dc-20c5-5f3a-bc9c-8afdcee48684 ← NVME\VEN_1C5C&DEV_174A&REV_00
│                         34e06406-f620-5b9d-a9cd-7f3a4e648633 ← NVME\VEN_1C5C&DEV_174A&SUBSYS_1C5C174A
│                         34f9d6fa-e1d6-5d7d-909e-0406644298c5 ← NVME\VEN_1C5C&DEV_174A&SUBSYS_1C5C174A&REV_00
│                         a9fd7eb6-3738-5c03-9e0e-fdfd6efd01fb ← SK hynix PC711 HFS001TDE9X073N
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│                         • Signed Payload
│
├─Ryzen 7 PRO 5850U with Radeon Graphics:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Vendor:             AMD
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f ← cpu
│                         22f9ecf4-588d-5c0a-8326-6ebff3655c6d ← CPUID\PRO_0&FAM_19
│                         79759cdc-94db-5098-be7b-eb02521fbbec ← CPUID\PRO_0&FAM_19&MOD_50
│                         20b595b0-5892-5870-8e4c-688133ad6e34 ← CPUID\PRO_0&FAM_19&MOD_50&STP_0
│     Device Flags:       • Internal device
│
├─System Firmware:
│     Device ID:          a45df35ac0e948ee180fe216a5f703f32dda163f
│     Summary:            UEFI ESRT device
│     Current version:    251920384
│     Minimum Version:    251920384
│     Vendor:             HP (DMI:Insyde)
│     Update State:       Needs reboot
│     Last modified:      2022-07-13 12:56
│     GUIDs:              9f7962c7-7d0e-473d-8a01-c168912f1f14
│                         230c8b18-8d9b-53ec-838b-6cfc0383493a ← main-system-firmware
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Cryptographic hash verification is available
│                         • Device is usable for the duration of the update
│
└─TPM:
      Device ID:          c6a80ac3a22083423992a3cb15018989f37834d6
      Current version:    7.85.17.51968
      Vendor:             Infineon (TPM:IFX)
      GUIDs:              ff71992e-52f7-5eea-94ef-883e56e034c6 ← system-tpm
                          5eebb112-75ad-5536-b173-a11eb3399402 ← TPM\VEN_IFX&DEV_0000
                          ddf995da-1b32-5a8a-bc1b-8d5af4b38b51 ← TPM\VEN_IFX&MOD_SLB9670
                          6d81ab63-db2e-50ac-934f-6be9accf5e02 ← TPM\VEN_IFX&DEV_0000&VER_2.0
                          301555de-680d-5ddc-b995-7553fc9138f1 ← TPM\VEN_IFX&MOD_SLB9670&VER_2.0
      Device Flags:       • Internal device
```

**Additional questions**

- Operating system and version: Fedora Workstation 36
- Have you tried rebooting? Yes
- Is this a regression? Unknown

Also, after running `fwupdmgr upgrade` and telling it *not* to reboot, my boot order doesn't appear to be changed:
```
$ efibootmgr
BootCurrent: 0001
Timeout: 0 seconds
BootOrder: 0001,2001,3001,2002,2004
Boot0001* Fedora
Boot0002* Fedora
Boot2001* EFI USB Device
Boot3001* Internal Hard Disk or Solid State Disk
```

hughsie, 2022-07-13:
@jackpot51 hey I see https://github.com/pop-os/fwupd/commit/c7132baee36019a3b6c039a28825e4c13b7167d6 -- did you mean to upstream a version of that? Does the HP Dev One advertise CoD support but not actually use it?

hughsie, 2022-07-13:
@maxamillion does the `DisableCapsuleUpdateOnDisk=true` trick work? If so, we should probably add a quirk based on the HwID to control that per-model.

superm1, 2022-07-13:
Fwiw I see something similar on the Insyde BIOS core on some other platforms that it advertised capsule support but it doesn't work in fwupd.

I wonder if we shouldn't be offering it at all on x86 right now...

maxamillion, 2022-07-13:
> @maxamillion does the `DisableCapsuleUpdateOnDisk=true` trick work? If so, we should probably add a quirk based on the HwID to control that per-model.

That worked! :)

superm1, 2022-07-13:
Too bad we JUST tagged 1.8.2. this would have been perfect to put in :/

hughsie, 2022-07-13:
Ahh, releases are cheap; 1.8.2 just took a long time because it was 6 weeks since the last release and I insisted on making lots of low level changes to most of the plugins :)

maxamillion, 2022-07-13:
Since that change in the configuration file worked, I'm happy to close unless there's something further y'all would like to address in this issue ticket. Thank you all! <3

closed 2022-07-14 by hughsie