fwupd/fwupd #4826: BIOS Control read incorrectly on APL and GLK

Sean-StarLabs, 2022-07-13:
fwupd seems to think that SPI is locked and write isn't enabled with coreboot on APL and GLK so fwupd won't allow updating or swtiching-branch.

I wrote the patches to set BIOS Control, so I'm 99.9% sure it's correct (before, it was empty so unsurprisingly didn't work!). fwupd currently shows:
```
[1mHSI-1[0m
✔ Platform Debugging:            [32m[1mDisabled[0m
✔ SPI lock:                      [32m[1mEnabled[0m
✔ SPI write:                     [32m[1mDisabled[0m
✔ Supported CPU:                 [32m[1mValid[0m
✔ TPM empty PCRs:                [32m[1mValid[0m
✔ TPM v2.0:                      [32m[1mFound[0m
✔ UEFI platform key:             [32m[1mValid[0m
✘ SPI BIOS region:               [31m[1mUnlocked[0m
```

On APL and GLK, for device 31, 0xdc, I have:
`0x8b = 1000 1010`

For comparison, AMI (which fwupd reports SPI write and lock correctly for) shows:
`0x82 = 1000 0010`

Bits 2:3 are to do with pre-fetching and caching so unrelated.

Issue is present with master. Any ideas?

hughsie, 2022-07-14:
> for device 31, 0xdc, I have

Is that the correct device and address on APL and GLK? Nothing obvious springs to mind tbh.

Sean-StarLabs, 2022-07-14:
Think so ;)

Quirk to ignore BIOS Control when on coreboot and be done with it?

hughsie, 2022-07-14:
> Quirk to ignore BIOS Control when on coreboot and be done with it?

Maybe, although that seems like working around the root issue. Can you debug a bit more and see what you can discover?

closed 2022-08-16 by Sean-StarLabs