fwupd/fwupd #4827: UEFI Update fails to update for Carbon X1 Gen 8

maddler, 2022-07-13:
**Describe the bug**
Firmware  update fails to install on a Lenovo Carbon X1 Gen 8.
Updates are downloaded corectly but install fails with "No supported devices found". The update description seems to be seems to be pointing update for T14s are being downloaded (see get-updates output). 

**Steps to Reproduce**

1. refresh available updates with fwupdmgr get-updates
```
root@nosferatu:~# fwupdmgr get-updates
Devices with no available firmware updates: 
 • Bolt Receiver
 • Integrated Camera
 • Radio
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI dbx
Devices with the latest available firmware version:
 • Embedded Controller
 • Intel Management Engine
 • Prometheus
 • Prometheus IOTA Config
 • System Firmware
 • WDC PC SN730 SDBQNTY-1T00-1001
20U9CTO1WW
│
└─UEFI Device Firmware:
  │   Device ID:          f37fb01122dd62c773f4e84ec89737e059712d59
  │   Summary:            UEFI ESRT device
  │   Current version:    0.0.22552
  │   Minimum Version:    0.0.1
  │   Vendor:             Lenovo (DMI:LENOVO)
  │   Update State:       Success
  │   Update Message:     		<p>ThinkPad Camera System Firmware version 6004</p> 		<p>The computer will be restarted and perform the Camera Firmware update. Do NOT turn off your computer or remove the AC adaptor while update is in progress. </p>
  │   GUID:               9e302497-0dc6-40f0-bf65-0108862ec500
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Device is usable for the duration of the update
  │ 
  ├─ThinkPad T14s:
  │     New version:      0.0.24580
  │     Remote ID:        lvfs
  │     Release ID:       13531
  │     Summary:          ThinkPad T14s Narrow Hybrid Camera Firmware
  │     License:          Proprietary
  │     Size:             187.4 kB
  │     Created:          2016-07-08
  │     Urgency:          High
  │     Vendor:           Lenovo
  │     Update Message:   		<p>ThinkPad Camera System Firmware version 6004</p> 		<p>The computer will be restarted and perform the Camera Firmware update. Do NOT turn off your computer or remove the AC adaptor while update is in progress. </p>
  │     Release Flags:    • Is upgrade
  │     Description:      
  │     ThinkPad Camera System Firmware version 6004
  │     
  │     The computer will be restarted and perform the Camera Firmware update.
  │     
  │     Do NOT turn off your computer or remove the AC adaptor while update is in progress.
  │   
  └─ThinkPad T14s:
        New version:      0.0.22808
        Remote ID:        lvfs
        Release ID:       12652
        Summary:          ThinkPad T14s Narrow Hybrid Camera Firmware
        License:          Proprietary
        Size:             187.4 kB
        Created:          2016-07-08
        Urgency:          High
        Vendor:           Lenovo
[... 341 lines cut ...]
├── System.map-5.17.15-76051715-generic
├── System.map-5.17.5-76051705-generic
├── vmlinuz -> vmlinuz-5.15.0-41-generic
├── vmlinuz-5.15.0-41-generic
├── vmlinuz-5.17.15-76051715-generic
├── vmlinuz-5.17.5-76051705-generic
└── vmlinuz.old -> vmlinuz-5.17.15-76051715-generic

13 directories, 40 files
```

**Additional questions**

- Operating system and version:
Ubuntu 22.04 LTS
- Have you tried rebooting?
Yes
- Is this a regression?
No
- Are you using an NVMe disk?
No
- Is secure boot enabled?
No
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS?
No

hughsie, 2022-07-13:
> No supported devices found

This is *weird*. If you use the recently-released 1.8.2 do you get a better error?

maddler, 2022-07-13:
Yup, pretty weird indeed. And, to me, that "  ├─ThinkPad T14s" in the get-updates output looks weird too.
Can't ben 100% but I don't thin I've ever seen that before.

hughsie, 2022-07-14:
> that " ├─ThinkPad T14s" in the get-updates output looks weird too

Ya, Lenovo pushed two bad updates. I've moved them back to embargo, and told them about all the broken things and asked them to fix it before they push it again.

maddler, 2022-07-14:
Two consecutive borked updates is evil but, at least, this makes (some sort of) sense.

hughsie, 2022-07-15:
@maddler Can you try with https://fwupd.org/downloads/1d4880e48999d1cf0b00a04e8081000823e2f154186fd7e7e61a20d579eab85a-Lenovo-ThinkPad-X1Carbon8th-X1Yoga5th-CameraFirmware-5918.cab please.

maddler, 2022-07-15:
Ok, confirmed working.
Thanks!

closed 2022-07-16 by hughsie

thekidconnect, 2023-11-09:
> @maddler Can you try with https://fwupd.org/downloads/1d4880e48999d1cf0b00a04e8081000823e2f154186fd7e7e61a20d579eab85a-Lenovo-ThinkPad-X1Carbon8th-X1Yoga5th-CameraFirmware-5918.cab please.

Will this fix Ubuntu not detecting my X1 Gen 8 camera? This link is now 404

mrhpearson, 2023-11-15:
X1 Carbon G8?
If Ubuntu isn't detecting the camera you likely have a HW issue - the FW update was (I believe) just supposed to improve image quality.