fwupd/fwupd #4830: vbe firmware-accessible state

sjg20, 2022-07-14:
This came up in the VBE discussion.

As per the doc [1] we need a way for firmware to report the state of the last update. I'd like it to say the time of the update and its status. Can we discuss again what state information the firmware can easily read? This is separate from the 'firmware state', which is a very small amount of non-volatile data for state signalling.

[1] https://docs.google.com/document/d/1mKahOBDZ2r9ZmoD0uuwuNDva_AHZoGScdTTfeW06X4I/edit?resourcekey=0-nLwaOkk6opE2w-s4zkOmvA#bookmark=id.fcxs3tvl5e7h

hughsie, 2022-07-14:
Can we use the existing `pending.db` history database? https://github.com/fwupd/fwupd/blob/main/src/fu-history.c#L980

i.e. we're storing `device_modified` there and `update_state` and so it seems a perfect fit; no?

hughsie, 2022-07-14:
e.g. try `sqlite3 /var/lib/fwupd/pending.db .d`

sjg20, 2022-07-14:
Yes the info looks useful although I suspect VBE will want more detail in some areas.

But I don't think we can use sqlite in firmware :-) The device tree is quite compact, small code size and already supported there. Can we build a parallel struct in a .dtb file, perhaps?

hughsie, 2023-02-20:
We talked about this off-bug -- I think we've got a plan now.

closed 2023-02-20 by hughsie