fwupd/fwupd #4837: Thunderbolt host controller -- No vendor ID set

jacobrosenthal, 2022-07-15:
**Describe the bug**
Thunderbolt controllers show `Update Error:       No vendor ID set`

Related?
https://github.com/fwupd/fwupd/issues/2200
https://github.com/fwupd/fwupd/issues/4438

Random Note I had to do `OverrideESPMountPoint=/boot/efi` fix as well

**fwupd version information**
```shell
$ fwupdmgr --version
runtime   org.freedesktop.fwupd         1.7.5
compile   org.freedesktop.gusb          0.3.10
runtime   com.dell.libsmbios            2.4
runtime   org.kernel                    5.15.0-41-generic
compile   org.freedesktop.fwupd         1.7.5
runtime   org.freedesktop.gusb          0.3.10
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
Apt

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
XPS 13 9300
│
├─Core™ i5-1035G1 CPU @ 1.00GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x000000b0
│     Vendor:             Intel
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f
│                         30249f37-d140-5d3e-9319-186b1bd5cac3
│                         d70b62bc-d9c6-5156-bd86-4f7440ccb34d
│                         95b2821b-aba1-51cc-ac93-bf7518dbceb0
│     Device Flags:       • Internal device
│   
├─Integrated Webcam HD:
│     Device ID:          4295296d98b3ba38c72f6baa33d24f03a1d428f6
│     Current version:    96.28
│     Vendor:             CN0PW36V8LG009CIA697A00 (USB:0x0BDA)
│     GUIDs:              d784b200-4ead-55f2-8277-15e1618cbe7b
│                         9a1becb6-4d9f-5855-99d0-145b1700da10
│                         d35eb209-22f9-59a5-a918-33d7307edab9
│     Device Flags:       • Updatable
│   
├─Iris Plus Graphics G1 (Ice Lake):
│     Device ID:          5792b48846ce271fab11c4a545f7a3df0d36e00a
│     Current version:    07
│     Vendor:             Intel Corporation (PCI:0x8086)
│     GUIDs:              9c1b0857-f7a6-5f00-836f-315f8b2851d9
│                         53f72484-2cb6-54b3-983e-2b4a560f34ad
│                         459c55aa-0a15-5b28-a9a5-8fd8c679fdfe
│                         45a1f72f-2603-529b-95ff-de701e682c9a
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
├─KBG40ZNS256G NVMe KIOXIA 256GB:
│     Device ID:          ac1ce3be70e444b92f99fe08c9b957efb9d0ee53
│     Summary:            NVM Express solid state drive
│     Current version:    10410106
│     Vendor:             KIOXIA Corporation (NVME:0x1E0F)
│     GUIDs:              35de42d4-bd35-5bcd-89fa-6e71cbd4d0ed
│                         5264a089-86d6-e811-9f8b-f2801f1b9fd1
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─System Firmware:
│ │   Device ID:          a45df35ac0e948ee180fe216a5f703f32dda163f
│ │   Summary:            UEFI ESRT device
│ │   Current version:    1.13.0
│ │   Minimum Version:    1.13.0
│ │   Vendor:             Dell (DMI:Dell Inc.)
│ │   Update State:       Success
│ │   GUIDs:              e8292593-e66e-4878-b051-f152535ab130
│ │                       230c8b18-8d9b-53ec-838b-6cfc0383493a
│ │                       29a2266e-9abb-57c9-b89e-58f2093f917b
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Device is usable for the duration of the update
│ │ 
│ └─UEFI dbx:
│       Device ID:        362301da643102b9f38477387e2193e57abaa590
│       Summary:          UEFI revocation database
│       Current version:  238
│       Minimum Version:  238
│       Vendor:           UEFI:Linux Foundation
│       Install Duration: 1 second
│       GUIDs:            00fe3755-a4d8-5ef7-ba5f-47979fbb3423
│                         4a6cd2cb-8741-5257-9d1f-89a275dacca7
│                         c6682ade-b5ec-57c4-b687-676351208742
│                         f8ba2887-9411-5c36-9cee-88995bb39731
│       Device Flags:     • Internal device
│                         • Updatable
│                         • Needs a reboot after installation
│                         • Only version upgrades are allowed
│     
├─Thunderbolt host controller:
│     Device ID:          8a49069ea116f1a1a42f5a47a1e5a869fafd332e
│     Summary:            Unmatched performance for high-speed I/O
│     Update Error:       No vendor ID set
│     GUID:               e72e778e-94f7-5ed2-b560-1c1262ee217c
│     Device Flags:       • Internal device
│                         • System requires external power source
│                         • Updatable
│   
└─Thunderbolt host controller:
      Device ID:          550be74e0bf57bbc1f64ddc1efa4635dd5c24db0
      Summary:            Unmatched performance for high-speed I/O
      Update Error:       No vendor ID set
      GUID:               e72e778e-94f7-5ed2-b560-1c1262ee217c
      Device Flags:       • Internal device
                          • System requires external power source
                          • Updatable

```

**Additional questions**

- Operating system and version:
Dell xps 13 9300
ubuntu 22.04
- Have you tried rebooting?
 Yep
- Is this a regression?
Never seen it before

hughsie, 2022-07-17:
@superm1 maybe the devices are in safe mode?

superm1, 2022-07-17:
You can't update firmware for soc integrated controllers from fwupd directly. 
This is a duplicate of some of those linked bugs you mentioned. If you look at the outcome of them it was to not mark it updatable or set an error string.

closed 2022-07-17 by superm1