fwupd/fwupd #4838: Lenovo M80t firmware update failure

kroeckx, 2022-07-17:
I've been trying to install a lenovo update:
```
11EVCTO1WW
│
└─System Firmware:
  │   Device ID:          a45df35ac0e948ee180fe216a5f703f32dda163f
  │   Summary:            UEFI ESRT device
  │   Current version:    65607
  │   Minimum Version:    65607
  │   Vendor:             Lenovo (DMI:LENOVO)
  │   Update State:       Success
  │   GUIDs:              5deafc1a-69b6-4a95-8572-2c888cabf533
  │                       230c8b18-8d9b-53ec-838b-6cfc0383493a ← main-system-firmware
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Cryptographic hash verification is available
  │                       • Device is usable for the duration of the update
  │ 
  └─ThinkCentre M90ts_ThinkCentre M80ts_ThinkCentre M70ts:
        New version:      65615
        Remote ID:        lvfs
        Release ID:       13217
        Summary:          System Firmware for ThinkCentre M90ts_ThinkCentre M80ts_ThinkCentre M70ts
        License:          Proprietary
        Size:             16.9 MB
        Created:          2022-04-29
        Urgency:          Low
        Vendor:           Lenovo
        Release Flags:    • Is upgrade
        Description:      
        This stable release fixes the following issues:
        
        • Enhancement to address security vulnerability. 
```

So I try to install it:
```
# fwupdmgr update
Devices with no available firmware updates: 
 • UEFI Device Firmware
 • UEFI dbx
Devices with the latest available firmware version:
 • Embedded Controller
 • Intel Management Engine
 • MZVLB1T0HBLR-000L7
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade System Firmware from 65607 to 65615?                                 ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This stable release fixes the following issues:                              ║
║                                                                              ║
║ • Enhancement to address security vulnerability.                             ║
║                                                                              ║
║ 11EVCTO1WW must remain plugged into a power source for the duration of the   ║
║ update to avoid damage.                                                      ║
╚══════════════════════════════════════════════════════════════════════════════╝

Perform operation? [Y|n]: 
Downloading…             [***************************************]
Downloading…             [***************************************]
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Waiting…                 [************************************** ]Successfully installed firmware
Waiting…                 [***************************************]

An update requires a reboot to complete. Restart now? [y|N]: 
```

[... 559 lines cut ...]
├── System.map-5.18.0-2-amd64
├── vmlinuz-5.15.0-1-amd64
├── vmlinuz-5.15.0-2-amd64
├── vmlinuz-5.15.0-3-amd64
├── vmlinuz-5.16.0-1-amd64
├── vmlinuz-5.16.0-3-amd64
├── vmlinuz-5.16.0-5-amd64
├── vmlinuz-5.16.0-6-amd64
├── vmlinuz-5.17.0-1-amd64
├── vmlinuz-5.18.0-1-amd64
└── vmlinuz-5.18.0-2-amd64

8 directories, 371 files
```

**Additional questions**

- Is this a regression?
  No idea, first time an update is available on this machine.
- Are you using an NVMe disk?
  Yes
- Is secure boot enabled?
  Yes
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS?
  I tried to look for a boot order lock in the bios but did not find any.

hughsie, 2022-07-17:
Does it work without secure boot turned on, by any chance?

kroeckx, 2022-07-17:
Yes, then it works.

hughsie, 2023-02-20:
I think we can close this; Lenovo had some updates go out unsigned, but I think it's been much better lately. If you can reproduce with anything released in the last 6 months, please reopen.